polarofficial/polar-ble-sdk #91: Demo app button "start recording" returns error 106 on H10 device

dephillipsmichael, 2020-08-20:
Here is the log stack from the sample app:

2020-08-20 09:37:49.083145-0700 polarBleSdkTestApp[358:12884] [BLE] PS-FTP new query operation
2020-08-20 09:37:49.147281-0700 polarBleSdkTestApp[358:12884] [BLE] RFC76 message response: 106
2020-08-20 09:37:49.148041-0700 polarBleSdkTestApp[358:12884] [BLE] PS-FTP query operation DISPOSED
2020-08-20 09:37:49.148631-0700 polarBleSdkTestApp[358:12668] recording start fail: responseError(errorCode: 106)

Any idea what is going on?  Do I need to put the H10 in a certain state before I can start recording?

erkki-silvola, 2020-09-06:
Hi, 

recording is probably already enabled, check status with "recording status". And if recording is on stop it first.

pablobablo, 2020-11-18:
I have the same error. I've tried everything - no results. Polar H10 (3.0.56). 
Any ideas?

JOikarinen, 2020-12-15:
@pablobablo did you try to request the _recording status_? If the H10 is already in record _ongoing_  state then the start  operation is not permitted?

JOikarinen, 2020-12-29:
> @pablobablo did you try to request the _recording status_? If the H10 is already in record _ongoing_ state then the start operation is not permitted?

@pablobablo and @dephillipsmichael update to my previous post. Error code 106 means "OPERATION NOT PERMITTED".

H10 returns such error in few cases:

1. recording is already started and `startRecoding` attempted
2. recording is already stopped and `stopRecoding` attempted
3. recording start is attempt with already existing `exerciseId` found in H10

If using Sample app the case 3 is likely to happen as recording is always started with same `exerciseId`. You may Remove previously recorded exercise with command `removeExercise`

closed 2020-12-29 by JOikarinen

bartekhugo, 2021-01-20:
From my observations, you can only have **one** existing recording stored in memory, to start another you need to remove existing first. At leat for H10 device. not sure how big that one recording can be

JOikarinen, 2021-01-21:
@bartekhugo good correction. H10 can have only one exercise stored in the memory at once.  Correction to my previous statement:

> 3. recording start is attempt with already existing `exerciseId` found in H10
3. recording start is attempt if there is already existing exercise stored in H10 memory.

JOikarinen, 2021-01-21:
About the H10 exercise storage:
- [Polar H10 support material](https://support.polar.com/uk-en/support/how_long_a_training_session_can_i_record_with_h10) says +30 hours
- 30h is good estimate when exercise recording is started with options: 1s recording interval and using sample type HR: 
`api.startRecording(DEVICE_ID, "TEST_APP_ID", PolarBleApi.RecordingInterval.INTERVAL_1S, PolarBleApi.SampleType.HR)`