polarofficial/polar-ble-sdk #92: Pmd response failed in receive in timeline

jul1u5, 2020-09-07:
Hi,
My app sometimes randomly crashes with this error:
```
E/AndroidRuntime: FATAL EXCEPTION: DefaultDispatcher-worker-6
    Process: com.github.jul1u5.app, PID: 12072
    java.lang.Exception: Pmd response failed in receive in timeline
        at com.androidcommunications.polar.api.ble.model.gatt.client.BlePMDClient.receiveControlPointPacket(BlePMDClient.java:673)
        at com.androidcommunications.polar.api.ble.model.gatt.client.BlePMDClient.sendPmdCommand(BlePMDClient.java:678)
        at com.androidcommunications.polar.api.ble.model.gatt.client.BlePMDClient.lambda$sendControlPointCommand$10$BlePMDClient(BlePMDClient.java:701)
        at com.androidcommunications.polar.api.ble.model.gatt.client.-$$Lambda$BlePMDClient$46eXajt5wXOZTLKH-N3q6T3puF0.subscribe(Unknown Source:6)
        at io.reactivex.rxjava3.internal.operators.single.SingleCreate.subscribeActual(SingleCreate.java:40)
        at io.reactivex.rxjava3.core.Single.subscribe(Single.java:4813)
        at io.reactivex.rxjava3.internal.operators.single.SingleSubscribeOn$SubscribeOnObserver.run(SingleSubscribeOn.java:89)
        at io.reactivex.rxjava3.core.Scheduler$DisposeTask.run(Scheduler.java:589)
        at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.run(ScheduledRunnable.java:65)
        at io.reactivex.rxjava3.internal.schedulers.ScheduledRunnable.call(ScheduledRunnable.java:56)
        at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:301)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at java.lang.Thread.run(Thread.java:919)
```

What is `Pmd response`? What could be the reason for the error?

jul1u5, 2020-09-08:
I noticed that this only happens when I connect to OH1 with low battery level.

JOikarinen, 2020-12-15:
@jul1u5 it seems that command send to the OH1 is timing out, so OH1 never responded to command you initialised. The reason most probably being the battery is too low in OH1. Best you can do is handle the exception on your code with Rx onError()

JOikarinen, 2020-12-15:
We will have a look, how to enhance the API to better expose if sensor is not responding because of low battery.

closed 2020-12-15 by JOikarinen