polarofficial/polar-ble-sdk #93: Connect 30/35 Polar H10 sensor using SDK

jams032, 2020-09-08:
Hi, SDK seems perfect.

Can we connect 30/35 Polar sensors at a time using this SDK ? We are planning to develop an iOS app to perform group session of 30/35 users. 

What would be workaround if we want to do so ?

Thanks.

JOikarinen, 2020-11-29:
@jams032 the maximum connected devices is phone and operating system dependant. 

One option is to listen data from multiple sensors via broadcast. From SDK you may use https://github.com/polarofficial/polar-ble-sdk/blob/9d7f21f8aa6b1967e6a3b92c9015ca8fee026578/sources/iOS/ios-communications/iOSCommunications/sdk/api/PolarBleApi.swift#L188

closed 2020-11-29 by JOikarinen

igorgcustodio, 2021-07-14:
Sorry for opening this issue again, but I prefer to reopen it than create a new one.

I was searching about BLE and I saw that I can connect up to 7 devices. In this case, connecting 30 or more, do I just connect the devices to iPad and use this startListenForPolarHrBroadcasts or I need to do something different?

JOikarinen, 2021-07-15:
Hi @igorgcustodio, 

> I was searching about BLE and I saw that I can connect up to 7 devices. 
- I still believe it is phone, OS and  OS version dependant how many BLE connections you are able to create. Let say your application is only installed on iPad and the maximum amount BLE connections on the iPad is satisfying your needs then you could  implement your system so that you create BLE connection on to the each of the interesting devices. 

> In this case, connecting 30 or more, do I just connect the devices to iPad and use this startListenForPolarHrBroadcasts or I need to do something different?

- when listening the broadcast you do not need to create connection to devices your are interested. You may read more about BLE network topology from docs found in internet, e.g. https://www.oreilly.com/library/view/getting-started-with/9781491900550/ch01.html

-  `startListenForPolarHrBroadcasts` implementation scans for nearby devices and their broadcast data. If any Polar devices is found broadcasting then it tries to parse broadcast packet to see if it contains heart rate data. If heart rate data is found from broadcast then that [data](https://github.com/polarofficial/polar-ble-sdk/blob/master/sources/Android/android-communications/src/sdk/java/com/polar/sdk/api/model/PolarHrBroadcastData.java) is passed to API client (i.e. your application) on reactive stream.