polarofficial/polar-ble-sdk #95: OH1 ignores `setLocalTime` until restart

jul1u5, 2020-09-08:
In my app, I synchronize the connected devices via `setLocalTime` method. H10 sets its local time immediately and that's great. However, OH1 only applies the time when it has been turned off and on again.

JOikarinen, 2021-01-15:
@jul1u5 , could you please provide more detailed steps? Are you using iOS or Android SDK? I have been trying to repeat the problem but in my testing the time is set correctly every time. Please reopen if you can reproduce the issue still.

closed 2021-01-15 by JOikarinen

JOikarinen, 2021-06-08:
@jul1u5 today I realised the obvious steps how to reproduce this.

STEPS:
- start receive ACC stream and observe the timestamp:
    - observed time stamp is 536580371079656926 
- set the time with SDK:  time set to device Tue Jun 08 08:40:09 GMT+03:00 2021
- start receive ACC stream and observe the timestamp
    - observed time stamp is 536580557724485604 
- power off/ power on
- start receive ACC stream and observe the timestamp
    - observed time stamp is 676456950432535404