polarofficial/polar-ble-sdk #97: Swift 5.3

NiwreRelbag, 2020-09-17:
Can you compile for XCode 12, Swift 5.3 please.
Thx.

hm1205, 2020-09-23:
Hi, 
Any update on the subject ? Due to Apple updates of Xcode and timeline of updates, hope a soon update of polar sdk; if not could be annoying to update ours apps. 

best regards

outsbook, 2020-09-29:
Any update on this issue?

hm1205, 2020-10-01:
Hi, framework build from source as i had an unexpected free time during work time and seems to work ( both branches with or without swiftprotobuf). 
i can keep on working on my app, but waiting for polar official release.  
best regards

outsbook, 2020-10-01:
@hm1205 Can you please share the frameworks which you generated for swift 5.3, thanks

hm1205, 2020-10-01:
@outsbook better to give you the process I've been true to build

first update Carthage version to latest version

second fix Carthage to work with swift 5.3 and Xcode 12 : 

```sh 
#!/usr/bin/env bash

# carthage.sh
# Usage example: ./carthage.sh build --platform iOS

set -euo pipefail

xcconfig=$(mktemp /tmp/static.xcconfig.XXXXXX)
trap 'rm -f "$xcconfig"' INT TERM HUP EXIT

# For Xcode 12 make sure EXCLUDED_ARCHS is set to arm architectures otherwise
# the build will fail on lipo due to duplicate architectures.
echo 'EXCLUDED_ARCHS__EFFECTIVE_PLATFORM_SUFFIX_simulator__NATIVE_ARCH_64_BIT_x86_64__XCODE_1200 = arm64 arm64e armv7 armv7s armv6 armv8' >> $xcconfig
echo 'EXCLUDED_ARCHS = $(inherited) $(EXCLUDED_ARCHS__EFFECTIVE_PLATFORM_SUFFIX_$(EFFECTIVE_PLATFORM_SUFFIX)__NATIVE_ARCH_64_BIT_$(NATIVE_ARCH_64_BIT)__XCODE_$(XCODE_VERSION_MAJOR))' >> $xcconfig

export XCODE_XCCONFIG_FILE="$xcconfig"
carthage "$@"
```

credit to : rastersize  see [this post](https://github.com/Carthage/Carthage/issues/3019#issuecomment-699143260 ) 

then rebuild or build Rxswift from Carthage (using previous script) with '--no-use-binaries' ( avoid bitcode error ) and '--platform IOS' as arguments 

finally run build script:   build_sdk.sh

Hope will work for you. best regards

outsbook, 2020-10-01:
@hm1205 Thank you very much for your help, it's seems to work

ymatt345, 2020-10-17:
I am waiting for the official update of polar-bye-sdk.
I wish to provide us soon.

ymatt345, 2020-10-20:
Please inform us of your planned update release for Swift5.3.
All of our development efforts are suspended now.
Best regards,

josephlarralde, 2020-11-03:
@hm1205 Thanks ! It seems to work for me too. I just had to copy the generated frameworks from `Carthage/Build/iOS/RxSwift.framework` and `3rd_party_sdk/PolarBleSdk.framework` into `polar-sdk-ios` and the ios example now builds and runs fine.

Arwain, 2020-11-04:
@hm1205 how did you `3rd_party_sdk`? When I run carthage build I get `Carthage/Build/iOS/RxSwift.framework` and `Carthage/Checkouts/polar-ble-sdk/polar-sdk-ios/PolarBleSdk.framework`.

josephlarralde, 2020-11-04:
@Arwain to be precise, I ran `./carthage.sh update RxSwift --no-use-binaries --platform iOS`, which generated the `Carthage/Build/iOS/RxSwift.framework`, then I ran `./build_sdk.h` which generated the `3rd_party_sdk/PolarBleSdk.framework`. But `3rd_party_sdk` already existed anyway ... Does that help ?

Arwain, 2020-11-04:
@josephlarralde thank you for your reply and for breaking this down! I'm with you through the initial steps, however, after I run `./carthage.sh update RxSwift --no-use-binaries --platform iOS` I get the RxSwift.framework. 

No `./build_sdk.h` is generated. I think this is what I'm missing. What/where is this script generated? 
`zsh: no such file or directory: ./build_sdk.h`

josephlarralde, 2020-11-05:
@Arwain are you typing your commands from `sources/iOS/ios-communications` ?
Because that's where I run both scripts from.
`carthage.sh` is just @hm1205 's script, and `build_sdk.sh` (sorry just noticed the `.h` typo in my previous answer) is already in there.
That's also where the `3rd_party_sdk` folder is, in which the new PolarBleSdk.framework will be built.

closed 2020-11-23 by JOikarinen