polarofficial/polar-ble-sdk #98: Does the API work on a desktop environment?

kothiga, 2020-09-17:
Hi there,

I am wanting to build a desktop client that will be running from a Linux (Ubuntu) computer. I am wondering if the java version of the sdk (polar-ble-sdk.aar) will be able to interface with the devices from the bluetooth on said desktop computer; or is it a dead end because the sdk is relying on some android specific things.

Are there any alternative for easily interfacing with the Polar H10 and OH1? I've seen from the data specifications sheet the serial commands you would send to the devices to initiate streaming of data, but I am wondering if there are better solutions.

Thank you for your help.

erikboto, 2020-10-22:
I did my own implementation for reading e.g. PPI-values from the OH1 using python-gatt, but you could also do this using any framework that has decent support for handling bluetooth low energy. Something like Qt might be suitable if you're doing something with a GUI. I haven't seen anything ready-made that is ready to use.

weliem, 2020-10-22:
I wrote a Java library for using Bluetooth on Linux. It comes with an example app that also implements the HR service which will work with the H10/OH1. But it doesn't implement all the other functionality of those devices. You would still have to 'port' that yourself if you need it...

You can find the library here: https://github.com/weliem/blessed-bluez

nepho5, 2020-11-11:
> I did my own implementation for reading e.g. PPI-values from the OH1 using python-gatt, but you could also do this using any framework that has decent support for handling bluetooth low energy. [...]

Hi @erikboto,
I'm currently trying to achive the same: Reading PPI-measurement from the Polar OH1 with Python. Unfortunately, i can't get the sensor to start recording the data. I tried to follow the provided documentation and nothing seems zo help.
Since you were successfull, is there a chance i could get insights into your code? 
Best regards

JOikarinen, 2021-01-13:
Unfortunately, we are only able to support Android and iOS. The SKD includes source code and you can feel free to utilize / port the code in other platforms too but that is something we are not able to provide support for.

closed 2021-01-13 by JOikarinen