polarofficial/polar-ble-sdk #103: Is it  possible to get HR data(Polar H10) without connecting device?

kcbaruri, 2020-10-22:
I am working on a project to read HR data from Polar H10 but not able to connect more than 14 devices. 

Is there is any way to read HR data from Polar H10 without connecting to any device? i.e. if the person wearing the device comes in the boundary of the device where app is running. Looking for a solution like WASP based on ant+.

Thanks

kcbaruri, 2020-10-22:
In an iOS application, we have connected up to 14 devices and our requirement is to connect 35 devices.

jams032, 2020-10-22:
Adding queries to the ticket : 
What is the public advertising data/key you provide to get basic heart rate data without BLE connection ? Is it public for all  ? 
In iOS , i got this response : 

<CBPeripheral: 0x282c1cb40, identifier = 9439A626-2E5E-5F37-7110-F9726BF3451E, name = Polar H10 E7BAC71A, state = disconnected>

from func centralManager(_ central: CBCentralManager, didDiscover peripheral: CBPeripheral,
                      advertisementData: [String : Any], rssi RSSI: NSNumber) {
}

Anything to change here  ? 

let heartRateServiceCBUUID = CBUUID(string: "0x180D")
let heartRateMeasurementCharacteristicCBUUID = CBUUID(string: "2A37")
let bodySensorLocationCharacteristicCBUUID = CBUUID(string: "2A38")

jams032, 2020-10-26:
Solutions i have found : 

sol1: UInt16(manufacturerData[5]) , last part is HR data 

sol2 : broadcast = api.startListenForPolarHrBroadcasts(nil).observeOn(MainScheduler.instance).subscribe can provide the HR value.

Hope this is the right procedure.

JOikarinen, 2021-01-15:
> Solutions i have found :
> Solution 2 :
> 
> broadcast = api.startListenForPolarHrBroadcasts(nil).observeOn(MainScheduler.instance).subscribe
> can provide the HR value using SDK.
> 
> Hope these are the right procedures. Correct me if i'm wrong.

@kcbaruri as @jams032 suggests if you need to listen multiple sensors, in your case as many as 14,  the solution is to listen heart rate from the broadcast.

closed 2021-01-15 by JOikarinen

jaesungj, 2021-05-17:
Hi @JOikarinen  sorry for the dumb question here, but original question was  "Is it possible to get HR data(Polar H10) without connecting to device" how does this API : startListenForPolarHrBroadcasts work so that this doesn't require device connection?

JOikarinen, 2021-05-18:
@jaesungj no dump questions here, more questions more knowledge.👍

> 1. What happens when people stay in the range for a few seconds and moves away? Do the data get dropped? or if he comes back to the range will the uningested data will get processed?

- Data will get dropped, unfortunately that is down side in BLE broadcasting. If the the packet is lost then the data is lost too. 

> 2\. Do you have any good customer experience suggestion for the problem setting which I mentioned above?

- I understand your problem. Polar has solution for [Clubs](https://www.polar.com/en/club). In the Polar Club the  head unit (usually iPad) is receiving the heart rate from heart rate sensors via broadcasting. In the Polar Club solution the broadcasting is working well, it can tolerate single lost packets every now and then.  

 - If your solution cannot tolerate the lost BLE packets (i.e. lost heart rate), then the broadcasting is not the best approach. Then you could think of the approach that you save the data into the H10. The way to do that is roughly this:
 1. Start recording https://github.com/polarofficial/polar-ble-sdk/blob/54ec381aaff68c959ae4eed4500220568e2cc09e/sources/Android/android-communications/src/main/java/polar/com/sdk/api/PolarBleApi.java#L236
 2. Stop recording https://github.com/polarofficial/polar-ble-sdk/blob/54ec381aaff68c959ae4eed4500220568e2cc09e/sources/Android/android-communications/src/main/java/polar/com/sdk/api/PolarBleApi.java#L247
 3. Fetch exercises https://github.com/polarofficial/polar-ble-sdk/blob/54ec381aaff68c959ae4eed4500220568e2cc09e/sources/Android/android-communications/src/main/java/polar/com/sdk/api/PolarBleApi.java#L272

JOikarinen, 2021-05-18:
> 3\. Why does the api input take the list of device ids? what difference does it make from the response perspective when we provide null, and the actual list of devices? Since I would be listening to those 40 devices does it mean I have to provide 40 devices ids?
- the list is for filtering the devices your are interested in. If you set parameter null you get broadcast data of all Polar devices around and by providing the list of device id's you get the data from devices your are interested in.

JOikarinen, 2021-05-18:
> 3\. Why does the api input take the list of device ids? what difference does it make from the response perspective when we provide null, and the actual list of devices? Since I would be listening to those 40 devices does it mean I have to provide 40 devices ids?

- I made small fix to the [startListenForPolarHrBroadcasts]( https://github.com/polarofficial/polar-ble-sdk/blob/e63c70aa713c6f13f6bf2022c7bad7c45ca3221e/sources/Android/android-communications/src/main/java/polar/com/sdk/api/PolarBleApi.java#L304 ) API documentation in e63c70aa713c6f13f6bf2022c7bad7c45ca3221e as the documentation was a bit of misleading.

jaesungj, 2021-05-18:
> @jaesungj no dump questions here, more questions more knowledge.👍
>
>
> * Data will get dropped, unfortunately that is down side in BLE broadcasting. If the the packet is lost then the data is lost [...]

So just to confirm if I were to build some sort of application using the record approach. 

This will be the CX
1. Customer hits the button in the mobile app to start recording
2. Customer starts to do some exercising 
3. Customer hits the button in the mobile app to stop recording
4. Customer hits the button in the mobile app to fetch data (Business logic will get executed acted on the data)

How many hours worth of hear-beat data can be stored?

JOikarinen, 2021-05-19:
> **Q. So just to confirm if I were to build some sort of application using the record approach.**
> This will be the CX
> 
> 1. Customer hits the button in the mobile app to start recording
> 2. Customer starts to do some exercising
> 3. Customer hits the button in the mobile app to stop recording
> 4. Customer hits the button in the mobile app to fetch data (Business logic will get executed acted on the data)

- sounds feasible approach.

JOikarinen, 2021-05-19:
> Q. How many hours worth of hear-beat data can be stored?
- it depends a lot of the parameters given in `startRecording`
```
 public abstract Completable startRecording(@NonNull final String identifier, @NonNull @Size(min = 1, max = 64) final String exerciseId, @NonNull RecordingInterval interval, @NonNull SampleType type);
```
 If `RecordingInterval` is 1s and `SampleType` is HR then the [statement](https://support.polar.com/en/support/how_long_a_training_session_can_i_record_with_h10?product_id=89452&category=training) from H10 support page says: "the training session can have data up to +30 hours". If the `RecordingInterval` is 5s and `SampleType` is HR then training session can have data up to 5x30hours.

JOikarinen, 2021-05-19:
> **Q. For the broadcasting mobile solutions to work what is the blue tooth coverage where people have to stay within?**
> **Q. For the broadcasting club solutions to work what is the blue tooth coverage where people have to stay within?**
- here is good explanation:
https://support.polar.com/en/support/what_is_the_bluetooth_transmission_range_of_polar_h10_hr_sensor

JOikarinen, 2021-05-19:
> Q. if the coverage is different between the broadcasting mobile, and broadcasting club why are they different? Even though both uses the same broadcasting api?
- the coverage is same

JOikarinen, 2021-05-19:
@jaesungj if you choose to record the data into the H10 there is limitation that only one file at the time can be saved on H10 memory (I don't know where that limitation origins). However, it means that `startRecording` will return error if there is already existing exercise file saved in H10 memory. It is good practise to call `removeExercise`, when ever data is fetched data device by `fetchExercise` call

jaesungj, 2021-05-19:
Thanks a lot!