polarofficial/polar-ble-sdk #104: Different sampling rate for PPG and Acc?

SijieZhuo, 2020-10-22:
HI, I have been using the same way to stream data as shown in the example provided.

public void streamPPG() {
        if (ppgDisposable == null) {
            ppgDisposable =
                    api.requestPpgSettings(DEVICE_ID).toFlowable().flatMap(new Function<PolarSensorSetting, Publisher<PolarOhrPPGData>>() {
                        @Override
                        public Publisher<PolarOhrPPGData> apply(PolarSensorSetting sensorSetting) throws Exception {
                            return api.startOhrPPGStreaming(DEVICE_ID,
                                    sensorSetting.maxSettings());
                        }
                    }).observeOn(AndroidSchedulers.mainThread()).subscribe(
                            new Consumer<PolarOhrPPGData>() {
                                @Override
                                public void accept(PolarOhrPPGData polarPPGData) throws Exception {
                                    for (PolarOhrPPGData.PolarOhrPPGSample data : polarPPGData.samples) {
//                                        ppgDataObservable.onNext(data);
                                        ppgData = (data.ppg0 + data.ppg1 + data.ppg2) / 3.0f;

                                        Intent intent = new Intent("updatingState");
                                        intent.putExtra("ppg", String.valueOf(ppgData));
                                        sendBroadcast(intent);

                                        if (sent) {
                                            writeToCsv();
                                        }
                                    }
                                }
                            },
                            new Consumer<Throwable>() {
                                @Override
                                public void accept(Throwable throwable) throws Exception {
                                    Log.e(TAG,
                                            "" + throwable.getLocalizedMessage());
                                    ppgDisposable = null;
                                }
                            },
                            new Action() {
                                @Override
                                public void run() throws Exception {
                                    Log.d(TAG, "complete");
                                }
                            }
                    );
        } else {
            // NOTE stops streaming if it is "running"
            ppgDisposable.dispose();
            ppgDisposable = null;
        }
    }

writeToCsv() was used to save the data, it takes the most updated value and store in csv, but from what i can see, from the data (PPG, PPI and ACC) I recorded, PPG signal seems to sample at 120Hz, however, PPI and ACC data all sample at about 1.5 Hz, is this normal?

or how to make the ACC and PPI sample in a higher frequency.  and for PolarAccelerometerData etc., why a timestamp is used for a list of samples, how do we get the exact timestamp for the samples?

jams032, 2020-10-25:
@SijieZhuo Hi, as you are working recently, could you please make idea regarding this ? Check my last line of the comment. 

https://github.com/polarofficial/polar-ble-sdk/issues/103

erkki-silvola, 2020-10-31:
Hi,  the timestamp is the last sample timestamp. One needs to buffer 2 PolarAccelerometerData "samples" and calculate the timestamps for each acc sample...  something like this

` api.requestAccSettings(s).observeOn(AndroidSchedulers.mainThread()).toFlowable().flatMap((Function<PolarSensorSetting, Flowable<Map<PolarSensorSetting.SettingType, Integer>>>) sensorSetting -> {
            Log.d(TAG,"acc max available: " + sensorSetting.maxSettings().settings);
            return Flowable.just(maxSettingsSingle(sensorSetting));
        }).observeOn(AndroidSchedulers.mainThread()).flatMap((Function<Map<PolarSensorSetting.SettingType, Integer>, Publisher<PolarAccelerometerData>>) selected -> {
            String name = Objects.requireNonNull(sensStringTextViewMap.get(s)).name;
            Objects.requireNonNull(dataCollectorMap.get(s)).startAccStream(name);
            return api.startAccStreaming(s,new PolarSensorSetting(selected));
        }).buffer(2,1).observeOn(AndroidSchedulers.mainThread()).map(new Function<List<PolarAccelerometerData>, List<Pair<Long, PolarAccelerometerData.PolarAccelerometerSample>>>() {
            @Override
            public List<Pair<Long, PolarAccelerometerData.PolarAccelerometerSample>> apply(List<PolarAccelerometerData> datas) throws Throwable {
                long delta1 = ((datas.get(1).timeStamp - datas.get(0).timeStamp) / datas.get(1).samples.size());
                long baseTimeStamp1 = datas.get(0).timeStamp - ((datas.get(0).samples.size()-1)*delta1);
                List<Pair<Long, PolarAccelerometerData.PolarAccelerometerSample>> list = new ArrayList<>();
                for(PolarAccelerometerData.PolarAccelerometerSample sample : datas.get(0).samples){
                        list.add(new Pair<>(baseTimeStamp1,sample));
                        baseTimeStamp1 += delta1;
                    }
                return list;
            }}).observeOn(AndroidSchedulers.mainThread()).subscribe(
                pairs -> {//
                    // do something with the data
                },
                throwable -> Log.e(TAG, "ACC stream failed: " + throwable.getLocalizedMessage()),
                () -> {

                }
        );
`

closed 2020-11-18 by SijieZhuo