polarofficial/polar-ble-sdk #105: OH1 sensor seems to disconnect after a while

SijieZhuo, 2020-10-22:
Hi,
I have been test OH1 for a while, and i am keep having this issue. after streaming data to android app for some time, it would disconnect itself and turn off the service. is it the problem of the device or just my code?

i am using foreground service for the recording, and i am saving the streamed data on the machine. when it stops working, the following message shows up.

2020-10-23 10:56:57.044 1985-4604/com.szhu842.hr_ppg_monitor D/BluetoothGatt: onClientConnectionState() - status=14 clientIf=6 device=A0:9E:1A:71:2A:F2
2020-10-23 10:56:57.072 1985-1985/com.szhu842.hr_ppg_monitor D/BluetoothAdapter: isLeEnabled(): ON
2020-10-23 10:56:57.073 1985-1985/com.szhu842.hr_ppg_monitor E/CheckPermission: _bluetooth code = 10 
2020-10-23 10:56:57.085 1985-4604/com.szhu842.hr_ppg_monitor D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=5 mScannerId=0
2020-10-23 10:56:57.120 1985-1985/com.szhu842.hr_ppg_monitor D/DataService: Device disconnected polar.com.sdk.api.model.PolarDeviceInfo@c6e418c
2020-10-23 10:56:57.143 1985-1985/com.szhu842.hr_ppg_monitor D/BluetoothGatt: close()
2020-10-23 10:56:57.143 1985-1985/com.szhu842.hr_ppg_monitor D/BluetoothGatt: unregisterApp() - mClientIf=6
2020-10-23 10:56:57.148 1985-1985/com.szhu842.hr_ppg_monitor E/DataService: null
2020-10-23 10:56:57.148 1985-1985/com.szhu842.hr_ppg_monitor I/chatty: uid=10424(com.szhu842.hr_ppg_monitor) identical 1 line
2020-10-23 10:56:57.148 1985-1985/com.szhu842.hr_ppg_monitor E/DataService: null

BluetoothGatt status 14 seems to be "unlikely error", and due to this issue, the device is disconnected and cause the service to shutdown. can you help me with this?

SijieZhuo, 2020-10-22:
ps. the service can run for 20 mins or up to a few hours before it stop working

JOikarinen, 2020-11-24:
@SijieZhuo latest [firmware update](https://support.polar.com/en/updates/polar-oh1-firrmware-update-2120?product=OH1) to OH1 shall fix the problem.

JOikarinen, 2020-11-29:
@SijieZhuo please reopen the issue if experiencing the issue still with OH1 firmware [2.1.20](https://support.polar.com/en/updates/polar-oh1-firrmware-update-2120?product=OH1)

closed 2020-11-29 by JOikarinen