polarofficial/polar-ble-sdk #106: Xcode 12.1 build issue - Module compiled with Swift 5.2 cannot be imported by the Swift 5.3 compiler

jams032, 2020-10-25:
SDK is working/building successfully in xcode 11.5 but not in xcode 12.1. It shows error like : 

Module compiled with Swift 5.2 cannot be imported by the Swift 5.3 compiler: /Users/-----/polarBleSdkTestApp-ios/polar-sdk-ios/PolarBleSdk.framework/Modules/PolarBleSdk.swiftmodule/arm64-apple-ios.swiftmodule

Hope the attachemnt help you to fix.

Thanks in advance. 

[image]

[image]

erkki-silvola, 2020-10-31:
Hi, sdk will be updated later, meanwhile compile from sources

jet-lee-m, 2020-11-02:
Hi, I tried to compile from sources, but it is not possible. I used the build script from the repository (/sources/iOS/ios-communications)

I get the following error:

```
error: failed to build module 'RxSwift' from its module interface; the compiler that produced it, 'Apple Swift version 5.2 (swiftlang-1103.0.32.1 clang-1103.0.32.29)', may have used features that aren't supported by this compiler, 'Apple Swift version 5.3 (swiftlang-1200.0.29.2 clang-1200.0.30.1)'
import RxSwif
```

josephlarralde, 2020-11-03:
Please see #97 for a workaround (I had the same exact errors and now everything builds fine)

closed 2020-11-23 by JOikarinen