polarofficial/polar-ble-sdk #108: BLE SDK IOS v2.2.2 resulted in IPA processing failed and Code signing "PolarBleSdk.framework" failed.

ymatt345, 2020-11-25:
The lates BleSdk IOS v.2.2.2 failed to make Adhoc IPA file and AppStore Updoad IPA file.

I have to tried to make Adhoc IPA file using latest IOS BleSdk but the massage "IPA processing failed" was observed and failed to make IPA file.

IDEDistribution.standard.log was as follows.
2020-11-25 06:24:22 +0000  /Applications/Xcode.app/Contents/Developer/usr/bin/ipatool exited with 1
2020-11-25 06:24:22 +0000  ipatool JSON: {
    alerts =     (
                {
            code = 3736;
            description = "Configuration issue: platform AppleTVSimulator.platform doesn't have any non-simulator SDKs; ignoring it";
            info =             {
            };
            level = WARN;
        },
                {
            code = 3736;
            description = "Configuration issue: platform WatchSimulator.platform doesn't have any non-simulator SDKs; ignoring it";
            info =             {
            };
            level = WARN;
        },
                {
            code = 3736;
            description = "Configuration issue: platform iPhoneSimulator.platform doesn't have any non-simulator SDKs; ignoring it";
            info =             {
            };
            level = WARN;
        }
    );
}

I also tried to upLoad my App to Appstore, the message "Code signing "PolarBleSdk.framework" failed." was observed.

IDEDistributionPipeline.log was as follows.
2020-11-25 06:16:30 +0000  Running /usr/bin/codesign '-vvv' '--force' '--sign' 'BBBE0821F1C2106883563D26C605DA989EE0E16A' '--entitlements' '/var/folders/n8/b4qnsw0d38s464m03dvzfmx00000gp/T/XcodeDistPipeline.~~~bK4BYX/entitlements~~~0DXIz0' '--preserve-metadata=identifier,flags,runtime' '/var/folders/n8/b4qnsw0d38s464m03dvzfmx00000gp/T/XcodeDistPipeline.~~~bK4BYX/Root/Payload/VitalScope.app/Frameworks/PolarBleSdk.framework'
2020-11-25 06:16:32 +0000  /var/folders/n8/b4qnsw0d38s464m03dvzfmx00000gp/T/XcodeDistPipeline.~~~bK4BYX/Root/Payload/VitalScope.app/Frameworks/PolarBleSdk.framework: replacing existing signature
2020-11-25 06:16:32 +0000  /var/folders/n8/b4qnsw0d38s464m03dvzfmx00000gp/T/XcodeDistPipeline.~~~bK4BYX/Root/Payload/VitalScope.app/Frameworks/PolarBleSdk.framework: code object is not signed at all
2020-11-25 06:16:32 +0000  /usr/bin/codesign exited with 1

Please make  adhoc IPA file and advise me a solution.
Best regards,

JOikarinen, 2020-11-29:
@ymatt345 could you test  whether the same problem still happens with [v.2.2.3-beta0](https://github.com/polarofficial/polar-ble-sdk/releases/tag/v.2.2.3-beta0)

ymatt345, 2020-11-30:
Hi,
v.2.2.3-beta0 still failed to make adHoc IPA file.
My app always failed to make Adhoc IPA file using v.2.2.3-beat0.

You can duplicate the problem using example-ios app.
You may not face the problem when making example-ios Adhoc IPA file  for the first time after rebooting mac.
However you can observe this problem when making example-ios Adhoc IPA file for the second time and later.

I have been suspending our development works  for more than 2 months due this swift5.3 issue.
I am waiting for your prompt response.
Best regards.

ymatt345, 2021-01-06:
Hi,

After Installing additional component to Xcode 12.3 , the problem symptom was changed as follows.

Build error 
/Users/ymatt/Dropbox/Polar/polar-ble-sdk-v.2.2.3-beta0/examples/example-iosV2/polarBleSdkTestApp.xcodeproj Building for iOS, but the linked and embedded framework 'PolarBleSdk.framework' was built for iOS + iOS Simulator.

I googled this symptom and applied the RunScript in the following messages, them the problem of making adhoc IPA was solved.
https://developer.apple.com/forums/thread/669411

I hope you to follow the apples suggestion listed bellow.
"The only correct way to resolve this is to rebuild the framework as an XCFramework."
Thank you,
Yoshimasa

closed 2021-01-13 by JOikarinen

JOikarinen, 2021-01-13:
@ymatt345 Polar BLE SDK is now shared as XCFramework and along with that the Xcode 12 and Swift 5 issues are solved. Originally you faced the problem with the IPA file building and code signing. However, I am not been able to reproduce such problems locally so I may need more detailed steps from you still. Please don't hesitate to open new issue or reopen this one if problems still exists.