polarofficial/polar-ble-sdk #120: Cannot upload to Apple App Store because of too many RxSwift Bundles

poml88, 2021-01-17:
Hi,

I just updated my app from xcode 11.7 and the Polar Ble Sdk v2.2.2 to Xcode 12.3 and Polar Ble SDK v.2.3.0.

The app works fine, but when uploading to the App store I get the errors below. It seems that the RxSwift is included also in the PolarBleSdk, and so there is some problem. I do not know how to solve this.

I downloaded both xcframework releases and added them to my project.

Any idea?

App Store Connect Operation Error
ERROR ITMS-90685: "CFBundleIdentifier Collision. There is more than one bundle with the CFBundleIdentifier value 'RxSwift' under the iOS application 'nameofmyapp.app'."

App Store Connect Operation Error
ERROR ITMS-90205: "Invalid Bundle. The bundle at 'nameofmyapp.app/Frameworks/PolarBleSdk.framework' contains disallowed nested bundles."

App Store Connect Operation Error
ERROR ITMS-90206: "Invalid Bundle. The bundle at 'nameofmyapp.app/Frameworks/PolarBleSdk.framework' contains disallowed file 'Frameworks'."

App Store Connect Operation Warning
WARNING ITMS-90806: "CFBundleIdentifier collision. Each bundle must have a unique bundle identifier. The bundle identifier 'RxSwift' is used in the bundles '[Payload/nameofmyapp.app/Frameworks/PolarBleSdk.framework/Frameworks/RxSwift.framework, Payload/nameofmyapp.app/Frameworks/RxSwift.framework]'"

poml88, 2021-01-18:
I now tried to delete ALL "Framwork" directories from within the PolarBleSdk.xcframework folder (and emtpy the bin) and now it uploaded.

Is that something obvious I should have known?

JOikarinen, 2021-01-18:
Hi @poml88,  interesting finding. Let's try to figure out what's going on. 

> I just updated my app from xcode 11.7 and the Polar Ble Sdk v2.2.2 to Xcode 12.3 and Polar Ble SDK v.2.3.0.

Just to double check, I would like to know you followed these steps when upgrading your app from PolarBleSDK v2.2.2 to v2.3.0

1. Reference to PolarBleSDK v2.2.2 framework was completely removed from your app: 
-- Xcode -> your app target(s) -> Build phases -> Link binary with libraries is not referencing to PolarBleSDK
-- Xcode -> your app target(s) -> Build settings -> search paths are not referencing to PolarBleSDK

2. Adding PolarBleSdk v.2.3.0 (as xcframework) into your app 
- adding XCFramework see https://developer.apple.com/videos/play/wwdc2019/416/ at around time 3:50

> It seems that the RxSwift is included also in the PolarBleSdk, and so there is some problem.

Yes, PolarBleSdk depends on RxSwift, so RxSwift is included in PolarBleSdk Xcode project.  Your app must also depend on RxSwift if want to use PolarBleSdk. 

> I downloaded both xcframework releases and added them to my project.

- You downloaded PolarBleSDK XCFramework here: https://github.com/polarofficial/polar-ble-sdk/releases/tag/2.3.0
- You downloaded RxSwift XCFramework here: https://github.com/ReactiveX/RxSwift/releases/tag/6.0.0

> Any idea?
- to troubleshoot the problem could you step through the section **Inspecting A Binary's Linkage** in the [embedding framework documentation](https://developer.apple.com/library/archive/technotes/tn2435/_index.html#//apple_ref/doc/uid/DTS40017543-CH1-TROUBLESHOOTING-BUNDLE_ERRORS) for your app. You shall see something like this:
`	@rpath/RxSwift.framework/RxSwift (compatibility version 0.0.0, current version 0.0.0)
	@rpath/PolarBleSdk.framework/PolarBleSdk (compatibility version 1.0.0, current version 1.0.0)
`

> I now tried to delete ALL "Framwork" directories from within the PolarBleSdk.xcframework folder (and emtpy the bin) and now it uploaded.
- this should not be needed. I wonder, how your app can compile if deleting the content of  PolarBleSdk.xcframework/

poml88, 2021-01-18:
Hi @JOikarinen, and thanks for the quick response!

I checked all the steps and I think I did it correctly. However, I can repeat them later today.

I think the problem is, that inside the PolarBleSdk.xcframework folder there is a folder called Frameworks and in there there is RxSwift. Apple does not like / allow this.

In fact then **I really have two RxSwift in my Project**:
1) The one I downloaded and added to my project
2) The one that is sitting inside the PolarBleSdk.xcframework.

That explains the errors I get:
```
App Store Connect Operation Error
ERROR ITMS-90685: "CFBundleIdentifier Collision. There is more than one bundle with the CFBundleIdentifier value 'RxSwift' under the iOS application 'nameofmyapp.app'."

App Store Connect Operation Error
ERROR ITMS-90205: "Invalid Bundle. The bundle at 'nameofmyapp.app/Frameworks/PolarBleSdk.framework' contains disallowed nested bundles."

App Store Connect Operation Error
ERROR ITMS-90206: "Invalid Bundle. The bundle at 'nameofmyapp.app/Frameworks/PolarBleSdk.framework' contains disallowed file 'Frameworks'."
```

Isn't it like this?

I will gladly do more tests. It is a bit slow, because the errors show only when the app was uploaded and checked by the appstore system.

hm1205, 2021-01-18:
Hi, 

I'm encountering the same errors; I've followed the key steps well and I'm handling other xcframeworks in other apps without any worries. 

best regards

JOikarinen, 2021-01-20:
Issue fixed in release https://github.com/polarofficial/polar-ble-sdk/releases/tag/2.3.1. The problem was that RxSwift framework was embedded to Polar BLE SDK XCFramework. As a consequence client application has two instance of RxSwift bundled which is not accepted by App Store Connect.

closed 2021-01-20 by JOikarinen

poml88, 2021-01-20:
> 
> 
> Issue fixed in release https://github.com/polarofficial/polar-ble-sdk/releases/tag/2.3.1. The problem was that RxSwift framework was embedded to Polar BLE SDK XCFramework. As a consequence client application has two instance of RxSwift bundled which is not accepted by App Store Connect.

Thank you @JOikarinen. Exactly that was the issue.