polarofficial/polar-ble-sdk #122: Question on updating firmware of H10

poml88, 2021-01-29:
Apparantly there is a new firmware 3.1.1 for the H10.

1) Any changes for the SDK? Especially the ECG part?
2) It is not shown to me in the Polar Beat App. Even though I am logged in as a use. What am I doing wrong?

JOikarinen, 2021-02-02:
> 1. Any changes for the SDK? Especially the ECG part?
- in [3.1.1](https://support.polar.com/en/updates/polar-h10-firmware-update-311) there is enhancements how H10 handles training recording. No changes related to ECG. 

> 2. It is not shown to me in the Polar Beat App. Even though I am logged in as a use. What am I doing wrong?
- could you double check you followed these instructions: https://support.polar.com/en/support/how_can_i_update_polar_h10

closed 2021-02-02 by JOikarinen

poml88, 2021-02-26:
Hi, yes, I followed those steps, still there is no update offerend fpr my H10. No idea why. Maybe I have to ask Polar Support.

JOikarinen, 2021-03-01:
@poml88 did you try with Polar Beat or with Polar Flow app?

poml88, 2021-03-01:
@JOikarinen I did try the Polar Beat App. I can try with the Polar Flow App as well.

JOikarinen, 2021-03-01:
@poml88 to get the firmware update notification you shall log in as registered user in Beat app. I am not sure is that clearly mentioned in the instructions here: https://support.polar.com/en/support/how_can_i_update_polar_h10

poml88, 2021-03-01:
@JOikarinen Yes, I am logged in as a user. Still, it does not offer me a firmware update. I can try with Polar Flow.

poml88, 2021-03-12:
@JOikarinen I tried and the Polar Flow app is offering the update. I did not install it yet, but it seems to work with Polar Flow.