polarofficial/polar-ble-sdk #123: real time processing

alyfreym, 2021-02-01:
Hello! In iOS you can create listener in to HealthApi and is called each time when something was changed, it returns a structure that stores data - calories, distance in meters, heart rate and so on. Does this SDK have this functionality? Or the developer must do all the calculations himself? If Yes, with respect to what data everything can be calculated? . For example if sdk give me distance  and  speed, i can count steps and calories. In SDK docs I see some data called - PolarEcgData, PolarPpgData etc..  Can I count in relation to them calories, distance?  There is a more detailed explanation of what these structures are. ? PolarPpgData, PolarPpiData etc.. 

I would be happy with a small example, let's say how to calculate the distance that the user has passed If possible.

I am not a user Polar watch. Therefore, you may think that I am asking stupid questions. I hope for your understanding 

Thank you for your attention.

JOikarinen, 2021-02-02:
@alyfreym good questions. Lets start discussion by splitting your questions to smaller pieces. 

> In iOS you can create listener in to HealthApi and is called each time when something was changed, it returns a structure that stores data - calories, distance in meters, heart rate and so on. Does this SDK have this functionality?
- Yes. Polar SDK API uses two approaches: function callbacks (i.e. listeners) and [asynchronous reactive API](http://reactivex.io/). The approach is chosen depending on the what the SDK is asked to do, for example BLE connection created with SDK uses callbacks to tell the status of connection whereas data streams requested from connected device uses reactive approach. On iOS side the Polar SDK reactive implementation is [RxSwift](https://github.com/ReactiveX/RxSwift)

> Or the developer must do all the calculations himself? If Yes, with respect to what data everything can be calculated?
- The SDK is not providing (at the moment of writing) speed or distance data. The data available and calculated by the SDK is explained here https://www.polar.com/en/developers/sdk. So whether the application using SDK needs to process or calculate the data even further depends on what that particular application wants to do with the data. 

> In SDK docs I see some data called - PolarEcgData, PolarPpgData etc.. Can I count in relation to them calories, distance? There is a more detailed explanation of what these structures are. ? PolarPpgData, PolarPpiData etc..
- [PolarEcgData](https://github.com/polarofficial/polar-ble-sdk/blob/ed4aab70603409e7694ce9d52e1d9dfb755626a2/sources/iOS/ios-communications/iOSCommunications/sdk/api/PolarBleApi.swift#L55): Electrocardiography (ECG) data in µV. https://en.wikipedia.org/wiki/Electrocardiography
- [PolarPpgData](https://github.com/polarofficial/polar-ble-sdk/blob/ed4aab70603409e7694ce9d52e1d9dfb755626a2/sources/iOS/ios-communications/iOSCommunications/sdk/api/PolarBleApi.swift#L67): https://en.wikipedia.org/wiki/Photoplethysmogram
- [PolarPpiData](https://github.com/polarofficial/polar-ble-sdk/blob/ed4aab70603409e7694ce9d52e1d9dfb755626a2/sources/iOS/ios-communications/iOSCommunications/sdk/api/PolarBleApi.swift#L77): peak-to-peak interval between heartbeats in milliseconds

> I would be happy with a small example, let's say how to calculate the distance that the user has passed If possible.
- The [iOS SDK example](https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-ios) is good starting point, which shows the usage available API's.

JOikarinen, 2021-02-05:
@alyfreym closing the issue. Please feel free to continue discussion and reopen the issue.

closed 2021-02-05 by JOikarinen