polarofficial/polar-ble-sdk #124: Changing accelerometer settings on H10

ossovskiy, 2021-02-02:
Hello everybody! I am very new in android programming, trying to change settings on H10 accelerometer and run into a problem. When I start Acc streaming with default settings (maxSettings), everything works fine, and Logcat outputs accelerometer data (first screenshot), but once I try to change sampling rate or the value of G, Logcat outputs nothing, so apparently H10 doesn't start at all (second screenshot). Could you please guys point out what do I do wrong? I include two screenshots.

[image]
[image]

JOikarinen, 2021-02-03:
Hi @ossovskiy! 
> but once I try to change sampling rate or the value of G,
- The problem is that while customising the settings the RESOLUTION setting is missing. H10 is sensitive if not all available settings are written. Acc stream fails and you may got error like this: `ACC stream failed: java.lang.Exception: Pmd cp failed: ERROR_INVALID_PARAMETER `

- here is the code which shall work:
```
 .flatMap((Function<PolarSensorSetting, Publisher<PolarAccelerometerData>>) settings -> {
    final Map<PolarSensorSetting.SettingType, Integer> sensorSettings = new HashMap<>();
    sensorSettings.put(PolarSensorSetting.SettingType.SAMPLE_RATE, 25); // Possible values for H10: 25, 50, 100, 200 Hz
    sensorSettings.put(PolarSensorSetting.SettingType.RANGE, 4); // Possible values for H10: 2 ,4, 8 G
    sensorSettings.put(PolarSensorSetting.SettingType.RESOLUTION, 16); // Possible values for H10: 16 bits
    return api.startAccStreaming(DEVICE_ID, new PolarSensorSetting(sensorSettings));
 })
 ```

closed 2021-02-03 by JOikarinen

ossovskiy, 2021-02-03:
Works! Works!!! Thank you Jukka!!!