polarofficial/polar-ble-sdk #125: Connect polar device to Android without Polar App

igorsan98v2, 2021-02-04:
Hello, I am trying to connect H10 belt to android device, by ID connection in my own app, but it`s fails at each android device in case if I did not  connected to belt before by Polar Beat App. Why So ? How to connect  polar devices without install Polar apps ?

JOikarinen, 2021-02-05:
HI @igorsan98v2, 
there is no need to install Polar Beat or Polar Flow to make H10 connectable. So lets investigate your case. You may try with the sample app found here: https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-android/androidBleSdkTestApp

In [MainActivity](https://github.com/polarofficial/polar-ble-sdk/blob/ed4aab70603409e7694ce9d52e1d9dfb755626a2/examples/example-android/androidBleSdkTestApp/app/src/main/java/polar/com/androidblesdk/MainActivity.java#L45) change the deviceId the one printed in your H10:

```
String DEVICE_ID = "12345FF"; // TODO replace with your device id
```

After installing the example app on your phone press the `connect` in the app. You may monitor whether any connection events happens in callbacks:

````
            @Override
            public void deviceConnected(@NonNull PolarDeviceInfo polarDeviceInfo) {
                Log.d(TAG, "CONNECTED: " + polarDeviceInfo.deviceId);
                DEVICE_ID = polarDeviceInfo.deviceId;
            }

            @Override
            public void deviceConnecting(@NonNull PolarDeviceInfo polarDeviceInfo) {
                Log.d(TAG, "CONNECTING: " + polarDeviceInfo.deviceId);
                DEVICE_ID = polarDeviceInfo.deviceId;
            }

            @Override
            public void deviceDisconnected(@NonNull PolarDeviceInfo polarDeviceInfo) {
                Log.d(TAG, "DISCONNECTED: " + polarDeviceInfo.deviceId);
                ecgDisposable = null;
                accDisposable = null;
                ppgDisposable = null;
                ppiDisposable = null;
            }
```

igorsan98v2, 2021-02-05:
> there is no need to install Polar Beat or Polar Flow to make H10 connectable. So lets investigate your case. You may try with the sample app found here: https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-android/androidBleSdkTestApp
Hello, @JOikarinen , thanks for advice, but in my case,  as I supose, the root of my problem was not turned on location detection :). After I turn on it. My own s app starts work great without any installation of polar beat. 👍

closed 2021-02-08 by JOikarinen