polarofficial/polar-ble-sdk #127: Do the new Polar Verity Sense support the same protocol as OH1?

erikboto, 2021-02-10:
Is the Verity Sense supported in the same way as the Polar OH1?

JOikarinen, 2021-02-11:
@erikboto yes it is. There is, however, minor enhancements made to API which causes that client using Polar SDK needs to be updated slightly. 

- individual stream setting requests replaced with more generic setting request function, i.e.``` requestStreamSettings()``` replaces call like ```requestAccSettings()```
- callback indicating the stream feature is ready is now replaced with more generic function, i.e. ```streamingFeaturesReady()``` replaces callback like ```accFeatureReady```

erikboto, 2021-02-16:
Thanks!

Is this due to changes in the underlying BLE protocol? I'm not actually using the SDK, but the information in Polar_Measurement_Data_Specification.pdf which allowed us to log data from from a regular linux system. So my main question is really if this information is still accurate, and if it isn't, can we expect the technical information to be update as well?

JOikarinen, 2021-02-23:
@erikboto there is updates in underlying protocol. Updates were required because of the bandwidth limits of the BLE. [Technical documentation](https://github.com/polarofficial/polar-ble-sdk/blob/master/technical_documentation/Polar_Measurement_Data_Specification.pdf) is now updated.

closed 2021-02-23 by JOikarinen

jussivirkkala, 2021-03-04:
Does this delta frame encoding mean that fast changes are limited to [-128 127]?

JOikarinen, 2021-03-08:
@jussivirkkala I suppose, your question is that whether delta packaging is limiting (big) changes between consecutive samples? That is not the case, as new encoded delta frame can be started with new reference value anytime. Also delta size can be modified on fly.

jussivirkkala, 2021-03-08:
Clever design. So delta samples can be not only 8 bit but also 16 bit or even 32 bits? I have to look raw data how this changes in practise.

JOikarinen, 2021-03-09:
> So delta samples can be not only 8 bit but also 16 bit or even 32 bits? 
- Delta frame encoding adapts dynamically to signal changes. Delta range is from 3bit to 32bit. One packet of data can contain multiple delta frames with different delta values.

> I have to look raw data how this changes in practise.
- 👍  observing the raw data shows you how it behaves in detailed