polarofficial/polar-ble-sdk #129: Can not establish connection without restarting the sensor

emrahozer-rightsoft, 2021-02-18:
I'm using Android SDK to do an integration on Quest 2. I can connect and receive heart data if I call autoconnect and IF I turn on the sensor after I launch the app. After connection I store the device id and use that for calling disconnect method. 

When I quit the app, I call disconnect and then shutdown. (Tried many combinations) 

The problem I'm facing is if I do not restart the sensor, when I **relaunch** the app it fails to connect back. Calling the connect with device id, or autoconnect doesn't seem to make any difference. 

I wanted to check with you if there's something do to make sure the connection is closed correctly?

`public class PolarDeviceBridge {
    private static final String LOG_TAG = "PolarDeviceBridge";
    private Context context;
    private PolarBleApi api;
    private boolean connected = false;
    private String deviceId = "";
    private int heartRate;
    private Disposable autoConnectDisposable;

    public PolarDeviceBridge(Context context) {
        this.context = context;
    }

    public void Init() {
        Log.d(LOG_TAG, "initializing");
        InitApi();
        ConnectToDevice();
    }

    public boolean IsConnected() {
        return connected;
    }

    public int GetHeartRate() {
        return heartRate;
    }

    public void Pause() {
        Log.d(LOG_TAG, "Pausing...");

        if (api != null) {
            Log.d(LOG_TAG, "Pause api not null");

            heartRate = 0;
            Log.d(LOG_TAG, "Api background entered");

            api.backgroundEntered();

            Log.d(LOG_TAG, "Dispose called");

            Dispose();
        }
    }

    public void Resume() {
        Log.d(LOG_TAG, "Resuming...");
        if (api != null) {

            Log.d(LOG_TAG, "Foreground entered");
            api.foregroundEntered();
        } else {
            Log.d(LOG_TAG, "Init Api Called");

            InitApi();
            Log.d(LOG_TAG, "Connect To Device");

            ConnectToDevice();
        }

    }

    public void Dispose() {

        Log.d(LOG_TAG, "Dispose is called." + (api == null));
        if (api != null) {
            Log.d(LOG_TAG, "Dispose Api is not null");

            DisconnectFromDevice();

            Log.d(LOG_TAG, "Shutting down api.");
            api.shutDown();
            Log.d(LOG_TAG, "Api shut down completed");
            api = null;
        }
    }

    private void InitApi() {
        Log.d(LOG_TAG, "Initialising Polar Api...");
        api = PolarBleApiDefaultImpl.defaultImplementation(context, PolarBleApi.FEATURE_HR);
        api.setAutomaticReconnection(true);
        api.setApiCallback(new PolarBleApiCallback() {
            @Override
            public void blePowerStateChanged(boolean powered) {
                Log.d(LOG_TAG, "BLE power: " + powered);
            }

            @Override
            public void deviceConnected(PolarDeviceInfo polarDeviceInfo) {
                Log.d(LOG_TAG, "Connected to device " + deviceId);
                deviceId = polarDeviceInfo.deviceId;
                connected = true;
            }

            @Override
            public void deviceConnecting(PolarDeviceInfo polarDeviceInfo) {
                Log.d(LOG_TAG, "Connecting to device " + polarDeviceInfo.deviceId);
            }

            @Override
            public void deviceDisconnected(PolarDeviceInfo polarDeviceInfo) {
                Log.d(LOG_TAG, "Disconnected from device " + polarDeviceInfo.deviceId);
                connected = false;
            }

            @Override
            public void hrFeatureReady(String identifier) {
                Log.d(LOG_TAG, "Heart rate ready: " + identifier);
            }

            @Override
            public void hrNotificationReceived(String identifier, PolarHrData data) {
                Log.d(LOG_TAG, "Heart rate received: " + data.hr);
                heartRate = data.hr;
            }
        });
        connected = false;
        Log.d(LOG_TAG, "Initialised Polar Api.");
    }

    public void ConnectToDevice() {
        if ((api != null) && (connected == false)) {
            if (deviceId == "") {
                if (autoConnectDisposable != null) {
                    autoConnectDisposable.dispose();
                    autoConnectDisposable = null;
                }
                
                Log.d(LOG_TAG, "Trying to connect to a nearby device...");
                autoConnectDisposable = api.autoConnectToDevice(-50, null, null).subscribe(() -> Log.d(LOG_TAG, "auto connect search complete"),
                        throwable -> Log.e(LOG_TAG, "" + throwable.toString()));
            } else {
                try {
                    Log.d(LOG_TAG, "Trying to connect to device " + deviceId);
                    api.connectToDevice(deviceId);
                } catch (PolarInvalidArgument polarInvalidArgument) {
                    Log.e(LOG_TAG, "Can't connect. Invalid device id: " + deviceId);
                }
            }
        }
    }

    private void DisconnectFromDevice() {
        Log.d(LOG_TAG, "Disconnect from device");

        if ((api != null)) {
            try {
                Log.d(LOG_TAG, "Disconnecting from device " + deviceId);
                api.disconnectFromDevice(deviceId);
            } catch (PolarInvalidArgument polarInvalidArgument) {
                Log.e(LOG_TAG, "Can't disconnect. Invalid device id: " + deviceId);
            }
        }
    }
}`

JOikarinen, 2021-02-18:
@emrahozer-rightsoft did I understood the steps correctly?

Step 1. Turn on the sensor
Step 2. Launch the app, which is causing the call to  `PolarDeviceBridge.Init()` 
...
Step 3. Quit the app, which is causing the call to `PolarDeviceBridge.Dispose()` 
Step 4. Relaunch the app, which is causing the call to `PolarDeviceBridge.Resume()`

Problem happens in step 4 and connection never happens in resume?

I copied your gist to my test project and I  tried the steps above. I couldn't reproduce the problem, everything worked as you assumed the SDK to work. Is it possible to share relevant parts your app logs?

emrahozer-rightsoft, 2021-02-19:
@JOikarinen I've created another gist and simplified the code for your reproduction: 

https://gist.github.com/emrahozer-rightsoft/79aeffa516d6ff6f6b6ce25fc5c44edc

I can also send the apk but you would need to run it on a Quest device.

Reproduction steps: 

- Turn off the device
- Turn on the device
- call **InitApi()**
- call **ConnectToDevice()**
 ( observe the connection is established and hr data is received from the callback)
 - call **Dispose()**
 - call **InitApi()**
 - call **ConnectToDevice()**
 (observe that connection is not established and hr data can't be received, at this stage you can try as many times to dispose the api / init and connect but it never connects back) 

 - Turn off the device
 - Turn on the device
 (observe that connection is established back again and hr data is received from the callback)

JOikarinen, 2021-02-22:
Good gist again, it is easy test your code.👍 

I don't have Quest 2 device for testing, but I used phone with Android 10 to match your environment. I tested your test code with Polar H10 (firmware 3.1.1) and still I am not able to see same problems as you describe. In second `ConnectToDevice()` call same H10 sensor found in first `ConnectToDevice()` call is connected and heart rate is received without a problem. 

Some questions and thoughts:
- Is the heart rate sensor your are connecting Polar H10 or some other Polar heart rate sensor? What is the firmware version of that heart rate sensor? You may read the firmware version by using feature `FEATURE_DEVICE_INFO` and listening `disInformationReceived()`
- do have possibility to try and test your code on the android phone to see whether you can reproduce the problem in phone as well?
- in your gist you use `autoConnectToDevice` api call. That is totally fine, however, there is no guarantee that in the second  `autoConnectToDevice` call you will get connected to same device than in first call. Giving the RSSI limit and the heart rate service UUID as parameter, like you have done in the gist, of course increases possibility you get connected to the same sensor again. Another approach would be , that you first search the devices with the `searchForDevice` call and then pick the device of interest from returned list. Once you pick the device of the interest you may use api calls `connectToDevice` and `disconnectFromDevice`.

emrahozer-rightsoft, 2021-02-22:
@JOikarinen thanks a lot for your response. 

I have two sensors for testing. OH1+ ( firmware 2.1.20) and Polar Verity Sense (firmware 1.0.0) . I've checked these values from Polar Beat app. I can not see any firmware updates on Polar Beat app for both sensors. 

Unfortunately I can not test the code with an android phone currently. I'll try to obtain one.

I tried to connect / disconnect with passing the ids as well in my previous attempts. I'll retry this today and let you know.

emrahozer-rightsoft, 2021-02-23:
[image]

@JOikarinen unfortunately connect with id / disconnect with id also doesn't solve the issue. It just works first time and after you disconnect you need to restart the device to be able to connect. 

Did you have chance to test with the OH1+ and Polar Verity Sense devices?

JOikarinen, 2021-02-24:
@emrahozer-rightsoft I tried with Polar H10 and Polar Verity Sense, both worked for me. Btw, you have good logs above from your running code. Could you set the SDK logging on with the `setApiLogger`, for example: `api.setApiLogger(s -> Log.d("API_LOGGER", s));`. That may give us some information what is going on inside the SDK when the problem occurs.

emrahozer-rightsoft, 2021-02-24:
Hi @JOikarinen, I purchased an android phone and tested the bug. I couldn't reproduce it, so it should be related to Quest somehow. Thanks for your effort.  I'll use broadcast feature instead of trying to establishing a direct connection to the device which works as expected.

JOikarinen, 2021-02-25:
So it seems Quest android platform is modified on some parts compared to phone android. If you are able to detect the problem on Quest, while doing you development, please let us know about your findings. It would be interesting to know what is the exactly the problem in Quest. If you have possibility take the SDK logs, I suggested in previous comment, that would still be pretty interesting to see too.

JOikarinen, 2021-02-25:
@emrahozer-rightsoft I will close this issue for now, but lets continue the discussion. And lets reopen the issue, if we come to conclusion, the bug is in SDK after all.

closed 2021-02-25 by JOikarinen