polarofficial/polar-ble-sdk #132: Add support for Swift Package Manager

kcede, 2021-03-03:
Currently, the iOS library contains mixed language source files, which prevents us from using the library as a Swift package.

Replacing the generated Objective-C protobuf files by using something like [Swift Protobuf](https://github.com/apple/swift-protobuf) would provide a nice starting point. 

Next steps would include:
 - adjusting the folder structure by moving the existing target source files to `Sources` and `Tests` folders
 - running `swift package init --type library` to create a `Package.swift`
 - adding the frameworks `RxSwift` and `RxTest` as dependencies.

I attempted this myself, but since the `.proto` schema is not public, I couldn't make any real progress.

JOikarinen, 2021-03-04:
@kcede good suggestion indeed, this have been in discussions. Some bugs related to Swift Package Manager have slightly reduced the motivation for this, for example https://bugs.swift.org/browse/SR-12303. However, let's add this item for TODO list. Unfortunately I cannot promise any timetable for this at the moment.

aegis965, 2021-03-04:
I tried adding the xcframework with the `binaryTarget` feature in swift 5.3. My [`Package.swift`](https://github.com/aegis965/polar-ble-sdk/blob/195394e7fddfcec3f6f1e80c791f5feba97c1a38/Package.swift). But I'm getting `dyld: Library not loaded`. Not sure if this is a Swift bug or not. I'll do more experiments. It would be ideal if we get SPM support at the source level.

edorphy, 2021-04-17:
What would an acceptable minimum OS version be for the library?
- As a suggestion, setting to 13/14 might be great for teams trying to use the latest Swift language features like Combine and SwiftUI (@ObservableObject)

Additionally, would you want the SP to support all four platforms? (iOS, tvOS, macOS, watchOS)?
- I don't see any technical limitations to supporting the watch and tv other than the systems limit the number of connections as well as the connection interval duration. --> So streaming data from the Polar Data Service might be choppy.

closed 2022-02-07 by JOikarinen