polarofficial/polar-ble-sdk #134: iOS SDK release 3.0.2 is compiled without bitcode

marmarelis, 2021-03-04:
After pulling the latest updates, I am confronted with the following error message from the linker:
`[...]PolarBleSdk.framework/PolarBleSdk' does not contain bitcode. You must rebuild it with bitcode enabled (Xcode setting ENABLE_BITCODE), obtain an updated library from the vendor, or disable bitcode for this target. [...]`

Do you know how to fix that?
Thanks.

JOikarinen, 2021-03-09:
The Bitcode enabled flag was accidentally changed from "default" (default is enabled) to "No" in commit 37e5f47. Fixed in 
773f229 by explicitly enabling the bitcode for PolarBleSDK target.

closed 2021-03-09 by marmarelis

JOikarinen, 2021-03-09:
Released the version [3.0.3](https://github.com/polarofficial/polar-ble-sdk/releases/tag/3.0.3) including the fix.