polarofficial/polar-ble-sdk #135: OH1 continues to send HR notifications

cedneo, 2021-03-05:
I am developing an application (Android) that uses a Polar OH1 to detect Heart rate of user. The user puts OH1 on his arm and then connect to it. It works fine, its heart rate is sending periodically to app. 
But when the user removes OH1 from his arm, without disconnecting it, OH1 continues to send periodical various heart rates ! I don't understand this behavior... is it a bug ? To work around this behaviour, is it possible to know if the sensor is on the skin ?

JOikarinen, 2021-03-08:
@cedneo please see the issue in #131. Is that answering to your question?

cedneo, 2021-03-11:
Thanks for your reply. 
To work around the behaviour on Polar OH1, what do you think about using accelerometer ? If accelerometer values change it means that the OH1 is worn by the user, not necessarily on the skin, but it is better than nothing. 
Is there any autonomy impact on OH1 if i use accelerometer ?
Thanks for your reply.

JOikarinen, 2021-03-15:
@cedneo in my opinion you suggestion would be possible approach for the problem. From technical point of view I don't see any reason why you suggestion would not work.

jimmyzumthurm, 2023-09-08:
Hi @cedneo , your suggestion might be possible to implement, however we are not supporting OH1 with new firmware updates anymore, so that could only be done with Verity Sense. In our ecosystem we prefer if OHR algorithm can determine by himself if skin contact is there or not, without having to rely on accelerometer for that, as user might be in deep sleep without without moving at all, etc.

closed 2023-09-08 by jimmyzumthurm