polarofficial/polar-ble-sdk #136: Crash under android: java.lang.NoClassDefFoundError: Failed resolution of: Lkotlin/jvm/internal/Intrinsics;

fmartin4, 2021-03-07:
How to reproduce it: downloading the sample code of this repository, building nd debugging the app in a mobile (Samsung galaxy note 10+) trying to link to a polar verity sense device. After we click on "CONNECT" buton we get this crash.

This worked "almost" fine with the previous version of the lib "polar-ble-sdk.aar" downloaded last week, at least to connect, scan, etc. But even with this previos version, the same crash happened when trying to list recordings.  Today I realized there were a new version of the library, I tried to test it, but it seems to be worse.

It seems that the sample code is enterily written in java, the not found kotlin classes seem to be referenced in the library.

2021-03-07 22:35:37.483 7174-7174/polar.com.androidblesdk I/ViewRootImpl@cb89706[MainActivity]: ViewPostIme pointer 0
2021-03-07 22:35:37.559 7174-7174/polar.com.androidblesdk I/ViewRootImpl@cb89706[MainActivity]: ViewPostIme pointer 1
2021-03-07 22:35:37.641 7174-7174/polar.com.androidblesdk D/API LOGGER: BDScanCallback/commandState state:IDLE action: CLIENT_START_SCAN
2021-03-07 22:35:37.642 7174-7174/polar.com.androidblesdk D/API LOGGER: BDScanCallback/commandState state:IDLE action: EXIT
2021-03-07 22:35:37.642 7174-7174/polar.com.androidblesdk D/API LOGGER: BDScanCallback/commandState state:SCANNING action: ENTRY
2021-03-07 22:35:37.642 7174-7174/polar.com.androidblesdk D/API LOGGER: BDScanCallback/timestamps left: 0
2021-03-07 22:35:37.643 7174-7174/polar.com.androidblesdk D/API LOGGER: BDScanCallback/Scan started -->
2021-03-07 22:35:37.644 7174-7174/polar.com.androidblesdk D/BluetoothAdapter: STATE_ON
2021-03-07 22:35:37.644 7174-7174/polar.com.androidblesdk I/chatty: uid=10338(polar.com.androidblesdk) identical 1 line
2021-03-07 22:35:37.646 7174-7174/polar.com.androidblesdk D/BluetoothAdapter: STATE_ON
2021-03-07 22:35:37.647 7174-7174/polar.com.androidblesdk D/BluetoothAdapter: BLE support array set: 111111
2021-03-07 22:35:37.647 7174-7174/polar.com.androidblesdk D/BluetoothLeScanner: Start Scan with callback
2021-03-07 22:35:37.651 7174-10994/polar.com.androidblesdk D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=7 mScannerId=0
2021-03-07 22:35:37.689 7174-7174/polar.com.androidblesdk D/API LOGGER: BDScanCallback/Scan started <--
2021-03-07 22:35:37.841 7174-7174/? D/AndroidRuntime: Shutting down VM
2021-03-07 22:35:37.846 7174-7174/? E/AndroidRuntime: FATAL EXCEPTION: main
    Process: polar.com.androidblesdk, PID: 7174
    java.lang.NoClassDefFoundError: Failed resolution of: Lkotlin/jvm/internal/Intrinsics;
        at com.androidcommunications.polar.api.ble.model.gatt.client.BleBattClient.<clinit>(BleBattClient.kt:20)
        at java.lang.reflect.Constructor.newInstance0(Native Method)
        at java.lang.reflect.Constructor.newInstance(Constructor.java:343)
        at com.androidcommunications.polar.api.ble.model.gatt.BleGattFactory.getRemoteServices(BleGattFactory.java:34)
        at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDDeviceSessionImpl.<init>(BDDeviceSessionImpl.java:80)
        at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDDeviceListenerImpl.deviceDiscovered(BDDeviceListenerImpl.java:299)
        at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDScanCallback.onScanResult(BDScanCallback.java:271)
        at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:580)
        at android.os.Handler.handleCallback(Handler.java:883)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:237)
        at android.app.ActivityThread.main(ActivityThread.java:8167)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:496)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1100)
     Caused by: java.lang.ClassNotFoundException: Didn't find class "kotlin.jvm.internal.Intrinsics" on path: DexPathList[[zip file "/data/app/polar.com.androidblesdk-T_t7uRGfblAF2gZB5cvMeg==/base.apk"],nativeLibraryDirectories=[/data/app/polar.com.androidblesdk-T_t7uRGfblAF2gZB5cvMeg==/lib/arm64, /system/lib64]]
        at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:196)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:379)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:312)
        at com.androidcommunications.polar.api.ble.model.gatt.client.BleBattClient.<clinit>(BleBattClient.kt:20) 
        at java.lang.reflect.Constructor.newInstance0(Native Method) 
        at java.lang.reflect.Constructor.newInstance(Constructor.java:343) 
        at com.androidcommunications.polar.api.ble.model.gatt.BleGattFactory.getRemoteServices(BleGattFactory.java:34) 
        at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDDeviceSessionImpl.<init>(BDDeviceSessionImpl.java:80) 
        at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDDeviceListenerImpl.deviceDiscovered(BDDeviceListenerImpl.java:299) 
        at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDScanCallback.onScanResult(BDScanCallback.java:271) 
        at android.bluetooth.le.BluetoothLeScanner$BleScanCallbackWrapper$1.run(BluetoothLeScanner.java:580) 
        at android.os.Handler.handleCallback(Handler.java:883) 
        at android.os.Handler.dispatchMessage(Handler.java:100) 
        at android.os.Looper.loop(Looper.java:237) 
        at android.app.ActivityThread.main(ActivityThread.java:8167) 
        at java.lang.reflect.Method.invoke(Native Method) 
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:496) 
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1100) 
2021-03-07 22:35:37.874 7174-7174/? I/Process: Sending signal. PID: 7174 SIG: 9

JOikarinen, 2021-03-09:
Fixed in 992015c. The reason was that Kotlin code was added to PolarBleSDK library but the example app did not have the Kotlin support.

JOikarinen, 2021-03-09:
Reopened. Polar BLE SDK has Kotlin code but it shouldn't effect on applications using [polar-ble-sdk.aar](https://github.com/polarofficial/polar-ble-sdk/blob/master/polar-sdk-android/libs/polar-ble-sdk.aar). Demo and example applications are fixed in 992015c, however the root cause on Polar Ble SDK remains.

JOikarinen, 2021-03-09:
This is my understanding about the issue at the moment. Please correct me if I am wrong with the analysis:

If the the Android library (i.e. .aar) is using Kotlin and the Android application using that library is pure Java application, then 
there will be a problem when functionality in .aar library tries to call Kotlin utilities. Kotlin utilities are not bundled with .aar and as consequence crash with error like `Didn't find class "kotlin.jvm.internal.Intrinsics"` will happen. 

The solution for the problem is that applications using Polar BLE SDK must have [Kotlin supported](https://developer.android.com/kotlin/add-kotlin).

Erhannis, 2021-05-07:
@JOikarinen That matches my experience.  I (just today) was getting the "Failed resolution" error, and after adding Kotlin support, the error went away.

Is the library _intended_ to require Kotlin support?

JOikarinen, 2021-05-10:
@Erhannis intend is not to require Kotlin support. However, Polar SDK library is now already containing some Kotlin code and all the future new implementation will happen preferably with Kotlin. Unfortunately, if the library is containing Kotlin, it seems that client application also needs to have Kotlin support. I did investigate the problem for some time, but as the Android development is very much moving towards Kotlin I decided the proposed fix (i.e. add of Kotlin support) may be sufficient

closed 2022-12-13 by JOikarinen