polarofficial/polar-ble-sdk #137: Getting data for untouched Polar devices as advertising data!

jams032, 2021-03-16:
I have implemented Polar SDK, using Polar H10 and getting HR data from advertising data(means without pairing with BLE API). Everything works fine. 

But there is a scenario. As I am not connecting/pairing through BLE, so it can advertise any polar devices. Suppose, you don't touch Polar devices, but it is near to the iPad(say it was touched 5 minutes ago, sometimes not even touched for a long time), but it shows/emits/advertises a fixed amount of HR data automatically. I find this sometimes. 

I believe it is happening for the signal power of Polar H10. But I want to discard/avoid data from devices that are not really touched. 

 Do you have any tricks to avoid data that are not really touched? I need to track the invalid data!

Please share your thought regarding this, it would be a nice help. 

Thanks in advance!

JOikarinen, 2021-03-17:
@jams032 let me double check I got your question correctly. 

STEPS:

1. H10 strap is around your chest and H10 is placed on the strap
2. Application utilising Polar iOS SDK is listening broadcast data from H10 sensor and data is good
3. H10 is removed from the strap and placed on the table
4. Application utilising Polar iOS SDK is still listening broadcast data from H10 sensor

EXPECTED RESULT:
- In step 4 the expected heart values are 0 

OBSEVED RESULT:
- In step 4 some random heart rate values are received from H10

jams032, 2021-03-18:
@JOikarinen, thanks for your reply.

Option 1 is happening here:
I left the H10 from the chest, but not removed it from the strap. I just put the polar(still attached with strap) on the table/near to me. 

In that case, I am receiving data wrong sometimes.

Q: Is it good practice/we can avoid abnormal data by removing it from the strap after use?

JOikarinen, 2021-03-18:
@jams032 I understand your setup now. 

> Q: Is it good practice/we can avoid abnormal data by removing it from the strap after use?
- yes,  the recommendation is the H10 module is removed from the strap after the use. If the H10 module is not removed there might be abnormal data still send by H10 module as the module may not detect that strap is not wear anymore.

- In additional here are some tips how shall the H10 maintained: https://support.polar.com/us-en/support/maintenance_of_heart_rate_sensor_with_textile_strap

I will close the issue, but don't hesitate to ask further questions on this topic if needed.

closed 2021-03-18 by JOikarinen

jams032, 2021-03-22:
ok, thanks.