polarofficial/polar-ble-sdk #139: Stream PPG, ACC and PPI at the same time

klandell, 2021-03-22:
Hello!
 
  I'm trying to stream the PPG, ACC, and PPI data at the same time.  When I stream both ACC and PPI at the same time, I always get the error `"Pmd response failed to receive in timeline"` which seems to be coming from `private byte[] receiveControlPointPacket()`.  Is streaming both of these features at the same time supported?  I wait until `streamingFeaturesReady` is called before starting the stream and I verify that PPI and ACC are in the results.
  
  I can calculate the PPI from the PPG if needed, so it isn't a huge deal if this isn't possible.

JOikarinen, 2021-03-23:
@klandell good finding. It took me a while to reproduce the problem, but now I can do it. The root cause maybe that start of PPI measurement actually resets the optical measurements together with accelerometer sensor. In my tests the same problem happens with both sensors OH1 and Polar Verity Sense. As a workaround I suggest that you change the start order of measurements to ACC, PPG and finally PPI. Or first PPI and then after some reasonable delay (let say 500ms) you start ACC and PPG. I didn't yet test the workaround, but I will continue try it later. I will also discuss with the engineers developing the firmware for OH1 and Polar Verity Sense whether we could spot the reason for you finding in sensor code.

JOikarinen, 2021-03-23:
@klandell I couldn't reproduce the bug if I change the start order Acc, Ppg and finally Ppi. I hope that's workaround you can utilise for now.

klandell, 2021-03-23:
@JOikarinen Thank you for looking into this so quickly!  Changing the start order to your suggestion of ACC, PPG, PPI seems to have resolved the issue in initial testing.

JOikarinen, 2021-03-24:
There will be fix for this issue in next Polar Verity Sense firmware, until then let's use the workarounds.

closed 2021-03-24 by JOikarinen