polarofficial/polar-ble-sdk #142: Polar OH1 PPG read failed - PolarServiceNotAvailable

TJUSIRIUS, 2021-03-25:
Hi,
i am trying to test the PPI and the PPG Feature of the OH1 sensor connected to your Android example application. 
i got 23:18:25.171 32086-32086/polar.com.androidblesdk D/MainActivity: HR value: 72 rrsMs: [] rr: [] contact: false,false
but when it comes to ppg, i got  ppg throwable  - ... - PolarServiceNotAvailable
i have update my framework to the newest...

Maybe you got an advice what I Need to do to make this work.

Thank you for your reply.
Best regards,
Sirius

JOikarinen, 2021-03-26:
@TJUSIRIUS I tried your steps but couldn't reproduce the problem. What is the phone model and Android version of your phone? 

Using the [example app](https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-android/androidBleSdkTestApp), my steps are
- I connect to OH1.
- From the logs I see heart rate callback is called `D/MainActivity: HR value: 49 rrsMs: [] rr: [] contact: false,false` 
- I press "PPG stream"
- From the logs I see ppg stream is started: `D/MainActivity:     ppg0: 863925 ppg1: 863925 ppg2: 863925 ambient: 863925`

TJUSIRIUS, 2021-03-29:
sorry i have fix my bug caused by wrong thread setting. (>.<)

closed 2021-03-29 by TJUSIRIUS