polarofficial/polar-ble-sdk #143: Stream HR, PPG, ACC on OH1, got HR

TJUSIRIUS, 2021-03-29:


TJUSIRIUS, 2021-03-29:
i also record the hr for a few minutes as a .txt file. the record shows the ble keeps receiving zero for seconds.
2021-03-26-16-32-37.424 hr: 84 rrsMs:  rr: 
2021-03-26-16-32-38.528 hr: 84 rrsMs:  rr: 
2021-03-26-16-32-39.446 hr: 0 rrsMs:  rr: 
2021-03-26-16-32-40.465 hr: 0 rrsMs:  rr: 
2021-03-26-16-32-41.426 hr: 0 rrsMs:  rr: 
2021-03-26-16-32-42.446 hr: 0 rrsMs:  rr: 
2021-03-26-16-32-43.436 hr: 0 rrsMs:  rr: 
2021-03-26-16-32-44.450 hr: 0 rrsMs:  rr: 
2021-03-26-16-32-45.477 hr: 0 rrsMs:  rr: 
2021-03-26-16-32-46.436 hr: 0 rrsMs:  rr: 
2021-03-26-16-32-47.426 hr: 0 rrsMs:  rr: 
2021-03-26-16-32-48.446 hr: 0 rrsMs:  rr: 
2021-03-26-16-32-49.466 hr: 0 rrsMs:  rr: 
2021-03-26-16-32-50.450 hr: 0 rrsMs:  rr: 
2021-03-26-16-32-51.445 hr: 0 rrsMs:  rr: 
2021-03-26-16-32-52.463 hr: 0 rrsMs:  rr: 
2021-03-26-16-32-53.451 hr: 0 rrsMs:  rr: 
2021-03-26-16-32-54.474 hr: 0 rrsMs:  rr: 
2021-03-26-16-32-55.442 hr: 0 rrsMs:  rr: 
2021-03-26-16-32-56.485 hr: 0 rrsMs:  rr: 
2021-03-26-16-32-57.446 hr: 0 rrsMs:  rr: 
2021-03-26-16-32-58.465 hr: 0 rrsMs:  rr: 
2021-03-26-16-32-59.435 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-00.451 hr: 72 rrsMs:  rr: 
2021-03-26-16-33-01.436 hr: 72 rrsMs:  rr: 
2021-03-26-16-33-02.427 hr: 72 rrsMs:  rr: 
2021-03-26-16-33-03.454 hr: 72 rrsMs:  rr: 
2021-03-26-16-33-04.436 hr: 72 rrsMs:  rr: 
2021-03-26-16-33-05.439 hr: 75 rrsMs:  rr: 
2021-03-26-16-33-06.476 hr: 75 rrsMs:  rr: 
2021-03-26-16-33-07.465 hr: 75 rrsMs:  rr: 
2021-03-26-16-33-08.458 hr: 75 rrsMs:  rr: 
2021-03-26-16-33-09.475 hr: 75 rrsMs:  rr: 
2021-03-26-16-33-10.435 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-11.465 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-12.445 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-13.460 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-14.425 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-15.506 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-16.445 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-17.447 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-18.445 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-19.468 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-20.425 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-21.451 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-22.437 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-23.473 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-24.444 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-25.436 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-26.435 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-27.446 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-28.436 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-29.428 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-30.444 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-31.438 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-32.489 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-33.446 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-34.469 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-35.426 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-36.474 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-37.436 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-38.479 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-39.475 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-40.470 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-41.456 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-42.450 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-43.466 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-44.430 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-45.475 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-46.439 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-47.428 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-48.446 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-49.465 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-50.425 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-51.476 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-52.435 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-53.449 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-54.445 hr: 0 rrsMs:  rr: 
2021-03-26-16-33-55.473 hr: 69 rrsMs:  rr: 
2021-03-26-16-33-56.425 hr: 69 rrsMs:  rr: 
2021-03-26-16-33-57.496 hr: 69 rrsMs:  rr: 
2021-03-26-16-33-58.436 hr: 69 rrsMs:  rr: 
2021-03-26-16-33-59.427 hr: 69 rrsMs:  rr: 
2021-03-26-16-34-00.476 hr: 48 rrsMs:  rr: 
2021-03-26-16-34-01.437 hr: 48 rrsMs:  rr: 
2021-03-26-16-34-02.455 hr: 48 rrsMs:  rr: 
2021-03-26-16-34-03.446 hr: 48 rrsMs:  rr: 
2021-03-26-16-34-04.435 hr: 48 rrsMs:  rr:

TJUSIRIUS, 2021-03-29:
Again i come here for your advice. : )
Maybe you got information what I Need to do to make this work.

Thank a lot for your reply.
Best regards,
Sirius

JOikarinen, 2021-03-29:
@TJUSIRIUS is it possible that heart rate is not reliable at that time you receive heart rate 0? Maybe the OH1 is not contacting the skin on those moments?

Even your app receives ACC and PPG streams it might be possible that received heart rate is 0. The algorithm calculating heart rate inside OH1 firmware is based ACC and PPG inputs, however the algorithm may consider that heart rate cannot be calculated for the inputs given. The algorithm doing the heart rate calculation is not open sourced.

TJUSIRIUS, 2021-03-29:
for the first, during testing, the band is constrained on my arm quite tightly.

and the second, you mean when the OH1 giving PPG and ACC, the inside algorithm would judge that, and decide not to calculate or give the heartrate?

JOikarinen, 2021-03-29:
> for the first, during testing, the band is constrained on my arm quite tightly.
> 
> and the second, you mean when the OH1 giving PPG and ACC, the inside algorithm would judge that, and decide not to calculate or give the heartrate?

I see your problem. At the moment, that is the only explanation I can figure out for the heart rate being 0 every now and then. I will do some own experiments whether I could spot something suspicious from SDK.  

This blog, loosely related on the topic, may be something to read: https://www.polar.com/blog/optical-heart-rate-tracking-polar/

erikboto, 2021-03-31:
I can verify that I see the same behavior. As long as I don't turn on PPI streaming I get good heart rate data over the standard BLE heartrate service, but when I turn on PPI I will get a lot more HR=0 values. I also noted that the HR value doesn't update every second anymore, and instead only update every fifth second in sync with when the new PPI values are sent.

JOikarinen, 2021-04-01:
I had discussion with the experts involved on sensor development. What I learnt is that when enabling the PPI stream the firmware switches on different heart rate calculation algorithm. The reason why algorithm is changed is that Pulse to Pulse interval (PPI) calculation is much more sensitive for any movements of the device or other possible interference. Algorithm calculates primary the PPI and then from the PPI the heart rate in bpm. That heart rate is then delivered with PPI stream and heart rate service. I believe the consequences of that algorithm switch is what @erikboto has observed and explained in previous comment. 

@TJUSIRIUS in your original description you don't mention PPI, but I assume you may have enabled PPI stream? Or do you see the problem also when you have ACC and PPG streams enabled (not PPI)?

JOikarinen, 2021-04-01:
The previous comment is effective for OH1 and Verity Sense sensors. While your solution is not requiring the PPI, then it maybe a good idea to stop PPI stream.

closed 2023-09-08 by jimmyzumthurm