polarofficial/polar-ble-sdk #144: E/libc: Access denied finding property "vendor.gralloc.disable_ahardware_buffer"

michaelscottsorenson, 2021-03-29:
I am attempting to set up an H10 heart rate sensor on android. I am seeing this error when setting up the sdk.

> E/libc: Access denied finding property "vendor.gralloc.disable_ahardware_buffer"

Here is the full log output

```
03/28 21:11:25: Launching 'app' on Google Pixel 3.
Install successfully finished in 451 ms.
$ adb shell am start -n "com.example.app/com.example.app.MainActivity" -a android.intent.action.MAIN -c android.intent.category.LAUNCHER
Connected to process 10364 on device 'google-pixel_3-8BVX1FXMC'.
Capturing and displaying logcat messages from application. This behavior can be disabled in the "Logcat output" section of the "Debugger" settings page.
D/NetworkSecurityConfig: No Network Security Config specified, using platform default
D/NetworkSecurityConfig: No Network Security Config specified, using platform default
W/mple.app: Accessing hidden method Landroid/view/View;->computeFitSystemWindows(Landroid/graphics/Rect;Landroid/graphics/Rect;)Z (greylist, reflection, allowed)
W/mple.app: Accessing hidden method Landroid/view/ViewGroup;->makeOptionalFitsSystemWindows()V (greylist, reflection, allowed)
D/AppName: BLE power: true
D/BluetoothAdapter: isLeEnabled(): ON
D/BluetoothLeScanner: onScannerRegistered() - status=0 scannerId=11 mScannerId=0
I/AdrenoGLES-0: QUALCOMM build                   : 781e7d0, I46ff5fc46f
    Build Date                       : 12/01/20
    OpenGL ES Shader Compiler Version: EV031.31.04.01
    Local Branch                     : QPR1
    Remote Branch                    : 
    Remote Branch                    : 
    Reconstruct Branch               : 
    Build Config                     : C P 11.0.1 AArch64
    Driver Path                      : /vendor/lib64/egl/libGLESv2_adreno.so
I/AdrenoGLES-0: PFP: 0x016ee189, ME: 0x00000000
W/AdrenoUtils: <ReadGpuID_from_sysfs:197>: Failed to open /sys/class/kgsl/kgsl-3d0/gpu_model
    <ReadGpuID:221>: Failed to read chip ID from gpu_model. Fallback to use the GSL path
I/Gralloc4: mapper 4.x is not supported
W/Gralloc3: mapper 3.x is not supported
E/libc: Access denied finding property "vendor.gralloc.disable_ahardware_buffer"
```

The sdk usage currently looks like this

```kt
        val api = PolarBleApiDefaultImpl.defaultImplementation(this, PolarBleApi.FEATURE_HR)

        api.setApiCallback(object : PolarBleApiCallback() {
            override fun blePowerStateChanged(powered: Boolean) {
                Log.d(appName, "BLE power: $powered")
            }

            override fun deviceConnected(polarDeviceInfo: PolarDeviceInfo) {
                Log.d(appName, "CONNECTED: " + polarDeviceInfo.deviceId)
            }

            override fun deviceConnecting(polarDeviceInfo: PolarDeviceInfo) {
                Log.d(appName, "CONNECTING: " + polarDeviceInfo.deviceId)
            }

            override fun deviceDisconnected(polarDeviceInfo: PolarDeviceInfo) {
                Log.d(appName, "DISCONNECTED: " + polarDeviceInfo.deviceId)
            }

            override fun streamingFeaturesReady(@NonNull identifier: String,
                                                @NonNull features: Set<DeviceStreamingFeature>) {
                for (feature in features) {
                    Log.d(appName, "Streaming feature $feature is ready")
                }
            }

            override fun hrFeatureReady(identifier: String) {
                Log.d(appName, "HR READY: $identifier")
            }

            override fun batteryLevelReceived(identifier: String, level: Int) {}
            override fun hrNotificationReceived(identifier: String, data: PolarHrData) {
                Log.d(appName, "HR: " + data.hr)
            }

            override fun polarFtpFeatureReady(s: String) {}
        })

        this.requestPermissions(arrayOf(Manifest.permission.ACCESS_COARSE_LOCATION, Manifest.permission.BLUETOOTH, Manifest.permission.BLUETOOTH_ADMIN), 1)
        
        api.autoConnectToDevice(-50, null, null).subscribe()
```

I have followed the installation instructions in the readme for android. I'm also using the app "Polar Beat", and the sensor is connected in that app. Do you have any advice for getting passed this error?

JOikarinen, 2021-03-29:
@michaelscottsorenson I tried to reproduce your issue by using your code snippet on Pixel 2, Pixel 4a and Pixel 3 simulator. However I cannot get the error `E/libc: Access denied finding property "vendor.gralloc.disable_ahardware_buffer"`

Could you try these actions:
- run your code on other device or simulator
- run the original [android example](https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-android/androidBleSdkTestApp) on your Pixel 3

JOikarinen, 2021-03-29:
@michaelscottsorenson actually now that I think again your original explanation I may have solution for you. Initially I thought your app is crash in log line `E/libc: Access denied finding property "vendor.gralloc.disable_ahardware_buffer"` but that might not be the case. 

I suppose the problem is the missing permission as bluetooth requires the `ACCESS_FINE_LOCATION`, at the moment your code snippet requests only `ACCESS_COARSE_LOCATION`. 

Another note related to permission request is that `api.autoConnectToDevice()` does nothing until the correct location permission is granted. 

If your modify your code like this, I believe it starts to work:

```
    this.requestPermissions(arrayOf(Manifest.permission.ACCESS_FINE_LOCATION, Manifest.permission.BLUETOOTH, 

    override fun onRequestPermissionsResult(requestCode: Int, permissions: Array<String>, grantResults: IntArray) {
        if (requestCode == 1) {
            api.autoConnectToDevice(-50, null, null).subscribe()
        }
    }

```

JOikarinen, 2021-03-29:
Also noticed that README section [Android: Getting started](https://github.com/polarofficial/polar-ble-sdk#android-getting-started) wasn't clear about the permissions. Fixed in https://github.com/polarofficial/polar-ble-sdk/commit/40ae0c4d934a117b67f66e520510dbdea4e2fbd6

michaelscottsorenson, 2021-04-02:
@JOikarinen I was able to get SDK connected my H10 after those permission changes you suggested. Thank you for your help. I am still seeing that error message. It hasn't stopped the SDK from connecting to the device.

JOikarinen, 2021-04-07:
@michaelscottsorenson the error may be some internal error in Qualcomm bluetooth stack. It is a bit mysterious. However if it is not seemingly effecting on any BLE communication maybe just ignoring for now. 

I will close this issue as resolved.

closed 2021-04-07 by JOikarinen