polarofficial/polar-ble-sdk #146: What is mean "error Estimate"

GLC-Kazetani, 2021-04-07:
Dear Polar team,
What is mean "error Estimate" in PolarOhrPPIData()？
SDK Document is "errorEstimate in ms" only
How should I use this value?

In my case, I want to know the elapsed time of measurement
Should I add up the PPI? Or is it better to add this value as well?
(Did you fix the bug that the timestamp is always 0?)

Kind regards

JOikarinen, 2021-04-07:
Hi @GLC-Kazetani,

here are my first thoughts of your questions. I will try to cover more later. 

> What is mean "error Estimate" in PolarOhrPPIData()？
- it is the expected estimation error for PP interval in milliseconds.

> How should I use this value?
- you may use it to detect the quality of received Pulse-to-Pulse interval. (ppErrorinMs/ppInMs) * 100 = error percentage

> In my case, I want to know the elapsed time of measurement
> Should I add up the PPI?
- as there is no timestamp (i.e. it is always 0) in the PPI data your option is to keep track of the time by your application. I don't recommend adding up the pulse-to-pulse interval as because of possible error in the interval your timing will start to drift.

> Or is it better to add this value as well?
- error estimate is unsigned value. I recommend not to add up as it is only estimation and in reality the sign of the value is not know.

GLC-Kazetani, 2021-04-07:
Hi @JOikarinen , Reply thank you!!

Oh, ppinMs + ppErrorinMs is not Elapsed time.
But, I can use PPI data of error percentage.
How many are good quality indicators? (Resting and exercising)

> your option is to keep track of the time by your application. 

Is it correct to get the time on the Android device?
But the time that BLE receives is every 5 seconds.

JOikarinen, 2021-04-07:
> Oh, ppinMs + ppErrorinMs is not Elapsed time.
- ppErrorInMs is absolute value

> But, I can use PPI data of error percentage.
> How many are good quality indicators? (Resting and exercising)
- the value indicates the quality of PP-intervals. When error estimate is below 10ms the PP-intervals are probably very accurate. Error estimate values over 30ms may be caused by movement artefact or too loose sensor-skin contact.

> Is it correct to get the time on the Android device?
- that's the best option at the moment
- note that one received [PolarOhrPPIData](https://github.com/polarofficial/polar-ble-sdk/blob/f8f71b58ac380e1c754daff4c71d7393136d5651/sources/Android/android-communications/src/main/java/polar/com/sdk/api/model/PolarOhrPPIData.java#L66) may contain multiple [samples](https://github.com/polarofficial/polar-ble-sdk/blob/f8f71b58ac380e1c754daff4c71d7393136d5651/sources/Android/android-communications/src/main/java/polar/com/sdk/api/model/PolarOhrPPIData.java#L64) so you may need to calculate time stamp estimate for each received sample

JOikarinen, 2021-04-07:
Documentation enhancements done in https://github.com/polarofficial/polar-ble-sdk/commit/f595dce3ec49a0edf97ea3bf47eb0d6eb7df3883

GLC-Kazetani, 2021-04-08:
> note that one received PolarOhrPPIData may contain multiple samples so you may need to calculate time stamp estimate for each received sample

Yes, I will probably receive as many samples as the number of beats measured in 5 seconds.
Is it better to estimate with the PPI integrated value of the received [PolarOhrPPIData](https://github.com/polarofficial/polar-ble-sdk/blob/f8f71b58ac380e1c754daff4c71d7393136d5651/sources/Android/android-communications/src/main/java/polar/com/sdk/api/model/PolarOhrPPIData.java#L66) based on the reception time of 5 seconds on the Android device?

JOikarinen, 2021-04-08:
> Is it better to estimate with the PP-intervals integrated value of the received [PolarOhrPPIData](https://github.com/polarofficial/polar-ble-sdk/blob/f8f71b58ac380e1c754daff4c71d7393136d5651/sources/Android/android-communications/src/main/java/polar/com/sdk/api/model/PolarOhrPPIData.java#L66) based on the reception time of 5 seconds on the Android device?

If I understand correctly, you would like to calculate timestamp for each peak, right? At the moment the PPI stream is not providing very convenient way of doing so. 

Here is snippet of the Android log where PolarOhrPPIData contains four samples:
```
2021-04-08 07:44:58.558  D/TESTING: sample.ppi 1313
2021-04-08 07:44:58.559  D/TESTING: sample.errorEstimate 10
2021-04-08 07:44:58.559  D/TESTING: sample.ppi 1246
2021-04-08 07:44:58.559  D/TESTING: sample.errorEstimate 7
2021-04-08 07:44:58.560  D/TESTING: sample.ppi 1295
2021-04-08 07:44:58.560  D/TESTING: sample.errorEstimate 6
2021-04-08 07:44:58.560  D/TESTING: sample.ppi 1308
2021-04-08 07:44:58.560  D/TESTING: sample.errorEstimate 7
```
In theory each peak could be calculated like this:
peak 0: 07:44:55.957 - 1.313 = 07:44:54.644
peak 1: 07:44:57.252 - 1.295 = 07:44:55.957
peak 2: 07:44:58.560 - 1.308 = 07:44:57.252
peak 3: 07:44:58.560

However, as peak3 timestamp is taken from Android phone we already know that it is not representing the real peak time. We cannot know how much it is wrong as there is lag before data gets from device to your application (i.e. processing on device, bluetooth communication and finally processing on your code).

As explained the solution is not perfect. I will trigger internal issue, whether it could be possible to add absolute timestamp for PPI data.

GLC-Kazetani, 2021-04-08:
> If I understand correctly, you would like to calculate timestamp for each peak, right? At the moment the PPI stream is not providing very convenient way of doing so.

Yes, I want like to calculate timestamp for each peak.
However I also understood that there is no timestamp associated with PPI.
Since there are variations in the timing of receiving BLE, I also found that there is a risk of replacing the previous PPI if simply calculated with PPI.

Thank you for your polite answer.

closed 2021-04-08 by GLC-Kazetani