polarofficial/polar-ble-sdk #148: Forcing H10 to stay on overnight

skylark1991, 2021-04-12:
Hello,

I am using polar sdk to make ECG readings from my H10 chest strap in python. I would like to make the readings overnight as I sleep, but the sensor keeps turning off after about 1 hour. I was wondering if there is a way to force it to stay on through SDK? Inaccuracy is tolerated if it falls off.

Also would have been nice to change sampling frequency to values other than 130Hz.

Thanks!

JOikarinen, 2021-04-19:
@skylark1991 interesting observation as H10 is designed to keep up and running when ECG stream is requested. If ECG stream is requested from H10 sensor, the H10 sensor shall continue sending ECG data endlessly. Only if ECG stream is stopped or bluetooth connection is closed, then H10 returns to its normal operation.

What is the firmware version of the H10 you are experimenting this problem? Could it be something in your environment timeouts the BLE connection after 1 hour?

I will test your use case using Android phone <-> H10, with ECG enabled whether I could reproduce your observation.

JOikarinen, 2021-04-19:
@skylark1991 with quick test I cannot reproduce the problem you descriped. I have H10 on table, not even connected to strap and chest and ECG stream has been now on for almost three hours.

JOikarinen, 2021-04-19:
> Also would have been nice to change sampling frequency to values other than 130Hz.

At the moment, there is no plans to add support for other sampling rates. However, could you give examples of use cases where you would need other sampling frequencies than 130Hz? I can work on feature request internally, practical use cases makes it easier to get attention for the request.

andrasferenczi, 2021-04-19:
To add to the discussion, I have also encountered something similar, but I couldn't entirely rule out if it was the device or some other issues on my phone (like maybe memory?). I connected to the `H10` device with `Android`, using the provided API.

I opted for measuring the acceleration data (with the least accuracy setting configured) and heart rate. I set up the recording from `20:23:20` `UTC` and acceleration data started missing at `21:50:58`, heart rate data was provided reasonably consistently, which means the connection was stable throughout the night. This was just a single measurement though, haven't tried since and couldn't check the logs what exactly caused it.

I will look into this further on my side what caused it and I'll keep an eye on this thread. It just seemed interesting that someone else encountered this issue as well.

poml88, 2021-04-20:
Using my iOS app I can record over night. No problem. Connection seems to be very stable.

skylark1991, 2021-04-20:
I would like to also confirm that I was able to record overnight without any problem using iOS app. The problem is my python code then. I know it's not part of your SDK, but do you happen to have a working python code example lying around for streaming ECG signal to a computer instead of a cellphone?

JOikarinen, 2021-04-25:
> I know it's not part of your SDK, but do you happen to have a working python code example lying around for streaming ECG signal to a computer instead of a cellphone?

@skylark1991 here is @erikboto's code example code for OH1 PPG streaming. The example may
giving you ideas for ECG streaming from H10.

closed 2023-09-08 by jimmyzumthurm