polarofficial/polar-ble-sdk #149: What is the PPG acquisition interval?

GLC-Kazetani, 2021-04-13:
Hi, Polar teams!
I am using VERITY SENSE to get PPG Stream.

What is the PPG acquisition interval?
The average BLE acquisition interval was 7ms.
However, it looks like PPI is getting at 1ms intervals

Kind regards.

JOikarinen, 2021-04-19:
PPG sampling rate is 135Hz. You may check that by reading the PPG stream settings. 

Android: https://github.com/polarofficial/polar-ble-sdk/blob/085bf6be0f8b19fa9d5222b9632768edfd9fa923/sources/Android/android-communications/src/main/java/polar/com/sdk/api/PolarBleApi.java#L193

iOS: https://github.com/polarofficial/polar-ble-sdk/blob/085bf6be0f8b19fa9d5222b9632768edfd9fa923/sources/iOS/ios-communications/iOSCommunications/sdk/api/PolarBleApi.swift#L228

JOikarinen, 2021-04-19:
> The average BLE acquisition interval was 7ms.

It is good to mention here that one BLE package may contain PPG multiple samples and timestamp of package is the timestamp of last sample. 

Android: https://github.com/polarofficial/polar-ble-sdk/blob/085bf6be0f8b19fa9d5222b9632768edfd9fa923/sources/Android/android-communications/src/main/java/polar/com/sdk/api/model/PolarOhrData.java#L58

iOS:https://github.com/polarofficial/polar-ble-sdk/blob/085bf6be0f8b19fa9d5222b9632768edfd9fa923/sources/iOS/ios-communications/iOSCommunications/sdk/api/PolarBleApi.swift#L99

GLC-Kazetani, 2021-04-22:
Reply thank you!!
Okay, PPG sampling rate is 135Hz = 7.40740740ms.

Of course, the BLE reception time is for reference only.
Measured at 135Hz on a regular basis.
And the device is sending PPG multiple samples at once with BLE.

Can you write the sampling rate of PPG in Readmd.md?
Many thanks.

closed 2021-04-22 by GLC-Kazetani

JOikarinen, 2021-04-25:
> Can you write the sampling rate of PPG in Readmd.md?

I understand your wish here. But I would like to enhance the Readme and knowledge of SDK user to direction that each stream have a setting which tells the capabilities of that particular stream, this is because SDK is supporting many Polar Devices (Polar H10, Polar OH1, Polar Verity sense etc.) so capabilities may vary between the devices. 

I would like to