polarofficial/polar-ble-sdk #152: Spikes in Polar PPG Sensor values

andykorth, 2021-04-22:
We have several Polar Verity Sensors, and are interested in logging the PPG signals. When we do so, we notice regular spikes in each of the channels (PPG1, PPG2, PPG3, and ambient), and these interfere with our analysis. Sometimes the spikes are a significant proportion of the PPG signal itself. It seems to vary by device or by user, but they do occur on all our devices to some extent or another.

Here is an unfiltered plot of the raw data off the sensors (shifted to remove the y-offsets):
http://kortham.homeip.net/files/PolarPPGSpikes.png

PPG0 is blue, PPG1 in orange, PPG2 in green, and ambient in red. 

These abnormalities strike me as potential electrical noise, just do the sharp upward spike and exponential decline. You'll notice there's a bit of a pattern as the signal starts on PPG1, then 2, then ambient, then 0. Sometimes there are 3 spikes, sometimes 5. 

In runs where the PPG signal is stronger, the spikes are perhaps easier to filter out with proper bandpass filters, but they always seem present.
http://kortham.homeip.net/files/chrome_2021-04-21_16-49-31.png

Zooming in, you can get a sense for the shape of these spikes:
http://kortham.homeip.net/files/chrome_2021-04-22_14-47-21.png

Is this expected behavior? Is it indicative of hardware issues?

andykorth, 2021-04-22:
In this particular run, the noise occurs for the first 8 seconds then stops:
[image]

slembcke, 2021-04-22:
On a second sensor, we get much worse spikes that have a much greater magnitude than the ppg wave itself. You can see approximately one full ppg wave in the output below.
[image]

On a third sensor we don't get any spikes, though the signal level just randomly shifts by a huge amount every once in a while. Zooming in on the level shifts, they all seem to happen simultaneously, but also have an exponential curve to them.
[image]
[image]

Again, these plots are all unfiltered raw data written out directly as they come to us through the API.

In order to process the PPG data, the spike issue has necessitated a very strong and narrow bandpass filter to remove them, and the second issue causes a nasty ringing step response using the same filter.

JOikarinen, 2021-04-25:
@andykorth and @slembcke thank you for the very informative descriptions of your observations. I will escalate Polar internal issue of your observation.

JOikarinen, 2021-04-26:
With quick test runs we couldn't reproduce the problem. 

Are you running the SDK on Android or iOS phone?
Is there other data streams enabled (e.g. Acceleration) at the time you notice the spikes?

slembcke, 2021-04-26:
This is on Android, and we only enable the PPG stream. Someone found a note that enabling other streams can interfere with the sampling rate of the PPG.

slembcke, 2021-04-26:
Sorry for the double post. We were discussing this again this morning and thought we weren't using any other callbacks because in the Unity/C# portion of the code we weren't using any other streams, but PPI was enabled in the Java code. We are disabling it now and will let you know what happens.

andykorth, 2021-04-26:
We found the source of one of these issues. All three of us have Vive lighthouses, which emit patterns of IR light.  I conducted a test with the Polar covered by a blanket, and then with an outstretched arm in front of the lighthouse. 

Here is the transition zone from blanket to no-blanket:

[image]

So this is a very conclusive finding for the spikes. Fortunately, this product isn't targeting the Vive, so we can avoid that issue.

The lighthouses run at a pretty high frequency, which may explain why they show up at different times on different channels. That probably makes it impossible to filter it out with the ambient sensor do to those time offsets. 

We do still see occasional discontinuous level shifts, as noted my @slembcke above. These have occurred on more than one sensor, but on average we only see them in <20% of our 5 minute sessions.

JOikarinen, 2021-04-27:
Excellent that source of the problem is now found. Thank you for very precise analysis.👍

> We do still see occasional discontinuous level shifts, as noted my @slembcke above. These have occurred on more than one sensor, but on average we only see them in <20% of our 5 minute sessions.

We will continue to investigations on this. Yesterday there were another report with similar observations.

JOikarinen, 2021-04-27:
> We do still see occasional discontinuous level shifts, as noted my @slembcke above. These have occurred on more than one sensor, but on average we only see them in <20% of our 5 minute sessions.

After discussion with engineers on Polar PPG development side it came obvious that reason seems to be a level change by autogain algorithm that keeps the signal at a desired level. Typical for the beginning of a measurement. The algorithm may do the level changes to find the best suitable light intensity level for the current conditions. 

For now the advice is that one can implement a high pass or a step filter or just ignore the first seconds of the recording, after which autogain won't do any further adjustments if the user of the device stays still and the DC-level remains stable.

For the future we will try to figure out better solution how the autogain effect could be hided from the API user.

slembcke, 2021-04-27:
I'm still suspicious that the spikes are from 60 Hz flicker of the lighting in the room vs the 50 Hz sample rate of the VR trackers since the spikes occur at ~15 hz intervals. Could explain why the spikes show up at different times in different channels if they are sampled sequentially. Anyway... kinda doesn't matter. We have a pretty strong bandpass filter in place that is working well enough for now.

As for the level shifts, as long as we know that it's expected behavior and not a bug we can deal with that. The transitions decay pretty quickly, so I think it should be easy to deal with them effectively.

JOikarinen, 2021-05-26:
@slembcke I will close this issue as resolved. I assume that you were able to resolve existing concerns highlighted on this thread.

closed 2021-05-26 by JOikarinen