polarofficial/polar-ble-sdk #155: Protobuf and protobuf-javalite

dusanpetren, 2021-05-02:
**Question concerns:**
- [ X] Android SDK
- [ ] iOS SDK
- [ ] Polar device
- [ ] General

**Description:**
Hello, I am trying to build an Android application with polar-sdk and google firebase firestory. 
I came across this bug where I can not use Firebase, becouse it is using protobuf-javalite and the Polar-SDK is using protobuf which are incompatible. 

Do you think you can provide any solution how to fix this problem ?

JOikarinen, 2021-05-04:
@dusanpetren there is now beta release https://github.com/polarofficial/polar-ble-sdk/releases/tag/3.0.6-beta0 where protobuf java is changed to protobuf java-lite. It shall help with the issue you have experienced.

dusanpetren, 2021-05-04:
@JOikarinen Thanks for the response. For now I replaced the Firestore lib with local SQLite DB. Will try the new release of the SDK in next iteration.

nikolmamai, 2022-04-06:
I'm facing the same issue with the latest releases of firebase and polar sdk for android client. Is this surely fixed?

JOikarinen, 2022-04-06:
@nikolmamai which of the firebase features is causing this? I could try to reproduce.

JOikarinen, 2022-04-07:
@nikolmamai the reason behind this problem is that PolarBleSdk and some of the Firebase features uses the Google Protocol Buffers internally. 

The workaround for the issue is to exclude the module `protolite-well-known-types` from that Firebase feature which is internally using the proto files.  For example, the Firebase In App Messaging uses the proto files internally. The workaround in your `build.gradle` may look something like this

```
dependencies {
...
  implementation platform('com.google.firebase:firebase-bom:29.2.1')
      // When using the BoM, don't specify versions in Firebase dependencies
      implementation 'com.google.firebase:firebase-analytics'
      implementation 'com.google.firebase:firebase-crashlytics'
      implementation('com.google.firebase:firebase-inappmessaging-display', {
          exclude group: 'com.google.firebase', module: 'protolite-well-known-types'
      })
...
}
```

nikolmamai, 2022-04-13:
Hi @JOikarinen,

nikolmamai, 2022-04-13:
Hi @JOikarinen ,
Thank you for your quick reply and I apologize for my delayed response. Please take a look at my dependencies

`api(platform(LibraryDependency.FIREBASE_BOM))
    api(LibraryDependency.FIREBASE_CORE)
    api(LibraryDependency.FIREBASE_AUTH)
    api(LibraryDependency.FIREBASE_ANALYTICS)
    api(LibraryDependency.FIREBASE_CRASHLYTICS)
    api(LibraryDependency.FIREBASE_FIRESTORE)
    api(LibraryDependency.FIREBASE_MESSAGING)`

The problem is caused by **firestore** dependency. I tried to exclude it as indicated at your comment but then my app is crashed, since I assume that is essential for firestore to work. These are some logs:

`java.lang.RuntimeException: Internal error in Cloud Firestore (24.1.0).
        at com.google.firebase.firestore.util.AsyncQueue.lambda$panic$3(AsyncQueue.java:539)
        at com.google.firebase.firestore.util.-$$Lambda$AsyncQueue$jx84dqgUsF4ojecSMurRqFLFD1Y.run(Unknown Source:2)
        at android.os.Handler.handleCallback(Handler.java:883)
        at android.os.Handler.dispatchMessage(Handler.java:100)
        at android.os.Looper.loop(Looper.java:224)
        at android.app.ActivityThread.main(ActivityThread.java:7562)
        at java.lang.reflect.Method.invoke(Native Method)
        at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:539)
        at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:950)
     Caused by: java.lang.NoClassDefFoundError: Failed resolution of: Lcom/google/type/LatLng;
        at com.google.firestore.v1.Value.dynamicMethod(Value.java:1626)
        at com.google.protobuf.GeneratedMessageLite.dynamicMethod(GeneratedMessageLite.java:256)
        at com.google.protobuf.GeneratedMessageLite.buildMessageInfo(GeneratedMessageLite.java:286)
        at com.google.protobuf.GeneratedMessageInfoFactory.messageInfoFor(GeneratedMessageInfoFactory.java:60)
        at com.google.protobuf.ManifestSchemaFactory$CompositeMessageInfoFactory.messageInfoFor(ManifestSchemaFactory.java:143)
        at com.google.protobuf.ManifestSchemaFactory.createSchema(ManifestSchemaFactory.java:55)
        at com.google.protobuf.Protobuf.schemaFor(Protobuf.java:90)
        at com.google.protobuf.Protobuf.schemaFor(Protobuf.java:104)
        at com.google.protobuf.GeneratedMessageLite.makeImmutable(GeneratedMessageLite.java:175)
        at com.google.protobuf.GeneratedMessageLite$Builder.buildPartial(GeneratedMessageLite.java:397)
        at com.google.protobuf.GeneratedMessageLite$Builder.build(GeneratedMessageLite.java:405)
        at com.google.firebase.firestore.model.ObjectValue.<init>(ObjectValue.java:60)
        at com.google.firebase.firestore.model.MutableDocument.newInvalidDocument(MutableDocument.java:94)
        at com.google.firebase.firestore.local.SQLiteRemoteDocumentCache.getAll(SQLiteRemoteDocumentCache.java:123)
        at com.google.firebase.firestore.local.SQLiteRemoteDocumentCache.get(SQLiteRemoteDocumentCache.java:111)
        at com.google.firebase.firestore.local.LocalDocumentsView.getBaseDocument(LocalDocumentsView.java:367)
        at com.google.firebase.firestore.local.LocalDocumentsView.getDocument(LocalDocumentsView.java:92)
        at com.google.firebase.firestore.local.LocalDocumentsView.getDocumentsMatchingDocumentQuery(LocalDocumentsView.java:248)
        at com.google.firebase.firestore.local.LocalDocumentsView.getDocumentsMatchingQuery(LocalDocumentsView.java:235)
        at com.google.firebase.firestore.local.QueryEngine.executeFullCollectionScan(QueryEngine.java:239)
        at com.google.firebase.firestore.local.QueryEngine.getDocumentsMatchingQuery(QueryEngine.java:99)
        at com.google.firebase.firestore.local.LocalStore.executeQuery(LocalStore.java:846)
        at com.google.firebase.firestore.core.SyncEngine.initializeViewAndComputeSnapshot(SyncEngine.java:213)
        at com.google.firebase.firestore.core.SyncEngine.listen(SyncEngine.java:206)
        at com.google.firebase.firestore.core.EventManager.addQueryListener(EventManager.java:104)
        at com.google.firebase.firestore.core.FirestoreClient.lambda$listen$7$FirestoreClient(FirestoreClient.java:176)
        at com.google.firebase.firestore.core.-$$Lambda$FirestoreClient$AxqBw4pFz0h3P8oIu9EDjIgL85c.run(Unknown Source:4)
        at com.google.firebase.firestore.util.AsyncQueue.lambda$enqueue$2(AsyncQueue.java:441)
        at com.google.firebase.firestore.util.-$$Lambda$AsyncQueue$Y9GO7zO2DX6MzQJ5TbKtKW_NzO0.call(Unknown Source:2)
        at com.google.firebase.firestore.util.AsyncQueue$SynchronizedShutdownAwareExecutor.lambda$executeAndReportResult$1(AsyncQueue.java:327)
        at com.google.firebase.firestore.util.-$$Lambda$AsyncQueue$SynchronizedShutdownAwareExecutor$pKSAgg-ozJ8RbUveSl-44aUfFQw.run(Unknown Source:4)
        at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:462)
2022-04-13 10:29:28.887 23014-23014/? E/AndroidRuntime:     at java.util.concurrent.FutureTask.run(FutureTask.java:266)
        at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:301)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641)
        at com.google.firebase.firestore.util.AsyncQueue$SynchronizedShutdownAwareExecutor$DelayedStartFactory.run(AsyncQueue.java:234)
        at java.lang.Thread.run(Thread.java:919)
[... cut ...] plit_feature_login_register.apk", zip file "/data/app/gr.auth.heart_inspector-RAv2LLFynO9NaS9HWOe4ww==/split_feature_profile.apk"],nativeLibraryDirectories=[/data/app/gr.auth.heart_inspector-RAv2LLFynO9NaS9HWOe4ww==/lib/arm64, /system/lib64, /system/product/lib64]]
        at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:230)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:379)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:312)
        	... 38 more`

JOikarinen, 2022-04-14:
> I tried to exclude it as indicated at your comment but then my app is crashed, since I assume that is essential for firestore to work.
Okay, sound logical. 

I can see the Firebase libraries (In app messaging, firestore etc.) are depending on the `protobuf-javalite:3.19.2` , whereas Polar BLE SDK is dependent on the `protobuf-javalite:3.20.0`. Well I guess, that is not the problem, but the fact the `protobuf-javalite` is missing some features. This is widely discussed issue, but I haven't find yet what is the proper and complete fix on this issue. 

@nikolmamai I will come back on this issue next week.

JOikarinen, 2022-04-19:
Hi @nikolmamai,

there is now attempt to fix the issue in branch https://github.com/polarofficial/polar-ble-sdk/tree/155-protobuf-and-protobuf-javalite

Is it possible you test that branch in your application? 
``` implementation 'com.github.polarofficial:polar-ble-sdk:155-protobuf-and-protobuf-javalite-SNAPSHOT'```

You can remove the `exclude group: 'com.google.firebase', module: 'protolite-well-known-types'` statements I suggested in previous posts. 

Your gradle dependencies may look something like this

```
dependencies {
  ...
  implementation 'com.github.polarofficial:polar-ble-sdk:155-protobuf-and-protobuf-javalite-SNAPSHOT'
  ...
  implementation platform('com.google.firebase:firebase-bom:29.2.1')
    // When using the BoM, don't specify versions in Firebase dependencies
    implementation 'com.google.firebase:firebase-analytics'
    implementation 'com.google.firebase:firebase-crashlytics'
    implementation 'com.google.firebase:firebase-inappmessaging-display'
    implementation 'com.google.firebase:firebase-firestore'
   ...
}
```

JOikarinen, 2022-04-22:
Hi @nikolmamai, 

> but a crash happens
the crash shown by logs snippet shall not be related to changes in branch https://github.com/polarofficial/polar-ble-sdk/commits/155-protobuf-and-protobuf-javalite

The above log snippet makes me thinking that in your code you release the resources of PolarBleSDK by calling [shutDown()](https://github.com/polarofficial/polar-ble-sdk/blob/610d5e1a4cb38d05d873794055e60a44616bcbe5/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L140). Since the releasing of the resources you need to get the new instance of PolarBleSDK by calling [PolarBleApiDefaultImpl.defaultImplementation()](https://github.com/polarofficial/polar-ble-sdk/blob/610d5e1a4cb38d05d873794055e60a44616bcbe5/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApiDefaultImpl.kt#L19)

nikolmamai, 2022-04-28:
Hi @JOikarinen ,
You are right. Something went wrong because of permissions and the new release of sdk since I used an old release before. Please inform me when the change about protobuf will be available. Thank you for the quick and detailed support!

closed 2022-05-12 by JOikarinen

JOikarinen, 2022-05-18:
Hi @nikolmamai, the fix for this issue is now officially released in [3.3.1](https://github.com/polarofficial/polar-ble-sdk/releases/tag/3.3.1)