polarofficial/polar-ble-sdk #156: Basic app fails to build: "More than one file was found with OS independent path 'google/protobuf/descriptor.proto'."

Erhannis, 2021-05-06:
**Describe the bug**
Following the instructions, I opened Android Studio 4.1.1 on Windows 10 Professional, created a new app with the minSdkVersion set to 21, copied the two aar files into app/libs/, added the given list of dependencies to the app's build.gradle, added the given permissions to AndroidManifest.xml, and added the given proguard rules to proguard-rules.pro.  Then I sync gradle, clean, and build.  I get the error `More than one file was found with OS independent path 'google/protobuf/descriptor.proto'.`.  See further below for a larger log.

**To Reproduce**
Do the steps given in the previous section.

**Expected behavior**
The app should build without error.

**Screenshots and logs**
Here's a log of `gradlew build`:
```
> Task :app:checkDebugDuplicateClasses FAILED
> Task :app:mergeDebugJavaResource FAILED

FAILURE: Build completed with 2 failures.

1: Task failed with an exception.
-----------
* What went wrong:
Execution failed for task ':app:checkDebugDuplicateClasses'.
> A failure occurred while executing com.android.build.gradle.internal.tasks.CheckDuplicatesRunnable
   > Duplicate class com.google.protobuf.DescriptorProtos found in modules jetified-polar-protobuf-release-runtime (polar-protobuf-release.aar) and jetified-protobuf-java-3.14.0 (com.google.protobuf:protobuf-java:3.14.0)
     Duplicate class com.google.protobuf.DescriptorProtos$1 found in modules jetified-polar-protobuf-release-runtime (polar-protobuf-release.aar) and jetified-protobuf-java-3.14.0 (com.google.protobuf:protobuf-java:3.14.0)
     Duplicate class com.google.protobuf.DescriptorProtos$DescriptorProto found in modules jetified-polar-protobuf-release-runtime (polar-protobuf-release.aar) and jetified-protobuf-java-3.14.0 (com.google.protobuf:protobuf-java:3.14.0)
     Duplicate class com.google.protobuf.DescriptorProtos$DescriptorProto$Builder found in modules jetified-polar-protobuf-release-runtime (polar-protobuf-release.aar) and jetified-protobuf-java-3.14.0 (com.google.protobuf:protobuf-java:3.14.0)
     Duplicate class com.google.protobuf.DescriptorProtos$DescriptorProto$ExtensionRange found in modules jetified-polar-protobuf-release-runtime (polar-protobuf-release.aar) and jetified-protobuf-java-3.14.0 (com.google.protobuf:protobuf-java:3.14.0)
     Duplicate class com.google.protobuf.DescriptorProtos$DescriptorProto$ExtensionRange$Builder found in modules jetified-polar-protobuf-release-runtime (polar-protobuf-release.aar) and jetified-protobuf-java-3.14.0 (com.google.protobuf:protobuf-java:3.14.0)
     Duplicate class com.google.protobuf.DescriptorProtos$DescriptorProto$ExtensionRangeOrBuilder found in modules jetified-polar-protobuf-release-runtime (polar-protobuf-release.aar) and jetified-protobuf-java-3.14.0 (com.google.protobuf:protobuf-java:3.14.0)
     Duplicate class com.google.protobuf.DescriptorProtos$DescriptorProto$ReservedRange found in modules jetified-polar-protobuf-release-runtime (polar-protobuf-release.aar) and jetified-protobuf-java-3.14.0 (com.google.protobuf:protobuf-java:3.14.0)
     Duplicate class com.google.protobuf.DescriptorProtos$DescriptorProto$ReservedRange$Builder found in modules jetified-polar-protobuf-release-runtime (polar-protobuf-release.aar) and jetified-protobuf-java-3.14.0 (com.google.protobuf:protobuf-java:3.14.0)
     Duplicate class com.google.protobuf.DescriptorProtos$DescriptorProto$ReservedRangeOrBuilder found in modules jetified-polar-protobuf-release-runtime (polar-protobuf-release.aar) and jetified-protobuf-java-3.14.0 (com.google.protobuf:protobuf-java:3.14.0)
     Duplicate class com.google.protobuf.DescriptorProtos$DescriptorProtoOrBuilder found in modules jetified-polar-protobuf-release-runtime (polar-protobuf-release.aar) and jetified-protobuf-java-3.14.0 (com.google.protobuf:protobuf-java:3.14.0)
     Duplicate class com.google.protobuf.DescriptorProtos$EnumDescriptorProto found in modules jetified-polar-protobuf-release-runtime (polar-protobuf-release.aar) and jetified-protobuf-java-3.14.0 (com.google.protobuf:protobuf-java:3.14.0)
     Duplicate class com.google.protobuf.DescriptorProtos$EnumDescriptorProto$Builder found in modules jetified-polar-protobuf-release-runtime (polar-protobuf-release.aar) and jetified-protobuf-java-3.14.0 (com.google.protobuf:protobuf-java:3.14.0)
     Duplicate class com.google.protobuf.DescriptorProtos$EnumDescriptorProto$EnumReservedRange found in modules jetified-polar-protobuf-release-runtime (polar-protobuf-release.aar) and jetified-protobuf-java-3.14.0 (com.google.protobuf:protobuf-java:3.14.0)
     Duplicate class com.google.protobuf.DescriptorProtos$EnumDescriptorProto$EnumReservedRange$Builder found in modules jetified-polar-protobuf-release-runtime (polar-protobuf-release.aar) and jetified-protobuf-java-3.14.0 (com.google.protobuf:protobuf-java:3.14.0)
     Duplicate class com.google.protobuf.DescriptorProtos$EnumDescriptorProto$EnumReservedRangeOrBuilder found in modules jetified-polar-protobuf-release-runtime (polar-protobuf-release.aar) and jetified-protobuf-java-3.14.0 (com.google.protobuf:protobuf-java:3.14.0)
     Duplicate class com.google.protobuf.DescriptorProtos$EnumDescriptorProtoOrBuilder found in modules jetified-polar-protobuf-release-runtime (polar-protobuf-release.aar) and jetified-protobuf-java-3.14.0 (com.google.protobuf:protobuf-java:3.14.0)
     Duplicate class com.google.protobuf.DescriptorProtos$EnumOptions found in modules jetified-polar-protobuf-release-runtime (polar-protobuf-release.aar) and jetified-protobuf-java-3.14.0 (com.google.protobuf:protobuf-java:3.14.0)
     Duplicate class com.google.protobuf.DescriptorProtos$EnumOptions$Builder found in modules jetified-polar-protobuf-release-runtime (polar-protobuf-release.aar) and jetified-protobuf-java-3.14.0 (com.google.protobuf:protobuf-java:3.14.0)
     Duplicate class com.google.protobuf.DescriptorProtos$EnumOptionsOrBuilder found in modules jetified-polar-protobuf-release-runtime (polar-protobuf-release.aar) and jetified-protobuf-java-3.14.0 (com.google.protobuf:protobuf-java:3.14.0)
     Duplicate class com.google.protobuf.DescriptorProtos$EnumValueDescriptorProto found in modules jetified-polar-protobuf-release-runtime (polar-protobuf-release.aar) and jetified-protobuf-java-3.14.0 (com.google.protobuf:protobuf-java:3.14.0)
     Duplicate class com.google.protobuf.DescriptorProtos$EnumValueDescriptorProto$Builder found in modules jetified-polar-protobuf-release-runtime (polar-protobuf-release.aar) and jetified-protobuf-java-3.14.0 (com.google.protobuf:protobuf-java:3.14.0)
     Duplicate class com.google.protobuf.DescriptorProtos$EnumValueDescriptorProtoOrBuilder found in modules jetified-polar-protobuf-release-runtime (polar-protobuf-release.aar) and jetified-protobuf-java-3.14.0 (com.google.protobuf:protobuf-java:3.14.0)
     Duplicate class com.google.protobuf.DescriptorProtos$EnumValueOptions found in modules jetified-polar-protobuf-release-runtime (polar-protobuf-release.aar) and jetified-protobuf-java-3.14.0 (com.google.protobuf:protobuf-java:3.14.0)
     Duplicate class com.google.protobuf.DescriptorProtos$EnumValueOptions$Builder found in modules jetified-polar-protobuf-release-runtime (polar-protobuf-release.aar) and jetified-protobuf-java-3.14.0 (com.google.protobuf:protobuf-java:3.14.0)
     Duplicate class com.google.protobuf.DescriptorProtos$EnumValueOptionsOrBuilder found in modules jetified-polar-protobuf-release-runtime (polar-protobuf-release.aar) and jetified-protobuf-java-3.14.0 (com.google.protobuf:protobuf-java:3.14.0)
     Duplicate class com.google.protobuf.DescriptorProtos$ExtensionRangeOptions found in modules jetified-polar-protobuf-release-runtime (polar-protobuf-release.aar) and jetified-protobuf-java-3.14.0 (com.google.protobuf:protobuf-java:3.14.0)
     Duplicate class com.google.protobuf.DescriptorProtos$ExtensionRangeOptions$Builder found in modules jetified-polar-protobuf-release-runtime (polar-protobuf-release.aar) and jetified-protobuf-java-3.14.0 (com.google.protobuf:protobuf-java:3.14.0)
     Duplicate class com.google.protobuf.DescriptorProtos$ExtensionRangeOptionsOrBuilder found in modules jetified-polar-protobuf-release-runtime (polar-protobuf-release.aar) and jetified-protobuf-java-3.14.0 (com.google.protobuf:protobuf-java:3.14.0)
     Duplicate class com.google.protobuf.DescriptorProtos$FieldDescriptorProto found in modules jetified-polar-protobuf-release-runtime (polar-protobuf-release.aar) and jetified-protobuf-java-3.14.0 ( [... cut ...]
   > More than one file was found with OS independent path 'google/protobuf/descriptor.proto'.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.
==============================================================================

* Get more help at https://help.gradle.org

BUILD FAILED in 1s
19 actionable tasks: 2 executed, 17 up-to-date
```

**Platform/Environment**
 - Device: Dell laptop
 - OS: Windows 10 Professional
Android Studio 4.1.1

Erhannis, 2021-05-06:
Additional note: I see a `google/protobuf/descriptor.proto` in both `polar-protobuf-release.aar` and `com.google.protobuf:protobuf-java:3.14.0`.  This may be the conflict the error describes.

Erhannis, 2021-05-07:
Additional note: the output of `gradlew --version` is:
```

------------------------------------------------------------
Gradle 6.5
------------------------------------------------------------

Build time:   2020-06-02 20:46:21 UTC
Revision:     a27f41e4ae5e8a41ab9b19f8dd6d86d7b384dad4

Kotlin:       1.3.72
Groovy:       2.5.11
Ant:          Apache Ant(TM) version 1.10.7 compiled on September 1 2019
JVM:          1.8.0_181 (Oracle Corporation 25.181-b13)
OS:           Windows 10 10.0 amd64
```

JOikarinen, 2021-05-07:
@Erhannis thank you for the finding. This must be related to the latest update where protobuf java was replaced with protobuf java-lite, the commit 5fd16b7e13032eeebdcb4500f1729631cb8cc3b9. I will investigate. 
Meanwhile you may work with the previous release version https://github.com/polarofficial/polar-ble-sdk/tree/3.0.5

JOikarinen, 2021-05-07:
@Erhannis the mistake is in README as it was not up to date. Now there is important change on README:
```
 implementation 'com.google.protobuf:protobuf-java:3.14.0'
```
changed to
```
implementation 'com.google.protobuf:protobuf-javalite:3.14.0'
```

Erhannis, 2021-05-07:
Great, that fixed it.  Thanks!

closed 2021-05-07 by Erhannis