polarofficial/polar-ble-sdk #158: PolarOH1 Sampling rate and PPG data unit

SiyuQian, 2021-05-14:
**Question concerns:**
- [x] Android SDK
- [ ] iOS SDK
- [x] Polar device
- [x] General

**Description:**
We are using the PolarOH1 device to produce the PPG and HR data during our research experiment.  Currently, we are using Neurokit2[https://neurokit2.readthedocs.io/en/latest/] library to calculate the HRV based on the PolarOH1 collected data. 

I have two questions:
1. Is there any way that I can make the sampling rate more stable? (as the watch is now collecting the data in a 120hz - 150hz range) 
2. What is the data unit of the PPG data? (I am sure if there are units for the PPG, not really familiar with this thing).But according to the sample data provided in the https://github.com/neuropsychology/NeuroKit/blob/master/data/bio_resting_5min_100hz.csv the PPG value is quite different with the ones which PolarOH1 produced. (I am using the PolarOH1 data diving 1000000 currently) ...

Thanks for your help.

JOikarinen, 2021-05-14:
> 1. Is there any way that I can make the sampling rate more stable? (as the watch is now collecting the data in a 120hz - 150hz range)
- Polar OH1 PPG sampling rate is 135Hz. If reading the sampling rate by [request stream settings](https://github.com/polarofficial/polar-ble-sdk/blob/034b414bbf1d7785f340f381760483a64586813f/sources/Android/android-communications/src/main/java/polar/com/sdk/api/PolarBleApi.java#L192) API the returned sampling rate is incorrectly said to be 130Hz. 
- You mention that sampling rate is varying between 120Hz-150Hz. One possible source of error is the moment when your app calculates the timestamp for each sample using the [timestamp](https://github.com/polarofficial/polar-ble-sdk/blob/034b414bbf1d7785f340f381760483a64586813f/sources/Android/android-communications/src/main/java/polar/com/sdk/api/model/PolarOhrData.java#L49) of received data packet. The [timestamp](https://github.com/polarofficial/polar-ble-sdk/blob/034b414bbf1d7785f340f381760483a64586813f/sources/Android/android-communications/src/main/java/polar/com/sdk/api/model/PolarOhrData.java#L49) given is the timestamp of the last sample in data packet. When calculating sample rate or timestamp for each sample, it needs to be taken into account that data packet size may vary.
- if the above is not the case, then I am interested to hear more details of your finding. In my own test I haven't be able to reproduce the problem. Do you have other streams enabled than PPG when you see the  sample rate changes between 120Hz-150Hz?

JOikarinen, 2021-05-14:
> 2\. What is the data unit of the PPG data? (I am sure if there are units for the PPG, not really familiar with this thing).But according to the sample data provided in the [...]

- PPG signal doesn't have agreed unit. It is raw value, each manufacturer having the own scale for the value. Because of autogain feature in PolarOH1 there can be jumps in PPG signal during measurements. That is discussed in previous issue, see the https://github.com/polarofficial/polar-ble-sdk/issues/152#issuecomment-827700658

SiyuQian, 2021-05-19:
> * Polar OH1 PPG sampling rate is 135Hz. If reading the sampling rate by [request stream [...]
@JOikarinen Thanks for your reply. I am going to reproduce the issue today. If I encouter the similiar issue then I will send you the data I collected from the PolarOH1 and so we might see what is happening to it.

SiyuQian, 2021-05-19:
Hi @JOikarinen 

[83D5852A_May,19_22-15-03.csv](https://github.com/polarofficial/polar-ble-sdk/files/6508756/83D5852A_May.19_22-15-03.csv)
This is a section of my whole experiment from the beginning. From what I understand, the sampling rate means the row of the data collected from the Polar device in a single second. You can see the first second got 127 rows and 138 for the second second from the CSV I provided.
Can you please help me out? Since the issue is causing an IndexError when I try to produce HRV value base on it, as I will need a sampling rate to pass in as a parameter for that function. (https://neurokit2.readthedocs.io/en/latest/_modules/neurokit2/hrv/hrv.html#hrv)

Thanks!

JOikarinen, 2021-05-20:
@SiyuQian I wonder what is the root cause of that problem you are facing. I have tried different conditions, but I always get steadily the 135Hz sampling for PPG. 

I added the logging to the code where I receive the PPG sample data and this is the result:
```
Packet #: 0 timestamp: 538660607946910208
Packet #: 0 samples size: 18
Packet #: 0 delta: NA
Packet #: 0 frequency: 135.09995437674542

Packet #: 1 timestamp: 538660608080144896
Packet #: 1 samples size: 18
Packet #: 1 delta: 7401927
Packet #: 1 frequency: 135.09995437674542
```
delta =  (packet1 timeStamp - packet0 timeStamp)/samples size = (538660608080144896ns-538660607946910208ns)/18 = 7401927ns 
frequency is (1/7401927ns ) x 1000 x 1000 x 1000 =  135Hz
...
```
Packet #: 1292 timestamp: 538660780057979392
Packet #: 1292 samples size: 18
Packet #: 1292 delta: 7400106
Packet #: 1292 frequency: 135.13319944335933

Packet #: 1293 timestamp: 538660780191181312
Packet #: 1293 samples size: 18
Packet #: 1293 delta: 7400106
Packet #: 1293 frequency: 135.13319944335933
```
delta =  (packet1293 timeStamp - packet1292 timeStamp)/samples size = (538660780191181312-538660780057979392)/18 = 7400106ns 
frequency is (1/7400106ns )x 1000 x 1000 x 1000 =  135Hz

SiyuQian, 2021-05-20:
@JOikarinen Ok..Thanks for your help. I found out that the random sampling rate issue caused by the ``for``` loop.

So basically, my code is very similar to this:
``` java
ppgDisposable = api.requestPpgSettings(DEVICE_ID).toFlowable().flatMap(new Function<PolarSensorSetting, Publisher<PolarOhrPPGData>>() {
                        @Override
                        public Publisher<PolarOhrPPGData> apply(PolarSensorSetting polarPPGSettings) throws Exception {
                            return api.startOhrPPGStreaming(DEVICE_ID,polarPPGSettings.maxSettings());
                        }
                    }).subscribe(
                        new Consumer<PolarOhrPPGData>() {
                            @Override
                            public void accept(PolarOhrPPGData polarOhrPPGData) throws Exception {
                                for( PolarOhrPPGData.PolarOhrPPGSample data : polarOhrPPGData.samples ){
                                    Log.d(TAG,"    ppg0: " + data.ppg0 + " ppg1: " + data.ppg1 + " ppg2: " + data.ppg2 + "ambient: " + data.ambient);
                                }
                            }
                        },
```
If I add the debug code you have provided inside the loop it would generate such results:
``` 
D/DEBUG: Packet #: 0 timestamp: 539553170493659136
D/DEBUG: Packet #: 0 samples size: 18
D/DEBUG: Packet #: 1 timestamp: 539553170493659136
D/DEBUG: Packet #: 1 samples size: 18
D/DEBUG: Packet #: 2 timestamp: 539553170493659136
D/DEBUG: Packet #: 2 samples size: 18
D/DEBUG: Packet #: 3 timestamp: 539553170493659136
D/DEBUG: Packet #: 3 samples size: 18
```
If I add the debug code outside of the loop, it would generate such results: (And after I remove the for loop, the sample rate become constant but the value is 225HZ, and it is weird, why the sampling rate becomes larger after I remove the loop ...)
``` 
D/DEBUG: Packet #: 0 timestamp: 539553042754033664
D/DEBUG: Packet #: 0 samples size: 18
D/DEBUG: Packet #: 1 timestamp: 539553042887202816
D/DEBUG: Packet #: 1 samples size: 18
```
Yes. Of couse it will act like that. So I think I did not understand why there is a loop in there and where should I add my custom actions (for exmaple, writeToCsv()). If I add the function inside the loop, the sample rate would become random. (120hz- 150hz)

If I add the function after the loop, how should I deal with the data inside the polarOhrPPGData.samples?

NOTE: Another difference is that I am using an Android service instead using the MainActivity.java. I am not sure if it is also related.

JOikarinen, 2021-05-21:
From the debug prints outside of for loop I get the sample rate of 135Hz. Here is my calculation:
delta = (packet0 timeStamp - packet1 timeStamp)/samples size = (539553042887202816-539553042754033664)/18 = 7398286ns
frequency is (1/7398286ns )x 1000 x 1000 x 1000 = 135Hz

The [timestamp](https://github.com/polarofficial/polar-ble-sdk/blob/034b414bbf1d7785f340f381760483a64586813f/sources/Android/android-communications/src/main/java/polar/com/sdk/api/model/PolarOhrData.java#L49) in PolarOhrData is the timestamp of the last sample in data packet. Last sample data timestamp is absolute (real value) and to get the timestamp for rest of the samples your code need to do some calculations. 

Option how to calculate: As the requested frequency is 135Hz (i.e. the time between samples is 0.0074s) you may calculate estimate for each sample once data packet is received. Last sample timestamp is known and the timestamp for each previous sample within the packet can estimated. 

Future thoughts:
- it is on my backlog to enhance the timestamp calculation. The SDK could do the work and calculate timestamp for each sample to makes API cleaner and easier to use.

SiyuQian, 2021-05-27:
> From the debug prints outside of for loop I get the sample rate of 135Hz. Here is my calculation:
> delta = (packet0 timeStamp - packet1 timeStamp)/samples size = [...]

[6D18B021_May,27_12-21-11.csv](https://github.com/polarofficial/polar-ble-sdk/files/6550300/6D18B021_May.27_12-21-11.csv)
So this is the CSV I get when I remove the ``` for ``` loop.  To be honest, I am a bit confused  .... Since the data was collected in ``` polarPPGData.samples ``` and currently, I am only getting the first sample (``` polarPPGDara.samples.get(1) ```) when the event triggered.

JOikarinen, 2021-05-27:
@SiyuQian, no worries, let's continue the discussion. I jump back to previous post by you:

> If I add the debug code outside of the loop, it would generate such results: (The sampling rate for each minute is now constant, 225 times / min)
> 
> ```
> D/DEBUG: Packet #: 0 timestamp: 539553042754033664
> D/DEBUG: Packet #: 0 samples size: 18
> D/DEBUG: Packet #: 1 timestamp: 539553042887202816
> D/DEBUG: Packet #: 1 samples size: 18
> ```

Above log output looks the correct output. You have received two data packets, both of them (packet 0 and packet 1) are containing 18 samples. So what you have is something like this if we write it out

```
Packet 0 Sample 0
Packet 0 Sample 1
Packet 0 Sample 2
Packet 0 Sample 3
Packet 0 Sample 4
Packet 0 Sample 5
Packet 0 Sample 6
Packet 0 Sample 7
Packet 0 Sample 8
Packet 0 Sample 8
Packet 0 Sample 8
Packet 0 Sample 8
Packet 0 Sample 8
```

JOikarinen, 2022-12-13:
Closing the issue. Please don't hesitate to reopen if needed.

closed 2022-12-13 by JOikarinen

aminhm2002, 2023-11-22:
I have the same problem 
Hi
I am working with the polar OH1.
I see the same problem with the sampling rate. I used your experiences but the quality of signal is really weak! it is 8 samples per cycle.
the picture of the signal is in below:
[image]

I do appreciate it if let me know how many peaks you receive in each second? and how many points you have in each PPG cycle?

Best Regards
Amin

https://github.com/polarofficial/polar-ble-sdk/issues/418