polarofficial/polar-ble-sdk #159: PolarOH1 Data Inconsistency

SiyuQian, 2021-05-20:
***Platform your question concerns:*** 
- [X] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [X] Polar OH1
- [ ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
Hi, I produced the following data by wearing two PolarOH1 on two arms but with one person.
[image]
[image]

I understand the value might not be that accurate and consistent but I wonder why the trend of both diagram could be quite different.

JOikarinen, 2021-05-21:
@SiyuQian two measurements are independent on each others and both OH1 devices may do they own decisions on level changes. The plot from two devices may look pretty different as both of the device experience different light conditions due the fact that both devices are positioned a bit differently against the skin. However if you filter out level changes then the two curve shall look pretty similar. 

Here is earlier discussion about the level change effect: https://docs.google.com/spreadsheets/d/10jjo0Nig1xKVJllF0RmCpWr0x6Xk_ONN7AQf7EcjKEc/edit?usp=drivesdk

@SiyuQian  Do you think this answer explains your observation?

nananaBananana, 2021-05-21:
@JOikarinen  I guess the link you provided is incorrect!

JOikarinen, 2021-05-22:
@nananaBananana thanks for pointing out.  Link is now pointing to discussion where the topic was earlier discussed.

SiyuQian, 2021-05-27:
> @SiyuQian two measurements are independent on each others and both OH1 devices may do they own decisions on level changes. The plot from two devices may look pretty different as both of the device experience different light conditions due the fact that both devices are positioned a bit differently against the skin. However if you filter out level changes then the two curve shall look pretty similar.
> 
> Here is earlier discussion about the level change effect: [#152 (comment)](https://github.com/polarofficial/polar-ble-sdk/issues/152#issuecomment-827700658)
> 
> @SiyuQian Do you think this answer explains your observation?

@JOikarinen Ok. Thanks.  One more question, how to filter out the level changes? In the previous issue I only found that you talked about to abandon the data collected from the first seconds.

JOikarinen, 2021-05-28:
@SiyuQian we discussed with the expert on Polar research team about the plot you shared. It is obvious that there is level changes on the plot, which can be filtered with step filter from the signal. Also there can be other ripple on the signal caused by movement of the patient wearing the OH1. The level changes are happening because of the PPG algorithm is changing the gain of signal to adapt for the light conditions seen by optical sensor. Conditions are changing because even small movement of the sensor causes light conditions to change. The light conditions are changing pretty dramatically if PPG signal is measured during some sports, like running. 

PPG signal is provided as raw data via Polar SDK. The purpose is that applications using Polar SDK may want to implement the signal manipulation and filtering from the raw signal. However, some intelligence is needed, if for example heart rate or PPI (pulse to pulse interval), is calculated from raw PPG signal. You may consider whether the heart rate or the PPI is matching requirements of your application. Both of mentioned are available from the Polar SDK too and then you save the time as you don't need to figure out PPG signal manipulation in your application.

SiyuQian, 2021-06-07:
@JOikarinen Unfortunately ... I reckon I have to use PPG data to calculate the HRV values and then I can detect the level of stress by analyzing the generated HRV values.

JOikarinen, 2021-06-08:
> I reckon I have to use **PPG data** to calculate the **HRV** values
- if the [HRV](https://www.polar.com/blog/heart-rate-variability-hrv) (heart rate variability) is your target could you use the PPI to calculate the HRV?

> .. Could you please provide more information about it(few links might helpful enough)?
- unfortunately the best help I can give is to explain phenomenons related on raw PPG signal. The reason is that the filtering of PPG signal is the core intellectual property owned by Polar.  The SDK provides output data like heart rate and PPI which are derivates from raw PPG signals calculated by Polar algorithms.

jimmyzumthurm, 2023-09-08:
@SiyuQian I'm closing this as this as not been updated for over 2+ years, feel free to re-open a new issue in case you have other questions.

closed 2023-09-08 by jimmyzumthurm