polarofficial/polar-ble-sdk #160: Having trouble calling startRecording API for Polar Verity Sense

jaesungj, 2021-05-20:
***Platform your question concerns:*** 
- [ ] Android 
- [V ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ V] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

I am trying to run through the sample example in this project via 
https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-ios/polarBleSdkTestApp
I am using the Polar Verity Sense, but it gives me unsupported error when I try to invoke "startRecording"  

recording start fail: PolarBleSdk.OperationNotSupported
I understand that this example is used for Polar H10 but in the API definition I don't see anything specific to H10

JOikarinen, 2021-05-21:
Hi @jaesungj 

> Q1) So I won't be able to use the api for it for a device other than h10? Do you possibly have any timeline for supporting it for other devices?
- I can confirm that only Polar H10 supports the recording start done by the SDK. At the moment of writing there is no plans to add the recording start/stop support to SDK for other Polar devices. For Polar OH1 and Polar Verity Sense the recording start can be done by the manual button found from the device. However, there is an extra step required for Polar OH1 and Polar Verity Sense to get recording working, those devices need to be first registered to Polar Ecosystem under some user to make recording possible. This extra requirement is making the recording use case in Polar OH1 and Polar Verity Sense a bit of impractical. 

> Q2) What does it mean by requires ftp to be ready in "request current recording status, only H10 supported, requires ftp to be ready? Is this saying that other devices are not supported because FTP is not ready or for us to use the startRecording, we need to have the ftp ready?
- SDK attempts to group api's related to each other together, the group of API's is the feature. The list of available features available are https://github.com/polarofficial/polar-ble-sdk/blob/e63c70aa713c6f13f6bf2022c7bad7c45ca3221e/sources/iOS/ios-communications/iOSCommunications/sdk/api/PolarBleApi.swift#L32
- when registering the SDK you may choose to support all the features or only the ones interested. Here is how it happens in example app https://github.com/polarofficial/polar-ble-sdk/blob/e63c70aa713c6f13f6bf2022c7bad7c45ca3221e/examples/example-ios/polarBleSdkTestApp/ViewController.swift#L17
- certain API's require the FTP feature is registered e.g. setLocalTime, startRecording, stopRecording, requestRecordingStatus etc. The need for FTP feature is mentioned in function documentation.

JOikarinen, 2021-05-21:
@jaesungj The API documentation is not too clear. In commit 7e61692 I tried to make it more explanatory.

jaesungj, 2021-05-21:
Thank you for addressing my questions so quickly!

closed 2021-05-21 by jaesungj