polarofficial/polar-ble-sdk #161: Feature request: H10 internal recording at 4Hz

jpcirrus, 2021-05-24:
Currently, with internal recording, HR is recorded at a maximum fidelity of 1Hz. This results in a loss of up to ~70% of HR data. If it was possible to record at 4Hz, this would still allow up to 7.5 hours of data to be stored (down from the current 30 hours) per training session. My use case is a research study which is collecting participants training HR data and comparing that to the various HRmax formalae, to determine at what percentage of HRmax masters athletes regulary train, and if the current HRmax formulae, and consequently training zones, need revision for older athletes.

JOikarinen, 2021-05-25:
@jpcirrus is it an option that you start recoding using `SampleType.RR`, then every heart beat gets recorded on Polar H10 memory and no data is lost.  

So you may [startRecording()](https://github.com/polarofficial/polar-ble-sdk/blob/7e616929cbbd516cf9014438b708afc3d32ca267/sources/Android/android-communications/src/main/java/polar/com/sdk/api/PolarBleApi.java#L237) using [SampleType.RR](https://github.com/polarofficial/polar-ble-sdk/blob/7e616929cbbd516cf9014438b708afc3d32ca267/sources/Android/android-communications/src/main/java/polar/com/sdk/api/PolarBleApi.java#L79). If using the [SampleType.RR](https://github.com/polarofficial/polar-ble-sdk/blob/7e616929cbbd516cf9014438b708afc3d32ca267/sources/Android/android-communications/src/main/java/polar/com/sdk/api/PolarBleApi.java#L79) then the [Recording interval](https://github.com/polarofficial/polar-ble-sdk/blob/7e616929cbbd516cf9014438b708afc3d32ca267/sources/Android/android-communications/src/main/java/polar/com/sdk/api/PolarBleApi.java#L59) does't have effect.

If using `SampleType.RR` for recording the size of memory used is dependant on the heart rate, higher the heart rate more data is recorded. If heart rate is steadily 150bpm then more than 9 hours of RR data can be recorded on H10 memory. Would that be enough for your use case?

jpcirrus, 2021-05-25:
@JOikarinen thanks very much for the fast response. Yes, that should work, and the memory is plenty for this study. Many thanks.

closed 2021-05-25 by jpcirrus

JOikarinen, 2021-05-25:
Updated the API documentation slightly. Now shall be more clear that if SampleType is RR then the interval parameter has no effect: 9626318b4208ac27400a274705e488bb3272ff94 and 3f042c0ca01232b953e88075cecc935b4127c4d5