polarofficial/polar-ble-sdk #163: Does the VERITY SENSE device have feature to change the sampling rate of PPG(s)?

S1S10, 2021-06-01:
Thank you very much for your support always. I would like to ask the following question.

In addition, the https://github.com/polarofficial/polar-ble-sdk/issues/149 Issue was very useful in asking this question.

***Platform your question concerns:*** 
- [ ] Android
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [x] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
I checked the sampling rate setting of PPG based on the following information.

https://github.com/polarofficial/polar-ble-sdk/blob/085bf6be0f8b19fa9d5222b9632768edfd9fa923/sources/iOS/ios-communications/iOSCommunications/sdk/api/PolarBleApi.swift#L228

The results was as follows.

`Check ppgSettings: PolarSensorSetting(settings: [PolarBleSdk.PolarSensorSetting.SettingType.channels: Set([4]), PolarBleSdk.PolarSensorSetting.SettingType.sampleRate: Set([135]), PolarBleSdk.PolarSensorSetting.SettingType.resolution: Set([22])])`

I thought that I should change this `PolarSensorSetting.SettingType.sampleRate: Set([135])` to 60Hz. Is this correct? Also, how should I change it?

Best regards,

JOikarinen, 2021-06-01:
@S1S10 the `requestStreamSettings` returns the values which are the accepted values  to start the stream.

Here is the snippet from [example app](https://github.com/polarofficial/polar-ble-sdk/blob/bd075478073dd323660ecd39adf0183953364d86/examples/example-ios/iosBleSdkTestApp/iosBleSdkTestApp/ViewModel.swift#L248) PPG stream start:

```
    func ppgToggle() {
        if ppgDisposable == nil, case .connected(let deviceId) = deviceConnectionState {
            ppgEnabled = true

           // 1)

            ppgDisposable = api.requestStreamSettings(deviceId, feature: DeviceStreamingFeature.ppg)
                .asObservable()
                .flatMap({ (settings) -> Observable<PolarOhrData> in

                    // 2)

                    return self.api.startOhrStreaming(self.deviceId, settings: settings.maxSettings())
                })
                .observe(on: MainScheduler.instance)
                .subscribe{ e in
                    switch e {
                    case .next(let data):
                        if(data.type == OhrDataType.ppg3_ambient1) {
                            for item in data.samples {
                                NSLog("    ppg0: \(item[0]) ppg1: \(item[1]) ppg2: \(item[2]) ambient: \(item[3])")
                            }
                        }
                    case .error(let err):
                        NSLog("PPG stream failed: \(err)")
                        self.ppgEnabled = false
                    case .completed:
                        NSLog("PPG stream completed")
                        self.ppgEnabled = false
                    }
                }
        } else {
            ppgEnabled = false
            ppgDisposable?.dispose()
            ppgDisposable = nil
        }
    }
```
1) The available stream settings are requested from the device
2) The example code is using the "maximum" settings available as the setting parameter for starting the PPG stream.

> If I want to set the sampling rate to 60 Hz, I thought that I should change this `PolarSensorSetting.SettingType.sampleRate: Set([135])` to 60Hz. Is this correct? Also, how should I change it?
- now with the pre explanation above I try to answer to your question. 
- Answer 1: you shall only pick the supported values returned by the `requestStreamSettings`.  In your question the returned value is 135Hz so that is the only choice for frequency. 
- Answer 2: what comes to Polar Verity Sense. Please see the latest [release 3.1.0](https://github.com/polarofficial/polar-ble-sdk/releases/tag/3.1.0) of Polar BLE SDK. There is now new mode available in Polar Verity Sense, so called SDK mode. Once the SDK mode is enabled the Polar Verity Sense offers Sampling rates 28Hz, 44Hz, 135Hz and 176Hz for PPG stream

S1S10, 2021-06-03:
Thank you for the answers.

Exactly, that your mentioned snippet is a good point. Then, I had to decipher the following source code.

https://github.com/polarofficial/polar-ble-sdk/blob/085bf6be0f8b19fa9d5222b9632768edfd9fa923/sources/iOS/ios-communications/iOSCommunications/sdk/api/model/PolarSensorSetting.swift#L56

```
    /// helper to retreive max settings available
    ///
    /// - Returns: PolarSensorSetting with max settings
    public func maxSettings() -> PolarSensorSetting {
        let selected = settings.reduce(into: [:]) { (result, arg1) in
            let (key, value) = arg1
            result[key] = value.max() ?? 0
        } as [SettingType : UInt32]
        return PolarSensorSetting(selected)
    }
```

I looked for a register specification to change the sample rate of PPG. However, I could not find it from the following document.

https://github.com/polarofficial/polar-ble-sdk/blob/085bf6be0f8b19fa9d5222b9632768edfd9fa923/technical_documentation/Polar_Measurement_Data_Specification.pdf

So, for the above source code, I tried to read and write what I thought was equivalent to registers, like the device driver of the firmware, by adding source code, but it did not work. It was the background for my question.

**<u>*Now, about your answer 1.*</u>**

>  In your question the returned value is 135Hz so that is the only choice for frequency.

Is it correct to say that the above returns a configurable value?

From the above source code, my understanding is that it is the maximum value that can be extracted from the register definition that matches the Key. But it seems to be my mistake.

**<u>*and, your answer 2.*</u>**

Thank you for the good news for me ;)

I will try the SDK mode. It's a pity that there is no 60Hz, but it looks strange if it is related to the clock division ratio.

JOikarinen, 2021-06-07:
> **_Now, about your answer 1._**
> 
> > In your question the returned value is 135Hz so that is the only choice for frequency.
> Is it correct to say that the above returns a configurable value?
> 
> From the above source code, my understanding is that it is the maximum value that can be extracted from the register definition that matches the Key. But it seems to be my mistake.

- let's take Polar Verity Sense as an example, if you read the settings by `api.requestStreamSettings(deviceId, feature: DeviceStreamingFeature.ppg)` you will get the settings available for the PPG stream, i.e. sample rate, resolution and channels. In case of the Polar Verity Sense PPG there is nothing you may choose from as you will get Sample Rate 135Hz, Resolution 22 and Channels 4. So in this case the `maxSettings()` is redundant as it will return Sample Rate 135Hz, Resolution 22 and Channels 4. However, the example uses the 'maxSettings()' because example shall work for all the scenarios, let say there is device which has more PPG stream settings available. 

And now with the firmware 1.1.5 available for Polar Verity Sense there is more options available for PPG stream. If you set the Polar Verity Sense to the SDK Mode then `api.requestStreamSettings(deviceId, feature: DeviceStreamingFeature.ppg)` actually returns the Sample Rate 28Hz, 44Hz, 135Hz and 176Hz  and Resolution 22 and Channels 4. With the maxSettings() returned values are Sample Rate 176Hz, Resolution 22 and Channels 4.  If that is what is desired then `maxSettings()` may be used. If you want other sample rates, then you may create instance of the `PolarSensorSetting` with you preferred settings, like Sample rate 44Hz, Resolution 22 and Channels 4.

merschel, 2021-09-10:
Hi

to change the sampling rate, SDK mode must be enabled in firmware 1.1.5. Will this function also be available for the normal mode and if so, can you estimate when that will be?

JOikarinen, 2021-09-11:
> to change the sampling rate, SDK mode must be enabled in firmware 1.1.5. Will this function also be available for the normal mode and if so, can you estimate when that will be?
- the PPG sampling is not possible to change in normal mode and for the moment there is no plans for making other sampling frequencies available.

jimmyzumthurm, 2023-09-08:
I'm closing this issue as the original question was answered. To further answer to @merschel , reason is that we have algorithms running that expects PPG at 55 Hz, and ACC + GYRO at 52 Hz. We could in theory configure ICs to higher ODR, and downsample before feeding to our algorithms so that Polar SDK can get higher sampling rates in normal mode, but this would require too much work and would need re-validation of our algorithms outputs. 

We have plans to allow **downsampling only** from normal mode for ACC + GYRO data types from 52 to 26/13 Hz for offline recordings, so that we can reduce the size of the files in memory to allow faster sync. This requires additional lowpass filtering to avoid aliasing distortion. We could not make that for PPG, as it would require to filter 4 more channels before decimation, and it is requiring too many resources for the device to have 10 simultaneous channel filtering ongoing, in term of memory and runtime execution. 

We chose ACC and GYRO over PPG, because we already provide highly accurate HR + PPI data, so there might be less demand for PPG raw data than for these 2, and we had to make some compromises. 

If you absolutely want lower sampling rate for PPG, I'm afraid SDK mode will be required until further notice.

closed 2023-09-08 by jimmyzumthurm