polarofficial/polar-ble-sdk #164: Polar H10 support simultaneous connections to two different processes?

sunshineinsandiego, 2021-06-04:
***Platform your question concerns:*** 
Mac OSX

***Device:***
Polar H10

**Description:**
Hi - I am developing with the Polar H10 using the BLEAK package for connections, and I'm wondering if the Polar H10 supports connections to two different processes at the same time? I'd like to record my polar data stream in one process and simultaneously visualize it. Thanks!

JOikarinen, 2021-06-07:
Hi @sunshineinsandiego,

it is possible to create two separate BLE connections with Polar H10. So if not mistaken, you shall be able to implement the scenario you explained. 

However, you need to enable 2 BLE connections option from Polar H10 using Polar Flow application.  Here are the quick steps how to:

**Step 1:** Install Polar Flow app on your mobile and register as a user.  Within the app navigate to Training -> Start tab and choose settings icon
[image]

**Step 2:** Pair with the H10
[image]

**Step 3:** Choose settings 
[image]

**Step 4:** Enable option "2 Bluetooth devices"
[image]

sunshineinsandiego, 2021-06-07:
Perfect, thank you! Appreciate the help.

closed 2021-06-07 by sunshineinsandiego