polarofficial/polar-ble-sdk #165: setLocalTime sets incorrect time and year

klandell, 2021-06-07:
**Platform on which you observed the bug:** 
- [ x ] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [ x  ] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
Calling `setLocalTime` does not properly set the timestamps in the acc/ppg/etc data.

**How to Reproduce**
See the following code:
```
@Override
public void polarFtpFeatureReady(@NonNull String s) {
    if (deviceId != null) {
        // set the local time for the device
        Calendar calendar = Calendar.getInstance(TimeZone.getTimeZone("UTC"));
        Log.d(TAG, "Now in ms: " + calendar.getTimeInMillis());
        polarApi.setLocalTime(deviceId, calendar)
                .subscribe(
                        () -> Log.d(TAG, "Time set to device"),
                        Throwable -> Log.d(TAG, "Set time failed: " + Throwable.toString()));
    }
}
```

And later...

```
polarAccelerometerData -> {
  Log.d(TAG, "ACC timeStamp: " + polarAccelerometerData.timeStamp);
},
```

Which produces the following log statements:
```
06-07 13:51:07.601 21479 21479 D POLAR   : Now in ms: 1623088267601
06-07 13:51:09.255 21479 22550 D POLAR   : Time set to device
06-07 13:51:13.253 21479 21479 D POLAR   : ACC timeStamp: 676389071436863759
```

`1623088267601` (ms timestamp) is Monday, June 7, 2021 5:51:07.601 PM **as expected**
`676389071436863759 ` (ns timestamp) is Saturday, June 8, 1991 1:51:11.436 PM **unexpected**

**Expected behavior**
I would expect the timestamp from the calendar and from the ACC data to reference the same date and _roughly_ the same time.

**Screenshots and logs**
(see above)

JOikarinen, 2021-06-08:
@klandell the explanation is that the "epoch" reference point is January 1, 2000 12:00:00 AM

> `676389071436863759 ` (ns timestamp) is Saturday, June 8, 1991 1:51:11.436 PM **unexpected**
- when "epoch" reference point is January 1, 2000 12:00:00 AM then `676389071436863759 ` converts to GMT: Monday, June 7, 2021 1:51:11.436 PM

JOikarinen, 2021-06-08:
there is an issue related to timestamp in Polar Verity Sense.

STEPS:
- start receive ACC stream and observe the timestamp:
    - observed time stamp is 536580371079656926
- set the time with SDK: time set to device Tue Jun 08 08:40:09 GMT+03:00 2021
- start receive ACC stream and observe the timestamp
    - observed time stamp is 536580557724485604 <- **NOK**
- power off and then power on
- start receive ACC stream and observe the timestamp
    - observed time stamp is 676456950432535404 <- **OK**

JOikarinen, 2021-06-08:
Ways to fix the epoch issue:
- Fix the docs: API documentation is not mentioning the epoch being 1.1.2000
- Fix the SDK: SDK can convert the received timestamp from the epoch 1.1.2000 to 1.1.1970
- Fix the sensor(s) software: Epoch reference point changed to 1.1.1970

klandell, 2021-06-08:
@JOikarinen Thank you for clarifying.  I will simply add 9.466848e17 to the timestamp for now then to convert to the 1.1.1970 epoch.  Thanks.

JOikarinen, 2021-06-09:
For now lets keep the epoch 1.1.2000

closed 2021-06-09 by JOikarinen