polarofficial/polar-ble-sdk #166: PPG sampling rate not as advertised in verity sense non SDK mode

labroskokkalas, 2021-06-15:
**Platform on which you observed the bug:** 
- [X] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [X] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
PPG sampling rate for Polar Verity Sense (firmware 1.1.5) is about 55 Hz,  while requestStreamSettings returns 135 Hz. When I enable SDK mode, and choose 135 Hz as sampling rate, then sampling rate is correct.

**How to Reproduce**
I used the example here https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-android/androidBleSdkTestApp, PPG data and calculated sampling rate 

**Expected behavior**
PPG sampling rate for non SDK mode should be 135 Hz as advertised by requestStreamSettings 

**Screenshots and logs**
If applicable, add screenshots or logs to help explain your problem.

JOikarinen, 2021-06-16:
@labroskokkalas your observation is correct, there is a bug in Polar Verity Sense firmware v.1.1.5. 

Added to [known issues](https://github.com/polarofficial/polar-ble-sdk/blob/master/technical_documentation/KnownIssues.md).

labroskokkalas, 2021-06-16:
@JOikarinen Thanks for the reply. I should have checked known issues first

jimmyzumthurm, 2023-09-08:
This is now fixed in 2.1.0 version