polarofficial/polar-ble-sdk #167: Acceleration data with gravity eliminated?

ma-c-bug-pro, 2021-06-17:
***Platform your question concerns:*** 
- [ ] Android 

***Device:***
- [ ] Polar Verity Sense
- [ ] Polar H10

**Description:**
Is there a way to extract clean Acceleration data from these two sensors, where gravity is eliminated from data?

sach16795, 2021-08-16:
Hello, I want to achieve the same result. Need acceleration data with gravity eliminated. Are there any updates on this? Thanks in advance.

jimmyzumthurm, 2023-09-08:
Hi, we currently do not provide a way of removing gravitational DC component from the accelerometer raw signal. Intention of the accelerometer raw data streaming is to provide reliable non-computed data read from our accelerometer and let the Polar SDK user interpret the data. 

For the reference, here is how the axis are defined for Verity Sense :  
[image]
And H10: 
[image]

closed 2023-09-08 by jimmyzumthurm