polarofficial/polar-ble-sdk #168: H10 RR recording interval

bartekhugo, 2021-06-29:
***Platform your question concerns:*** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
HI, Is there any effect when setting interval 1s vs 5s when recording RR intervals on Polar H10?
`polarApi.startRecording("device_id", exerciseId: "exercise_id", interval: .interval_5s, sampleType: .rr)` vs `polarApi.startRecording("device_id", exerciseId: "exercise_id", interval: .interval_1s, sampleType: .rr)`

JOikarinen, 2021-06-30:
Hi @bartekhugo 

> 1. Is there any effect when setting interval 1s vs 5s when recording RR intervals on Polar H10?
>    `polarApi.startRecording("device_id", exerciseId: "exercise_id", interval: .interval_5s, sampleType: .rr)` vs `polarApi.startRecording("device_id", exerciseId: "exercise_id", interval: .interval_1s, sampleType: .rr)`

If sampleType is .rr then the interval parameter has no effect as mentioned in [API documentation](https://github.com/polarofficial/polar-ble-sdk/blob/5f53bd19dcd5462ed4d5c14cacb206b40da1682c/sources/iOS/ios-communications/iOSCommunications/sdk/api/PolarBleApi.swift#L189). Only the RR intervals are saved into the Polar H10 memory.

> 2\. What is `PolarBleSdk.PolarExerciseEntry.date`? When I start recording, then take of the device so it turns off (and recording stops), and put it on again after more time the date seems to be a date of downloading the file and not the date the the device turning off (real session end). For me it doesn't make sense this way... 

Unfortunately the condition with the Polar H10 is that recoding start or end time is not known by SDK. The time in the `PolarBleSdk.PolarExerciseEntry.date` is the moment when exercise listing is done with function  [fetchStoredExerciseList](https://github.com/polarofficial/polar-ble-sdk/blob/5f53bd19dcd5462ed4d5c14cacb206b40da1682c/sources/iOS/ios-communications/iOSCommunications/sdk/api/PolarBleApi.swift#L221). I highly agree that implemented solution is a bit of suspicious. 

> To get the expected end date the only way is to manually save the start date and then sum up all the RR samples to get the end date correctly?

Exactly, that is the best way to get the start and end time for the recorded session.

JOikarinen, 2021-06-30:
> To get the expected end date the only way is to manually save the start date and then sum up all the RR samples to get the end date correctly?
- one possible approach is to give start time as exercise identifier (i.e. `exerciseId`) when starting the exercise with `func startRecording(_ identifier: String, exerciseId: String, interval: RecordingInterval, sampleType: SampleType) -> Completable`. By this approach you don't need to keep or save the start time locally.