polarofficial/polar-ble-sdk #169: Heart rate and RR interval metrics and relation to ECG data

seismann, 2021-07-02:
***Platform your question concerns:*** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
Is my understanding correct that the metrics of heart rate and RR interval are calculated based on the ECG data that is also provided i.e., given access to the algorithms, these metrics could be exactly reproduced from the ECG data?

JOikarinen, 2021-07-05:
Hi @seismann 
> i.e., given access to the algorithms, these metrics could be exactly reproduced from the ECG data?
- this statement is correct

seismann, 2021-07-05:
Thank you! Is the specific code that implements these algorithms publicly available?

JOikarinen, 2021-07-06:
Unfortunately it is not publicly available.

NiwreRelbag, 2022-01-24:
Hi, so the internal calculation (and therefore the accuracy of results) is based on the ECG data at 130Hz sampling rate. Correct?

JOikarinen, 2022-01-25:
@NiwreRelbag I understand your question. However I am not able to provide the concrete answer. ECG data is sampled by 130Hz when requested and streamed from Polar H10 sensor. However, I cannot give you exact details of how the H10 internal algorithm works when it calculates the heart rate from sensor data, e.g. what sampling rate it uses to measure ECG.

JOikarinen, 2022-01-25:
@seismann and @NiwreRelbag, I am closing this issue now, but don't hesitate to reopen for more questions.

closed 2022-01-25 by JOikarinen

NiwreRelbag, 2022-01-25:
For one of our apps, we have implemented an algorithm to calculate RR distances from the H10 ECG. The reason for this is that the RR distances computed by the H10 firmware are transmitted with a relatively high real-time delay (2-4 sec.) -  a serious disadvantage for the app. Therefore, it would be very important and interesting to know if the internal calculations in the H10 are based on a higher resolution signal.

This paper (https://www.polar.com/sites/default/files/static/science/white-papers/polar-h10-heart-rate-sensor-white-paper.pdf) says "H10 detects RR intervals with an accuracy of 2ms" and somewhere (not at Polar) I also read about "1KHz internal sampling".

Thanks!

JOikarinen, 2022-01-25:
> For one of our apps, we have implemented an algorithm to calculate RR distances from the H10 ECG. The reason for this is that the RR distances computed by the H10 firmware are transmitted with a relatively high real-time delay (2-4 sec.) - a serious disadvantage for the app.

I understand. I am a bit of surprised of such long delay as 2-4sec you mention. Let me copy a bit of the answer wrote in issue https://github.com/polarofficial/polar-ble-sdk/issues/227:

"With 130Hz sampling each sample is spaced ~7.69ms apart. The number of samples in one ECG data package depends on what kind of Bluetooth connection parameters are negotiated i.e. what is the Bluetooth data packet size used on BLE communication channel (in BLE terminology the data packet size on communication is called MTU, maximum data unit). The default MTU with H10 is 232 bytes and it results each ECG package contains 73 ECG samples. Roughly it means that one ECG package contains half second of sampled data (73samples/130Hz = 0.56s)."

That said, the oldest sample received by your application is at least 0.56s old. But your observation is that some samples are 2-4sec old, that observation means that there is some other and very reasonable delays from the moment ECG is sampled till the moment your application receives the sample. 
 
> Therefore, it would be very important and interesting to know if the internal calculations in the H10 are based on a higher resolution signal.
- ECG is sampled with 130Hz frequency when ECG stream is requested and streamed.

NiwreRelbag, 2022-01-25:
Sorry, I may have argued poorly. I saw the 2-4 second real time delay when I take the RR distances transmitted via the standard Bluetooth protocol for heart rate - so it is not related to the H10 SDK (ECG).

JOikarinen, 2022-01-25:
@NiwreRelbag, no problem, now I understand. Again little quote from the issue #227:
- The [PolarHrData](https://github.com/polarofficial/polar-ble-sdk/blob/38cb692dc9daf454325086de26425cd6d102a45c/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/model/PolarHrData.java#L10) given in [hrNotificationReceived](https://github.com/polarofficial/polar-ble-sdk/blob/38cb692dc9daf454325086de26425cd6d102a45c/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/PolarBleApiCallback.java#L115) callback is fired around once in a second. That data is coming from [BLE HR service](https://www.bluetooth.com/specifications/specs/heart-rate-service-1-0/), which is BLE standard service. BLE HR service specification says that heart value is sent around once a second. That is the reason why you may see multiple RR-values in [PolarHrData](https://github.com/polarofficial/polar-ble-sdk/blob/38cb692dc9daf454325086de26425cd6d102a45c/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/model/PolarHrData.java#L10). Whenever the heart is higher than 60bpm, then more than one RR value is received in [PolarHrData](https://github.com/polarofficial/polar-ble-sdk/blob/38cb692dc9daf454325086de26425cd6d102a45c/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/model/PolarHrData.java#L10) once in a while. When the heart rate is around 120bpm (RR is 500ms) then every [PolarHrData](https://github.com/polarofficial/polar-ble-sdk/blob/38cb692dc9daf454325086de26425cd6d102a45c/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/model/PolarHrData.java#L10) object start to list two RR values.  As a summary, all the detected RR-intervals are sent over, but not exactly at the moment of detection. 

I would say there is no realtime guarantee with standard BLE HR service. I believe the 2-4s delays are totally possible as you have observed. I suppose, the H10 internal sampling frequency doesn't play a big role in the delay. I mean even the internal sampling frequency is made higher the data is still sent over the BLE with same frequency and same delay. The delay is combination of the work H10 does internally, calculates heart rate from ECG, maybe does some time consuming filtering, waiting for next BLE sent moment and so on. I know don't the exact details of the firmware in H10, this is just my own speculation. I would expect similar delays on other manufacturer heart rate sensors too.

NiwreRelbag, 2022-01-25:
Thanks for the clarification.  "there is no real time guarantee with the standard BLE HR service": that's why we started analyzing the ECG data :-). But if the internal (firmware) calculation is able to provide higher quality results.... back to the core question :-).

 " I would expect similar delays on other manufacturer heart rate sensors too": In our experience, this depends heavily on the manufacturer. The delay with Polar (H9, H10) is relatively large - but ev. produces more stable results.

JOikarinen, 2022-01-25:
Okay, let's get back to your original question. 😊

> Hi, so the internal calculation (and therefore the accuracy of results) is based on the ECG data at 130Hz sampling rate. Correct?
- so you want to know if RR-values in [PolarHrData](https://github.com/polarofficial/polar-ble-sdk/blob/38cb692dc9daf454325086de26425cd6d102a45c/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/model/PolarHrData.java#L10) are more accurate compared what you can measure from streamed ECG data. I guess, you may find this document interesting: https://www.polar.com/sites/default/files/static/science/white-papers/polar-h10-heart-rate-sensor-white-paper.pdf. In that document there is information like this: 
```
H10 detects RR intervals within 2 ms accuracy
92.9 % in running,
99.3 % in cycling,
95.3 % in weight training,
95.6 % in all activities combined.
```

NiwreRelbag, 2022-01-25:
Yes I referred to this document 4 hours ago. 2ms would then indirectly imply a sampling rate of 500Hz?

JOikarinen, 2022-01-25:
> Yes I referred to this document 4 hours ago.
I am sorry, I totally ignored that. We would have same some time, but I was too excited to explain the aspects related to delay.

JOikarinen, 2022-01-25:
> 2ms would then indirectly imply a sampling rate of 500Hz?

This is the question which goes to the area of Polar expertise and is part the company know-how. Unfortunately, the document [referred]( https://www.polar.com/sites/default/files/static/science/white-papers/polar-h10-heart-rate-sensor-white-paper.pdf.) is the best I can tell.

NiwreRelbag, 2022-01-25:
Ok, thanks anyway