polarofficial/polar-ble-sdk #170: How to set dual-channel mode for H10 through SDK?

IanPeake, 2021-07-03:
***Platform your question concerns:*** 
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar H10

**Description:**
I have an application that effectively requires the H10 and dual-channel capability. How can I change this setting using the SDK? The user likely already has their own fitness device paired with the H10 and my app augments that with a further metric DFA alpha1. It's easy for the user to forget to enable dual-channel mode (in Polar Beat), in which case once an activity is started, the only way for the user to proceed is to abort the activity in order to start Polar Beat, since there is only one channel available.

JOikarinen, 2021-07-05:
@IanPeake thank you for the question, I understand your concern. Unfortunately, at the moment SDK is not supporting the dual channel enable (or disable). 
As you mentioned user can switch on the dual channel from the Polar Beat sensor settings and nowadays also from Polar Flow App, under device settings.

stuartlynne, 2022-08-24:
What is the setting to change via BLE to enable two BLE connections?

My application is not using the SDK, but coded directly in python using BLEAK. 

Looking at the polar-ble-sdk, there is mention of a .pfcRequestMultiConnectionSetting in the ios SDK file BlePfcClient.swift. 

Something to do with the PFC_SERVICE.

public enum PfcMessage: Int{
        /// configure hr broadcast setting, single byte parameter: 0 = off, 1 = on
        case pfcConfigureBroadcast = 1
        /// request current broadcast setting, payload as single byte with 0/1
        case pfcRequestBroadcastSetting = 2
        /// configure 5khz setting, single byte parameter: 0 = off, 1 = on
        case pfcConfigure5khz = 3
        /// request current 5khz setting, payload as single byte with 0/1
        case pfcRequest5khzSetting = 4
        /// configure adaptive tx power level (only in H10)
        case pfcConfigureAdaptiveTxPowerLevel = 5
        ///
        case pfcRequestAdaptiveTxPowerLevelSetting = 6
        ///
        case pfcConfigureBleMode = 7
        /// configure/enable multi connection = dual connection support in the device
        case pfcConfigureMultiConnection = 8 
        /// request current multi connection setting
        case pfcRequestMultiConnectionSetting = 9
        ///
        case pfcConfigureAntPlusSetting = 10
        ///
        case pfcRequestAntPlusSetting = 11

        public var description : String {
            switch self {
            case .pfcConfigureBroadcast:                     return "PFC_CONFIGURE_BROADCAST"
            case .pfcRequestBroadcastSetting:               return "PFC_REQUEST_BROADCAST_SETTING"
            case .pfcConfigure5khz:                          return "PFC_CONFIGURE_5KHZ"
            case .pfcRequest5khzSetting:                    return "PFC_REQUEST_5KHZ_SETTING"
            case .pfcConfigureAdaptiveTxPowerLevel:       return "PFC_CONFIGURE_ADAPTIVE_TX_POWER_LEVEL"
            case .pfcRequestAdaptiveTxPowerLevelSetting: return "PFC_REQUEST_ADAPTIVE_TX_POWER_LEVEL_SETTING"
            case .pfcConfigureBleMode: return "PFC_CONFIGURE_BLE_MODE"
            case .pfcConfigureMultiConnection:              return "PFC_CONFIGURE_MULTI_CONNECTION"
            case .pfcRequestMultiConnectionSetting:        return "PFC_REQUEST_MULTI_CONNECTION_SETTING"
            case .pfcConfigureAntPlusSetting:              return "PFC_CONFIGURE_ANT_PLUS_SETTING"
            case .pfcRequestAntPlusSetting:                return "PFC_REQUEST_ANT_PLUS_SETTING"
            }

    public static let PFC_SERVICE = CBUUID(string: "6217FF4B-FB31-1140-AD5A-A45545D7ECF3")
    let PFC_FEATURE               = CBUUID(string: "6217FF4C-C8EC-B1FB-1380-3AD986708E2D")
    public static let PFC_CP      = CBUUID(string: "6217FF4D-91BB-91D0-7E2A-7CD3BDA8A1F3")

rkangast, 2025-07-02:
This issue is old, therefore I am closing this issue. Please, feel free to create another issue if necessary.

closed 2025-07-02 by rkangast