polarofficial/polar-ble-sdk #173: Response Error 209

bartekhugo, 2021-07-14:
***Platform your question concerns:*** 
- [ ] Android 
- [x] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

I have a chain that 

- connects to a specific device
- reguests recording status
- stops recording (if recording)
- fetch exercise list
- fetch exercise (if exists)
- remove exercise (if exists)
- starts recording (RR intervals)
- disconnects from the device

It works fine for me but for one of the testers it fails within that chain. I dont have acces to physical devices to debug onsite and what I was able to get is that there is a "Response Error 209". Any idea what it means?

JOikarinen, 2021-07-15:
Error 209 is meaning "Battery too low". This error is probably returned on the moment of `start recoding` in your steps. 

I will change the label from the question to the bug as this could be fixed/enhanced so that textual description of the error reason is also returned together with error code.

jimmyzumthurm, 2023-09-08:
@samulimaa It seems that this is still not implemented, as recently someone complained about received error 202 : https://github.com/polarofficial/polar-ble-sdk/issues/374 , and it didn't seem to know it translates to SYSTEM_BUSY. Should we take actions about that ?

rkangast, 2025-07-02:
TC-63606

rkangast, 2025-09-09:
This has been fixed in release [6.6.0](https://github.com/polarofficial/polar-ble-sdk/tree/6.6.0)

closed 2025-09-09 by rkangast