polarofficial/polar-ble-sdk #174: Will  be there an adjustment in the near future for tge android example app?

franzda, 2021-07-15:
***Platform your question concerns:*** 
- [x ] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [ ] Polar Verity Sense
- [ x] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**

franzda, 2021-07-15:
Your example application is really great. But when the app  is translated with android studio there are several "unresolved references" and other gaps. 
Will there be an adjustment in the near future? Would be great!
greetings Franz

JOikarinen, 2021-07-15:
Hi @franzda, 

by "translating" you mean the building of the Example application? Could you point out the exact places in the code where you see unresolved references, lets fix them? Purpose is to keep example as clean and readable as possible. 

If I build the app from the terminal I can see the following lint warnings which will be cleaned in near future:
```
$ ./gradlew assembleDebug

> Task :app:compileDebugKotlin
w: androidBleSdkTestApp/app/src/main/java/com/polar/androidblesdk/MainActivity.kt: (99, 64): Type mismatch: inferred type is String? but String was expected
w: androidBleSdkTestApp/app/src/main/java/com/polar/androidblesdk/MainActivity.kt: (620, 62): 'getColor(Int): Int' is deprecated. Deprecated in Java
w: androidBleSdkTestApp/app/src/main/java/com/polar/androidblesdk/MainActivity.kt: (622, 62): 'getColor(Int): Int' is deprecated. Deprecated in Java
w: androidBleSdkTestApp/app/src/main/java/com/polar/androidblesdk/MainActivity.kt: (670, 19): Unchecked cast: Function<Pair<PolarSensorSetting, PolarSensorSetting>!, Flowable<PolarSensorSetting?>!> to Function<Pair<PolarSensorSetting, PolarSensorSetting>, Flowable<PolarSensorSetting>>
```

JOikarinen, 2021-07-15:
Or do you mean that example application is not using all the SDK API functions which are available in the API? There are some "gaps", as you told. The example application is not utilising all the possibilities from SDK, it is mostly giving the example of how to use main functionality. However, let's try to enhance the example all the time so that it will give enough information of the SDK to make start of SDK usage easier.

jimmyzumthurm, 2023-09-08:
@franzda As you didn't provide additional information, I will close this ticket. You could try with latest example app code to see if you can compile it.

closed 2023-09-08 by jimmyzumthurm