polarofficial/polar-ble-sdk #175: Scan Device

kumarankit-fujitsu, 2021-07-22:
***Platform your question concerns:*** 
- [ ] Android 

***Device:***
- [ ] Polar Verity Sense
- [ ] Other

**Description:**
I am using below code in my application to search polar device api.searchDevice().observeOn(AndroidSchedulers.mainThread()).subscribe(
                                polarDeviceInfo -> Log.d(TAG, "polar device found id: " + " isConnectable: " +polarDeviceInfo.isConnectable),
                                throwable -> Log.d(TAG, "" + throwable.getLocalizedMessage()),
                                () -> Log.d(TAG, "complete")
                        );

Case 1: When I am using Verity Sense
In this case, I get the result and polar device details

Case 2: When I use another device like polar unite
I don't get anything in the result. Basically, I want to handle the scenario where no compatible device was found after scanning.

Need help.

JOikarinen, 2021-07-23:
Hi @kumarankit-fujitsu,

I think the problem in the Case 2 is that Polar Unite is not doing BLE advertisement and for that reason the `api.searchDevice()` is not finding your Polar Unite. The Polar wrist units, like Polar Unite, have a bit different logic than the sensor category devices, like Polar Verity Sense. The Polar Unite is not advertising all the time and that is for battery saving. 

Your code snippet above looks similar what is found in [Polar SDK android example app](https://github.com/polarofficial/polar-ble-sdk/blob/0456ab9ea6b8f136a473d424f3641b18f407f363/examples/example-android/androidBleSdkTestApp/app/src/main/java/com/polar/androidblesdk/MainActivity.kt#L225). With these steps `api.searchDevice()` finds the Polar Unite:
1) When the Polar Unite is on watch mode (i.e. time of the day is shown on the screen), then do the long button press to get the Polar Unite to sync mode. 
2) When Polar Unite is on sync mode it starts to do BLE advertisement to make it self available
3) In [Polar SDK example app](https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-android/androidBleSdkTestApp) when pressing "SCAN DEVICES" I got to following log output:
```
2021-07-23 16:56:52.957 20532-20532/com.polar.androidblesdk D/MainActivity: polar device found id: 7338782C address: A0:9E:1A:73:38:78 rssi: -85 name: Polar Unite 7338782C isConnectable: true
```

> Basically, I want to handle the scenario where no compatible device was found after scanning.
- is the above explanation helping you with this?

kumarankit-fujitsu, 2021-08-05:
Hi, @JOikarinen Thanks for the help.
Need one more help.
I am getting error whenever I start GYRO Streaming
JavaBinder: *** Uncaught remote exception!  (Exceptions are not yet supported across processes.)
    java.lang.AssertionError: GYRO setting not stored

Can you please let me know the reason?

Thanks in Advance

JOikarinen, 2021-08-10:
> java.lang.AssertionError: GYRO setting not stored

@kumarankit-fujitsu by looking the SDK code I can see that assertion highlighted by you is possible. However, I am not able figure out the exact reason for getting the assert. I would be interested of what is the [PolarSensorSetting](https://github.com/polarofficial/polar-ble-sdk/blob/0456ab9ea6b8f136a473d424f3641b18f407f363/sources/Android/android-communications/src/sdk/java/com/polar/sdk/api/model/PolarSensorSetting.java#L13) parameter you pass for the [startGyroStreaming](https://github.com/polarofficial/polar-ble-sdk/blob/0456ab9ea6b8f136a473d424f3641b18f407f363/sources/Android/android-communications/src/sdk/java/com/polar/sdk/api/PolarBleApi.java#L460) method.

kumarankit-fujitsu, 2021-08-11:
@JOikarinen I am using the same code as it is in the given sample. The only difference is that I am using that code in the Service class and providing direct value for PolarSensorSetting parameters. Please look into my Service class code below.
```
class HRForegroundService : Service() {

    private val polarApi: PolarBleApi by lazy {
        PolarBleApiDefaultImpl.defaultImplementation(
            this, PolarBleApi.ALL_FEATURES
        )
    }
    private lateinit var notificationLayout: RemoteViews
    private var gyrDisposable: Disposable? = null

    companion object {
        private const val CHANNEL_ID = "HeartRate"
        private const val NOTIFICATION_ID = 101
        private const val ACTION_STOP = "STOP_ACTION"
        private const val ACTION_START = "START_ACTION"
        private var isServiceRunning: Boolean = false
        private lateinit var DEVICE_ID: String

        fun startHRService(mContext: Context, deviceID: String) {
            val serviceIntent = Intent(mContext, HRForegroundService::class.java)
            serviceIntent.action = ACTION_START
            isServiceRunning = true
            DEVICE_ID = deviceID
            ContextCompat.startForegroundService(mContext, serviceIntent)
        }

        fun stopHRService(mContext: Context) {
            val stopServiceIntent = Intent(mContext, HRForegroundService::class.java)
            stopServiceIntent.action = ACTION_STOP
            isServiceRunning = false
            mContext.stopService(stopServiceIntent)

        }
    }

    override fun onCreate() {
        super.onCreate()
        createNotificationChannel()
    }

    override fun onStartCommand(intent: Intent?, flags: Int, startId: Int): Int {
        when (intent?.action) {
            ACTION_START -> {
                val notification = getNotification()
                startForeground(NOTIFICATION_ID, notification)
                polarHeartRateSetup()
            }
            ACTION_STOP -> {
                stopForeground(true)
                stopSelf()
                if (DEVICE_ID.isNotEmpty()){
                    polarApi.disconnectFromDevice(DEVICE_ID)
                }
                gyrDisposable?.dispose()
                polarApi.shutDown()
            }
        }

        return START_NOT_STICKY
    }

    override fun onBind(intent: Intent?): IBinder? {
        return null
    }

    private fun getNotification(): Notification {
        val notificationIntent = Intent(this, SplashActivity::class.java)
        val pendingIntent = PendingIntent.getActivity(
[... 163 lines cut ...]
            )
    }

    private fun gyroSetting(
        available: Map<PolarSensorSetting.SettingType, Set<Int>>,
        all: Map<PolarSensorSetting.SettingType, Set<Int>>
    ): Single<PolarSensorSetting?> {

        return Single.create { e: SingleEmitter<PolarSensorSetting?> ->
            val selected: MutableMap<PolarSensorSetting.SettingType, Int> = EnumMap(
                PolarSensorSetting.SettingType::class.java
            )

            selected[PolarSensorSetting.SettingType.SAMPLE_RATE] = 52
            selected[PolarSensorSetting.SettingType.RESOLUTION] = 16
            selected[PolarSensorSetting.SettingType.RANGE] = 2000
            selected[PolarSensorSetting.SettingType.CHANNELS] = 3
            e.onSuccess(PolarSensorSetting(selected))
        }
    }

}
```

NOTE: When I use the same code inside a fragment or activity, then it is working fine as expected.

JOikarinen, 2021-08-12:
Hi @kumarankit-fujitsu, for now I don't spot the possible problem in your code, I continue investigating what could be the issue when using the SDK from android service.

JOikarinen, 2021-08-12:
@kumarankit-fujitsu one thing to double check is the restriction by Android OS related to services requiring location. Polar SDK requires the location for any Bluetooth operation. Here is the Google's guide: 
https://developer.android.com/guide/components/foreground-services#types

`
If your app targets Android 10 (API level 29) or higher and accesses location information in a foreground service, declare the location foreground service type as an attribute of your <service> component.
`

kumarankit-fujitsu, 2021-08-12:
Thanks, @JOikarinen I will check and get back to you.

kumarankit-fujitsu, 2021-08-12:
@JOikarinen Same error after applying location foreground service type as an attribute of <service> component. 
`<service
            android:name=".ui.views.service.HRForegroundService"
            android:enabled="true"
            android:foregroundServiceType="location"/>`

kumarankit-fujitsu, 2021-08-12:
Hi @JOikarinen I have tried with Android 8 (API level 27). Found same error. 
Only for few seconds I got gyro data and then app get crashed. Below are the logs, I am sharing with you. It might help you to understand the reason for this issue.

```
2021-08-12 18:10:32.214 8038-8038/com.fujitsu.dev D/Service: GYR    x: 1.05 y:  1.89 z: -1.4
2021-08-12 18:10:32.215 8038-8038/com.fujitsu.dev D/Service: GYR    x: 0.98 y:  1.47 z: -1.26
2021-08-12 18:10:32.215 8038-8038/com.fujitsu.dev D/Service: GYR    x: 0.91 y:  1.26 z: -1.19
2021-08-12 18:10:32.215 8038-8038/com.fujitsu.dev D/Service: GYR    x: 0.7 y:  1.4 z: -1.26
2021-08-12 18:10:32.216 8038-8038/com.fujitsu.dev D/Service: GYR    x: 0.63 y:  1.4 z: -1.33
2021-08-12 18:10:32.216 8038-8038/com.fujitsu.dev D/Service: GYR    x: 0.91 y:  1.05 z: -1.19
2021-08-12 18:10:32.217 8038-8038/com.fujitsu.dev D/Service: GYR    x: 1.19 y:  0.91 z: -1.19
2021-08-12 18:10:32.217 8038-8038/com.fujitsu.dev D/Service: GYR    x: 1.05 y:  0.42000002 z: -1.19
2021-08-12 18:10:32.226 8038-8055/com.fujitsu.dev E/JavaBinder: *** Uncaught remote exception!  (Exceptions are not yet supported across processes.)
    java.lang.AssertionError: GYRO setting not stored
        at com.polar.androidcommunications.common.ble.BleUtils.validate(BleUtils.java:110)
        at com.polar.androidcommunications.api.ble.model.gatt.client.BlePMDClient.fetchFactor(BlePMDClient.java:871)
        at com.polar.androidcommunications.api.ble.model.gatt.client.BlePMDClient.processServiceData(BlePMDClient.java:992)
        at com.polar.androidcommunications.enpoints.ble.bluedroid.host.BDDeviceSessionImpl.handleCharacteristicValueUpdated(BDDeviceSessionImpl.java:559)
        at com.polar.androidcommunications.enpoints.ble.bluedroid.host.BDGattCallback.onCharacteristicChanged(BDGattCallback.java:133)
        at android.bluetooth.BluetoothGatt$1$8.run(BluetoothGatt.java:443)
        at android.bluetooth.BluetoothGatt.runOrQueueCallback(BluetoothGatt.java:725)
        at android.bluetooth.BluetoothGatt.-wrap0(Unknown Source:0)
        at android.bluetooth.BluetoothGatt$1.onNotify(BluetoothGatt.java:437)
        at android.bluetooth.IBluetoothGattCallback$Stub.onTransact(IBluetoothGattCallback.java:184)
        at android.os.Binder.execTransact(Binder.java:697)
```

JOikarinen, 2021-08-13:
Hi @kumarankit-fujitsu, I have replicated your service setup as close as I can by copying your `HRForegroundService` code found in your previous comment. In my experiment the code is working perfectly and gyro stream data is received. I get stream of gyro data and produced log messages are looks like this: `D/HRForegroundService: GYR    x: -1.61 y:  0.56 z: 0.98`. 

So there is some aspect which is different in our setups. I would be interested of what kind of logs are produced by Polar BLE SDK once you run your code. Is it possible you setup the logging for Polar BLE SDK and provide logs? You may enable the logging by placing `polarApi.setApiLogger { s: String? -> Log.d("API_LOGGER", s!!) }` to your service `onStartCommand` function

JOikarinen, 2021-08-13:
Another thing which we could double check is the used device and its firmware. May you override the  `disInformationReceived` for `PolarBleApiCallback` 

```
override fun disInformationReceived(identifier: String, uuid: UUID, value: String) {
     Log.d("DisInformation", "DIS information $value")
}
```

kumarankit-fujitsu, 2021-08-13:
@JOikarinen Here are the logs that I got using `polarApi.setApiLogger { s: String? -> Log.d("API_LOGGER", s!!) }` in onStartCommand() function of Service Class.

```
`2021-08-13 11:19:06.535 13147-13147/com.fujitsu.dev D/API_LOGGER: BDScanCallback/commandState state:IDLE action: CLIENT_START_SCAN
2021-08-13 11:19:06.536 13147-13147/com.fujitsu.dev D/API_LOGGER: BDScanCallback/commandState state:IDLE action: EXIT
2021-08-13 11:19:06.536 13147-13147/com.fujitsu.dev D/API_LOGGER: BDScanCallback/commandState state:SCANNING action: ENTRY
2021-08-13 11:19:06.536 13147-13147/com.fujitsu.dev D/API_LOGGER: BDScanCallback/timestamps left: 0
2021-08-13 11:19:06.536 13147-13147/com.fujitsu.dev D/API_LOGGER: BDScanCallback/Scan started -->
2021-08-13 11:19:06.561 13147-13147/com.fujitsu.dev D/API_LOGGER: BDScanCallback/Scan started <--
2021-08-13 11:19:06.786 13147-13644/com.fujitsu.dev D/API_LOGGER: BDGattCallback/ phy updated tx: 2 rx: 2 status: 0
2021-08-13 11:19:07.076 13147-13147/com.fujitsu.dev D/API_LOGGER: BleGattBase/Added notification characteristic for fb005c51-02e7-f387-1cad-8acd2d8df0c8
2021-08-13 11:19:07.077 13147-13147/com.fujitsu.dev D/API_LOGGER: BleGattBase/Added notification characteristic for fb005c52-02e7-f387-1cad-8acd2d8df0c8
2021-08-13 11:19:07.078 13147-13147/com.fujitsu.dev D/API_LOGGER: BleGattBase/Added notification characteristic for fb005c81-02e7-f387-1cad-8acd2d8df0c8
2021-08-13 11:19:07.079 13147-13147/com.fujitsu.dev D/API_LOGGER: BleGattBase/Added notification characteristic for fb005c82-02e7-f387-1cad-8acd2d8df0c8
2021-08-13 11:19:07.080 13147-13147/com.fujitsu.dev D/API_LOGGER: BleGattBase/Added notification characteristic for 00002a19-0000-1000-8000-00805f9b34fb
2021-08-13 11:19:07.081 13147-13147/com.fujitsu.dev D/API_LOGGER: BDDeviceListenerImpl/new device allocated name: Polar Sense 923EE226
2021-08-13 11:19:07.081 13147-13147/com.fujitsu.dev D/API_LOGGER: BDDeviceList/new session added: Polar Sense 923EE226
2021-08-13 11:19:07.716 13147-13644/com.fujitsu.dev D/API_LOGGER: BDDeviceSessionImpl/ SERVICE: 00001800-0000-1000-8000-00805f9b34fb
2021-08-13 11:19:07.716 13147-13644/com.fujitsu.dev D/API_LOGGER: BDDeviceSessionImpl/No client found for SERVICE: 00001800-0000-1000-8000-00805f9b34fb chrs: 4
2021-08-13 11:19:07.717 13147-13644/com.fujitsu.dev D/API_LOGGER: BDDeviceSessionImpl/ SERVICE: 00001801-0000-1000-8000-00805f9b34fb
2021-08-13 11:19:07.717 13147-13644/com.fujitsu.dev D/API_LOGGER: BDDeviceSessionImpl/No client found for SERVICE: 00001801-0000-1000-8000-00805f9b34fb chrs: 1
2021-08-13 11:19:07.718 13147-13644/com.fujitsu.dev D/API_LOGGER: BDDeviceSessionImpl/ SERVICE: 0000180a-0000-1000-8000-00805f9b34fb
2021-08-13 11:19:07.718 13147-13644/com.fujitsu.dev D/API_LOGGER: BDDeviceSessionImpl/     CHARACTERISTIC: 00002a29-0000-1000-8000-00805f9b34fb PROPERTIES: 2
2021-08-13 11:19:07.720 13147-13644/com.fujitsu.dev D/API_LOGGER: BDDeviceSessionImpl/     CHARACTERISTIC: 00002a24-0000-1000-8000-00805f9b34fb PROPERTIES: 2
2021-08-13 11:19:07.721 13147-13644/com.fujitsu.dev D/API_LOGGER: BDDeviceSessionImpl/     CHARACTERISTIC: 00002a25-0000-1000-8000-00805f9b34fb PROPERTIES: 2
2021-08-13 11:19:07.722 13147-13644/com.fujitsu.dev D/API_LOGGER: BDDeviceSessionImpl/     CHARACTERISTIC: 00002a27-0000-1000-8000-00805f9b34fb PROPERTIES: 2
2021-08-13 11:19:07.723 13147-13644/com.fujitsu.dev D/API_LOGGER: BDDeviceSessionImpl/     CHARACTERISTIC: 00002a26-0000-1000-8000-00805f9b34fb PROPERTIES: 2
2021-08-13 11:19:07.723 13147-13644/com.fujitsu.dev D/API_LOGGER: BDDeviceSessionImpl/     CHARACTERISTIC: 00002a28-0000-1000-8000-00805f9b34fb PROPERTIES: 2
2021-08-13 11:19:07.724 13147-13644/com.fujitsu.dev D/API_LOGGER: BDDeviceSessionImpl/     CHARACTERISTIC: 00002a23-0000-1000-8000-00805f9b34fb PROPERTIES: 2
2021-08-13 11:19:07.725 13147-13644/com.fujitsu.dev D/API_LOGGER: BDDeviceSessionImpl/ SERVICE: 0000feee-0000-1000-8000-00805f9b34fb
2021-08-13 11:19:07.725 13147-13644/com.fujitsu.dev D/API_LOGGER: BDDeviceSessionImpl/     CHARACTERISTIC: fb005c51-02e7-f387-1cad-8acd2d8df0c8 PROPERTIES: 28
2021-08-13 11:19:07.725 13147-13644/com.fujitsu.dev D/API_LOGGER: BDDeviceSessionImpl/     CHARACTERISTIC: fb005c52-02e7-f387-1cad-8acd2d8df0c8 PROPERTIES: 16
2021-08-13 11:19:07.726 13147-13644/com.fujitsu.dev D/API_LOGGER: BDDeviceSessionImpl/     CHARACTERISTIC: fb005c53-02e7-f387-1cad-8acd2d8df0c8 PROPERTIES: 12
2021-08-13 11:19:07.726 13147-13644/com.fujitsu.dev D/API_LOGGER: BDDeviceSessionImpl/ SERVICE: 0000180d-0000-1000-8000-00805f9b34fb
2021-08-13 11:19:07.727 13147-13644/com.fujitsu.dev D/API_LOGGER: BDDeviceSessionImpl/     CHARACTERISTIC: 00002a37-0000-1000-8000-00805f9b34fb PROPERTIES: 16
2021-08-13 11:19:07.727 13147-13644/com.fujitsu.dev D/API_LOGGER: BDDeviceSessionImpl/ SERVICE: 0000180f-0000-1000-8000-00805f9b34fb
2021-08-13 11:19:07.728 13147-13644/com.fujitsu.dev D/API_LOGGER: BDDeviceSessionImpl/     CHARACTERISTIC: 00002a19-0000-1000-8000-00805f9b34fb PROPERTIES: 18
2021-08-13 11:19:07.728 13147-13644/com.fujitsu.dev D/API_LOGGER: BDDeviceSessionImpl/ SERVICE: fb005c20-02e7-f387-1cad-8acd2d8df0c8
2021-08-13 11:19:07.729 13147-13644/com.fujitsu.dev D/API_LOGGER: BDDeviceSessionImpl/No client found for SERVICE: fb005c20-02e7-f387-1cad-8acd2d8df0c8 chrs: 3
2021-08-13 11:19:07.729 13147-13644/com.fujitsu.dev D/API_LOGGER: BDDeviceSessionImpl/ SERVICE: 6217ff4b-fb31-1140-ad5a-a45545d7ecf3
[... cut ...]
2021-08-13 11:19:10.751 13147-13147/com.fujitsu.dev D/DisInformation: DIS information 1.1.5
2021-08-13 11:19:10.838 13147-13147/com.fujitsu.dev D/DisInformation: DIS information 0.1.5
2021-08-13 11:19:10.898 13147-13147/com.fujitsu.dev D/DisInformation: DIS information 00784292.02
2021-08-13 11:19:10.959 13147-13147/com.fujitsu.dev D/DisInformation: DIS information 923EE226��
2021-08-13 11:19:11.084 13147-13147/com.fujitsu.dev D/DisInformation: DIS information INW4J
2021-08-13 11:19:11.146 13147-13147/com.fujitsu.dev D/DisInformation: DIS information Polar Electro Oy��`
```

kumarankit-fujitsu, 2021-08-13:
@JOikarinen May I know, are you using any other permission other than I am using:
1. In Fragment
 ```
private fun requestForBluetoothAndLocation() {
        val mBluetoothAdapter = BluetoothAdapter.getDefaultAdapter()
        if (mBluetoothAdapter != null && !mBluetoothAdapter.isEnabled) {
            val enableBtIntent = Intent(BluetoothAdapter.ACTION_REQUEST_ENABLE)
            startActivityForResult(enableBtIntent, 2)
        }

        //requestPermissions() method needs to be called when the build SDK version is 23 or above
        if (Build.VERSION.SDK_INT >= 23) {
            requestPermissions(
                arrayOf(
                    Manifest.permission.ACCESS_COARSE_LOCATION,
                    Manifest.permission.ACCESS_FINE_LOCATION
                ), 1
            )
        }
    }
```
2. In Manifest 

```
<uses-permission android:name="android.permission.BLUETOOTH" />
    <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION" />
    <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION" />
    <uses-permission android:name="android.permission.FOREGROUND_SERVICE" />

```
```
 <service
            android:name=".ui.views.service.HRForegroundService"
            android:enabled="true"
            android:foregroundServiceType="location"/>
```

If I am missing any permission, then please let me know.
Many thanks.

kumarankit-fujitsu, 2021-08-13:
Hi, @JOikarinen As you said, "_In my experiment, the code is working perfectly and gyro stream data is received_".
Can you please share that working experimented code on Github with me. So that I can compare my code with yours and try to figure out the exact reason for crashing.
It will be a great help. 
Thank You.

JOikarinen, 2021-08-13:
@kumarankit-fujitsu thank you for the logs. I think the logs reveals the cause but maybe not reason yet. From the logs I can see there is two sessions created for the one device. Both of the sessions do operations for the device and as a consequence there is assert on other session. 

- session 1 is running in process id 13147-13644
```
2021-08-13 11:19:07.716 13147-13644/com.fujitsu.dev D/API_LOGGER: BDDeviceSessionImpl/ SERVICE: 00001800-0000-1000-8000-00805f9b34fb
```
- session 2  is running in process id 13147-15439
```
2021-08-13 11:19:09.910 13147-15439/com.fujitsu.dev D/API_LOGGER: BDDeviceSessionImpl/ SERVICE: 00001800-0000-1000-8000-00805f9b34fb
```

But why there are two sessions, is now a bit of mystery. Here are the first thoughts you could double check:
- in your phone bluetooth settings, remove the known Polar Sense devices and try again
- is there any change in your code, that `HRForegroundService` started twice, so that there would be two service instances running parallel?

kumarankit-fujitsu, 2021-08-13:
@JOikarinen Yes, there are two sessions purposely. Both sessions started on two different objects of PolarBleApi. One In the fragment when the user using the application and the other when the user closes the application, not using directly( in the Service)

JOikarinen, 2021-08-13:
@kumarankit-fujitsu I suppose that's the reason for the problem. The two instances of the PolarBleApi are now interfering each others. You shall refactor your code so that you have only one instance of PolarBleApi. I will have a look on documentation where would is the best place for highlighting that there shall be on instance created of PolarBleApi.

JOikarinen, 2021-08-16:
@kumarankit-fujitsu after some thinking I couldn't think of the situation that multiple instances of the PolarBleApi would be needed by client application. For the future proof,  in #175 the singleton pattern is taken into use to prevent the multiple instantiation of PolarBleApi.

closed 2021-08-16 by JOikarinen

kumarankit-fujitsu, 2021-08-17:
@JOikarinen Thanks a lot for the help. As you said, I refactor my code and now keeping only one instance of the PolarBleApi. Now it is working fine. 
Two questions I would like to ask you. 
1. I am using PPIStreaming and GyroStreaming both in service. Sometimes I get the GYRO setting is not available error. Below is the code snippet
```
` override fun streamingFeaturesReady(
                identifier: String,
                features: MutableSet<PolarBleApi.DeviceStreamingFeature>
            ) {
                for (feature in features) {
                    when (feature) {
                        PolarBleApi.DeviceStreamingFeature.GYRO -> {
                            if (DEVICE_ID.isNotEmpty()) {
                                startGyroStreaming()
                            }
                        }
                        PolarBleApi.DeviceStreamingFeature.PPI -> {
                            if (DEVICE_ID.isNotEmpty()) {
                                startHRPPIStreaming()
                            }
                        }
                        else -> {
                            //do nothing
                        }
                    }
                }

            }`
```
 The reason may be both calls get started simultaneously?
2. I am converting the HR value that I get from PPIStreaming into BPM and the formula I am using is 60*1000/PpiMs. Is it the correct way to convert because sometimes I get a 21 bpm converted value?

Thank You

JOikarinen, 2021-08-18:
> 1. I am using PPIStreaming and GyroStreaming both in service. Sometimes I get the GYRO setting is not available error. Below is the code snippet
- interesting problem. It sounds pretty much of timing issue. If you comment out the `startHRPPIStreaming()` do you still get the " GYRO setting is not available error" sometimes?

> 2. I am converting the HR value that I get from PPIStreaming into BPM and the formula I am using is 60*1000/PpiMs. Is it the correct way to convert because sometimes I get a 21 bpm converted value?
- there is [HR](https://github.com/polarofficial/polar-ble-sdk/blob/dc29a7cf0d17bf1a012bb1708e890a0d2e5c61f6/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/model/PolarOhrPPIData.java#L32) value provided in [PolarOhrPPISample](https://github.com/polarofficial/polar-ble-sdk/blob/dc29a7cf0d17bf1a012bb1708e890a0d2e5c61f6/sources/Android/android-communications/library/src/sdk/java/com/polar/sdk/api/model/PolarOhrPPIData.java#L14). However if you want to calculate it yourself the formula `60*1000/PpiMs` is correct

kumarankit-fujitsu, 2021-08-18:
@JOikarinen 

- After some testing, I didn't find " GYRO setting is not available error" when I commented out the startHRPPIStreaming()
- The HR value in PolarOhrPPISample is coming 0 sometimes. Please see below logs
- 2021-08-18 11:48:02.190 30795-30795/com.fujitsu.polargyro D/MainActivity: PPI    ppi: 713 hr: 0 blocker: false errorEstimate: 35
2021-08-18 11:48:02.190 30795-30795/com.fujitsu.polargyro D/MainActivity: PPI    ppi: 659 hr: 0 blocker: true errorEstimate: 39
2021-08-18 11:48:02.190 30795-30795/com.fujitsu.polargyro D/MainActivity: PPI    ppi: 651 hr: 0 blocker: true errorEstimate: 40
2021-08-18 11:48:02.190 30795-30795/com.fujitsu.polargyro D/MainActivity: PPI    ppi: 713 hr: 0 blocker: true errorEstimate: 34
2021-08-18 11:48:02.190 30795-30795/com.fujitsu.polargyro D/MainActivity: PPI    ppi: 616 hr: 0 blocker: true errorEstimate: 32
2021-08-18 11:48:02.206 30795-30795/com.fujitsu.polargyro D/MainActivity: PPI    ppi: 617 hr: 0 blocker: true errorEstimate: 28
2021-08-18 11:48:02.206 30795-30795/com.fujitsu.polargyro D/MainActivity: PPI    ppi: 680 hr: 0 blocker: true errorEstimate: 24
2021-08-18 11:48:02.206 30795-30795/com.fujitsu.polargyro D/MainActivity: PPI    ppi: 617 hr: 0 blocker: true errorEstimate: 21
2021-08-18 11:48:02.206 30795-30795/com.fujitsu.polargyro D/MainActivity: PPI    ppi: 684 hr: 0 blocker: true errorEstimate: 18
2021-08-18 11:48:02.435 30795-30795/com.fujitsu.polargyro D/MainActivity: HR value: 0 rrsMs: [] rr: [] contact: false , false
2021-08-18 11:48:03.422 30795-30795/com.fujitsu.polargyro D/MainActivity: HR value: 0 rrsMs: [] rr: [] contact: false , false
2021-08-18 11:48:04.386 30795-30795/com.fujitsu.polargyro D/MainActivity: HR value: 0 rrsMs: [] rr: [] contact: false , false
2021-08-18 11:48:05.402 30795-30795/com.fujitsu.polargyro D/MainActivity: HR value: 0 rrsMs: [] rr: [] contact: false , false
2021-08-18 11:48:06.396 30795-30795/com.fujitsu.polargyro D/MainActivity: HR value: 0 rrsMs: [] rr: [] contact: false , false
2021-08-18 11:48:06.938 30795-30795/com.fujitsu.polargyro D/MainActivity: PPI    ppi: 682 hr: 0 blocker: true errorEstimate: 16
2021-08-18 11:48:06.938 30795-30795/com.fujitsu.polargyro D/MainActivity: PPI    ppi: 724 hr: 0 blocker: true errorEstimate: 14
2021-08-18 11:48:06.938 30795-30795/com.fujitsu.polargyro D/MainActivity: PPI    ppi: 691 hr: 0 blocker: false errorEstimate: 13
2021-08-18 11:48:06.938 30795-30795/com.fujitsu.polargyro D/MainActivity: PPI    ppi: 682 hr: 0 blocker: true errorEstimate: 12
2021-08-18 11:48:06.938 30795-30795/com.fujitsu.polargyro D/MainActivity: PPI    ppi: 658 hr: 0 blocker: true errorEstimate: 12
2021-08-18 11:48:06.955 30795-30795/com.fujitsu.polargyro D/MainActivity: PPI    ppi: 740 hr: 0 blocker: true errorEstimate: 13
2021-08-18 11:48:06.955 30795-30795/com.fujitsu.polargyro D/MainActivity: PPI    ppi: 679 hr: 0 blocker: false errorEstimate: 13
2021-08-18 11:48:07.415 30795-30795/com.fujitsu.polargyro D/MainActivity: HR value: 0 rrsMs: [] rr: [] contact: false , false
2021-08-18 11:48:08.407 30795-30795/com.fujitsu.polargyro D/MainActivity: HR value: 0 rrsMs: [] rr: [] contact: false , false
2021-08-18 11:48:09.395 30795-30795/com.fujitsu.polargyro D/MainActivity: HR value: 0 rrsMs: [] rr: [] contact: false , false
2021-08-18 11:48:10.386 30795-30795/com.fujitsu.polargyro D/MainActivity: HR value: 0 rrsMs: [] rr: [] contact: false , false
2021-08-18 11:48:11.407 30795-30795/com.fujitsu.polargyro D/MainActivity: HR value: 0 rrsMs: [] rr: [] contact: false , false
2021-08-18 11:48:12.095 30795-30795/com.fujitsu.polargyro D/MainActivity: PPI    ppi: 675 hr: 88 blocker: false errorEstimate: 12
2021-08-18 11:48:12.095 30795-30795/com.fujitsu.polargyro D/MainActivity: PPI    ppi: 659 hr: 88 blocker: false errorEstimate: 10
2021-08-18 11:48:12.096 30795-30795/com.fujitsu.polargyro D/MainActivity: PPI    ppi: 673 hr: 88 blocker: false errorEstimate: 9
2021-08-18 11:48:12.096 30795-30795/com.fujitsu.polargyro D/MainActivity: PPI    ppi: 678 hr: 88 blocker: false errorEstimate: 9
2021-08-18 11:48:12.096 30795-30795/com.fujitsu.polargyro D/MainActivity: PPI    ppi: 740 hr: 88 blocker: false errorEstimate: 9
2021-08-18 11:48:12.111 30795-30795/com.fujitsu.polargyro D/MainActivity: PPI    ppi: 686 hr: 88 blocker: false errorEstimate: 11
2021-08-18 11:48:12.111 30795-30795/com.fujitsu.polargyro D/MainActivity: PPI    ppi: 685 hr: 88 blocker: false errorEstimate: 15
2021-08-18 11:48:12.395 30795-30795/com.fujitsu.polargyro D/MainActivity: HR value: 88 rrsMs: [] rr: [] contact: false , false
2021-08-18 11:48:13.384 30795-30795/com.fujitsu.polargyro D/MainActivity: HR value: 88 rrsMs: [] rr: [] contact: false , false
2021-08-18 11:48:14.402 30795-30795/com.fujitsu.polargyro D/MainActivity: HR value: 88 rrsMs: [] rr: [] contact: false , false
2021-08-18 11:48:15.394 30795-30795/com.fujitsu.polargyro D/MainActivity: HR value: 88 rrsMs: [] rr: [] contact: false , false
2021-08-18 11:48:16.385 30795-30795/com.fujitsu.polargyro D/MainActivity: HR value: 88 rrsMs: [] rr: [] contact: false , false
2021-08-18 11:48:17.150 30795-30795/com.fujitsu.polargyro D/MainActivity: PPI    ppi: 781 hr: 87 blocker: false errorEstimate: 13
2021-08-18 11:48:17.150 30795-30795/com.fujitsu.polargyro D/MainActivity: PPI    ppi: 677 hr: 87 blocker: false errorEstimate: 11
2021-08-18 11:48:17.150 30795-30795/com.fujitsu.polargyro D/MainActivity: PPI    ppi: 676 hr: 87 blocker: false errorEstimate: 11
[... cut ...]
2021-08-18 11:48:17.150 30795-30795/com.fujitsu.polargyro D/MainActivity: PPI    ppi: 670 hr: 87 blocker: false errorEstimate: 10
2021-08-18 11:48:17.159 30795-30795/com.fujitsu.polargyro D/MainActivity: PPI    ppi: 691 hr: 87 blocker: false errorEstimate: 9
2021-08-18 11:48:17.159 30795-30795/com.fujitsu.polargyro D/MainActivity: PPI    ppi: 721 hr: 87 blocker: false errorEstimate: 8
2021-08-18 11:48:17.403 30795-30795/com.fujitsu.polargyro D/MainActivity: HR value: 87 rrsMs: [] rr: [] contact: false , false

JOikarinen, 2021-08-18:
> So to draw the HR graph how do I rely on **PolarHrData** or **PolarOhrPPIData**?

- there is known reason why the heart rate is 0 every now and then, I have tried to explained here: 
https://github.com/polarofficial/polar-ble-sdk/issues/143#issuecomment-810892167
- so in your case the question comes back to the use case you are implementing, if you need reliable heart rate then it is good to stop the PPI stream and get the heart rate from traditional HeartRate data service (i.e. `PolarHrData`).

kumarankit-fujitsu, 2021-08-18:
Thanks a lot @JOikarinen for your patience and helping me out.