polarofficial/polar-ble-sdk #177: App crash when disconnecting the Polar Verity Sense

hussein-ha1601589, 2021-07-29:
**Platform on which you observed the bug:** 
- [X] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this bug

**Device on which you observed the bug:**
- [ ] Polar OH1
- [X] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this bug

**Describe the bug**
App crash when disconnecting the Polar Verity Sense, when the Sense was initially connected through auto reconnection. 

**How to Reproduce**
connect to the sensor, then turn off the sensor until it automatically disconnects from the app, turns on the sensor to trigger automatic reconnection, then performs a normal disconnection using the disconnectFromDevice function. 

**Expected behavior**
disconnects without crashing. 

**Screenshots and logs**
`io.reactivex.rxjava3.exceptions.UndeliverableException: The exception could not be delivered to the consumer because it has already canceled/disposed the flow or the exception has nowhere to go to begin with. Further reading: https://github.com/ReactiveX/RxJava/wiki/What's-different-in-2.0#error-handling | com.androidcommunications.polar.api.ble.exceptions.BleGattNotInitialized
   at io.reactivex.rxjava3.plugins.RxJavaPlugins.onError(RxJavaPlugins.java:367)
   at io.reactivex.rxjava3.internal.operators.flowable.FlowableDoFinally$DoFinallySubscriber.runFinally(FlowableDoFinally.java:149)
   at io.reactivex.rxjava3.internal.operators.flowable.FlowableDoFinally$DoFinallySubscriber.onError(FlowableDoFinally.java:90)
   at io.reactivex.rxjava3.internal.operators.flowable.FlowableConcatArray$ConcatArraySubscriber.onError(FlowableConcatArray.java:91)
   at io.reactivex.rxjava3.subscribers.SerializedSubscriber.onError(SerializedSubscriber.java:144)
   at io.reactivex.rxjava3.internal.operators.flowable.FlowableDoFinally$DoFinallySubscriber.onError(FlowableDoFinally.java:89)
   at io.reactivex.rxjava3.internal.operators.flowable.FlowableCreate$BaseEmitter.errorDownstream(FlowableCreate.java:299)
   at io.reactivex.rxjava3.internal.operators.flowable.FlowableCreate$BufferAsyncEmitter.drain(FlowableCreate.java:539)
   at io.reactivex.rxjava3.internal.operators.flowable.FlowableCreate$BufferAsyncEmitter.signalError(FlowableCreate.java:489)
   at io.reactivex.rxjava3.internal.operators.flowable.FlowableCreate$BaseEmitter.tryOnError(FlowableCreate.java:287)
   at com.androidcommunications.polar.common.ble.RxUtils.postError(RxUtils.java:39)
   at com.androidcommunications.polar.common.ble.RxUtils.postDisconnectedAndClearList(RxUtils.java:23)
   at com.androidcommunications.polar.api.ble.model.gatt.client.BleHrClient.reset(BleHrClient.java:62)
   at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDDeviceSessionImpl.handleDisconnection(BDDeviceSessionImpl.java:314)
   at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDDeviceListenerImpl.deviceDisconnected(BDDeviceListenerImpl.java:434)
   at com.androidcommunications.polar.enpoints.ble.bluedroid.host.connection.ConnectionHandler.deviceDisconnected(ConnectionHandler.java:85)
   at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDGattCallback.lambda$onConnectionStateChange$6$BDGattCallback(BDGattCallback.java:68)
   at com.androidcommunications.polar.enpoints.ble.bluedroid.host.-$$Lambda$BDGattCallback$pBHyn8LrywH1_WQ1nhkFIFEXlFE.run(Unknown Source:4)
   at android.os.Handler.handleCallback(Handler.java:789)
   at android.os.Handler.dispatchMessage(Handler.java:98)
   at android.os.Looper.loop(Looper.java:164)
   at android.app.ActivityThread.main(ActivityThread.java:6938)
   at java.lang.reflect.Method.invoke(Native Method)
   at com.android.internal.os.Zygote$MethodAndArgsCaller.run(Zygote.java:327)
   at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:1374)
Caused by: com.androidcommunications.polar.api.ble.exceptions.BleGattNotInitialized
   at com.androidcommunications.polar.enpoints.ble.bluedroid.host.BDDeviceSessionImpl.setCharacteristicNotify(BDDeviceSessionImpl.java:409)
   at com.androidcommunications.polar.api.ble.model.gatt.client.BleHrClient.lambda$observeHrNotifications$2$BleHrClient(BleHrClient.java:129)
   at com.androidcommunications.polar.api.ble.model.gatt.client.-$$Lambda$BleHrClient$7nqyUcGNkruGx8R-zBU9WPY4tas.run(Unknown Source:2)
   at io.reactivex.rxjava3.internal.operators.flowable.FlowableDoFinally$DoFinallySubscriber.runFinally(FlowableDoFinally.java:146)
	... 23 more
`

hussein-ha1601589, 2021-08-01:
I noticed that if I waited 5 seconds between calling the disconnectFromDevice function, and unbinding the service that hosts the API logging, I don't get the error. I hope this might give a clue about the problem.

JOikarinen, 2021-08-10:
@hussein-ha1601589 thanks for the bug report, I am on it. 

At the moment I cannot reproduce the bug, what I have tried is to repeat your steps with [example application](https://github.com/polarofficial/polar-ble-sdk/tree/master/examples/example-android/androidBleSdkTestApp).

Steps:
1. Connect to Polar Verity Sense sensor with Auto Connect button in example app
2. Power off the Polar Verity Sense sensor -> disconnection happens
3. Power on the Polar Verity Sense sensor -> connection happens
4. Disconnect from Polar Verity Sense sensor with Disconnect button in example app

Observation:
- no crash

Is it possible you provide the logs given by the SDK once your app crashes. The SDK logs can be set on like  shown in [example app](https://github.com/polarofficial/polar-ble-sdk/blob/0456ab9ea6b8f136a473d424f3641b18f407f363/examples/example-android/androidBleSdkTestApp/app/src/main/java/com/polar/androidblesdk/MainActivity.kt#L99).

jimmyzumthurm, 2023-09-08:
I'm closing this ticket as no more information was given for over 2 years. @hussein-ha1601589 , please feel free to remake a new bug report if you still encounter that with latest Polar SDK version.

closed 2023-09-08 by jimmyzumthurm