polarofficial/polar-ble-sdk #178: Polar Andriod SDK using as a service

sach16795, 2021-08-03:
***Platform your question concerns:*** 
- [*] Android 
- [ ] iOS
- [ ] Other
- [ ] Platform is not relevant for this question

***Device:***
- [ ] Polar OH1
- [*] Polar Verity Sense
- [ ] Polar H10
- [ ] Polar H9
- [ ] Other
- [ ] Device is not relevant for this question

**Description:**
I am using this Android SDK to connect my Polar device (Verity Sense). I can connect the device and get the HR/accelerometer values in the Main Activity. But If the activity is inactive/killed, the connection is lost and data is no longer recorded. How can I continue recording the data even if the main activity is paused/destroyed? I tried using service so that it can run irrespective of the main activity but I cannot use the requestStreamSettings method to get the data stream in the service since the method takes Activity as a parameter. Can you please help me setting up the required setup?

(I am new to ReactiveX and hence not able to rewrite/modify the required methods. Apologize if this is a trivial issue. Any help is appreciated.)

Thanks.

yedhukrrish, 2021-08-10:
Hi, have you figured it out? coz I too want to run the android sdk in services . I tried but only HR data is retrieved . I am not able to retrieve PPG data

sach16795, 2021-08-15:
@yedhukrrish Yes, I figured out a way to run the SDK in service. Not sure if it's the most optimized/correct way, but it's working for now. I am using ACC data but I think it's going to be a lot similar. 

api.requestStreamSettings(deviceId, ACC).toFlowable()
            .flatMap { setting: PolarSensorSetting -> api.startAccStreaming(deviceId, setting) }
            .observeOn(AndroidSchedulers.mainThread())
            .subscribe({ polarAccelerometerData: PolarAccelerometerData ->
                for (data in polarAccelerometerData.samples) {
                    //Do stuff here
                }
            })

sach16795, 2021-08-23:
Found a way to implement this. Closing.

closed 2021-08-23 by sach16795